jgromes/RadioLib #1243: [SX128x] Packet reception does not work with a preamble length higher than 30.

jacobeva, 2024-09-30:
**Describe the bug**
When the preamble length of an SX1280 radio is set higher than 30, no packets can be received by the remote. 
This is due to the following code in SX128X.cpp (line 868 onwards):
```
    // calculate exponent and mantissa values (use the next longer preamble if there's no exact match)
    uint8_t e = 1;
    uint8_t m = 1;
    uint32_t len = 0;
    for(; e <= 15; e++) {
      for(; m <= 15; m++) {
        len = m * (uint32_t(1) << e);
        if(len >= preambleLength) {
          break;
        }
      }
      if(len >= preambleLength) {
        break;
      }
    }
```

Now, this appears totally fine, right? Well, when I compiled this code and ran it locally with some debug output to let me see the `e` and `m` values, this is what they read:

```
Preamble length:200
Reached preamble length:30
e value:16
m value:16
```
Where preamble length is the target, and reached preamble length is the value that the loop exited with. You may also note the illegal `e` and `m` values, which are sure to cause trouble.

</p>
</details> 

**To Reproduce**
This is how I tested your code:
```c++
#include <stdio.h>
#include <stdint.h>
#include <math.h>

int main() {
    uint32_t preambleLength = 200;
    // calculate exponent and mantissa values (use the next longer preamble if there's no exact match)
    uint8_t e = 1;
    uint8_t m = 1;
    uint32_t len = 0;
    for(; e <= 15; e++) {
      for(; m <= 15; m++) {
        len = m * (1 << e);
        if(len >= preambleLength) {
          break;
        }
      }
      if(len >= preambleLength) {
        break;
      }
    }

    printf("Preamble length:%d\n\r", preambleLength);
    printf("Reached preamble length:%ld\n\r", len);

    printf("e value:%d\n\r", e);
    printf("m value:%d\n\r", m);
}
```

**Expected behavior**
This function should not be spitting out invalid `e` and `m` values, but it should also work up to the correct preamble length.

**Suggested fix**
Please find the below patch for your discretion.
```
diff --git a/src/modules/SX128x/SX128x.cpp b/src/modules/SX128x/SX128x.cpp
index a51ad836..58eb5c60 100644
--- a/src/modules/SX128x/SX128x.cpp
+++ b/src/modules/SX128x/SX128x.cpp
@@ -868,17 +868,17 @@ int16_t SX128x::setPreambleLength(uint32_t preambleLength) {
     // calculate exponent and mantissa values (use the next longer preamble if there's no exact match)
     uint8_t e = 1;
     uint8_t m = 1;
-    uint32_t len = 0;
-    for(; e <= 15; e++) {
-      for(; m <= 15; m++) {
-        len = m * (uint32_t(1) << e);
-        if(len >= preambleLength) {
-          break;
+    uint32_t len;
+
+    while (e <= 15) {
+        while (m <= 15) {
+            len = m * (pow(2,e));
+            if (len >= preambleLength) break;
+            m++;
         }
-      }
-      if(len >= preambleLength) {
-        break;
-      }
+        if (len >= preambleLength) break;
+        m = 0;
+        e++;
     }
 
     // update packet parameters
```

**Additional info:**
 - Library version 7.0.0

jgromes, 2024-09-30:
Thank you for reporting this - however, the actual fix is much simpler. The internal loop incrementing the mantissa value is simply missing a reset back to m = 0. With that, it calculates the following:

```
Preamble length:200
Reached preamble length:208
e value:4
m value:13
```

Could you test this also resolves the problem?

jacobeva, 2024-09-30:
I haven't tested it physically on the radio, but it does set the preamble correctly, so it's reasonable to assume this works, yes.

closed 2024-09-30 by jacobeva