jgromes/RadioLib #1254: [LoRaWAN] timeUntilUplink returns 0 when called immediately after a activateOTAA

MortenGuldager, 2024-10-04:
I’ve noticed `node.timeUntilUplink()` returns 0 immediately after calling `node.activateOTAA()`. My understanding is that, especially in the case of a failed join attempt, subsequent retries should still respect duty cycle limits before the node can attempt another uplink. However, the immediate 0 return seems to suggest that the node is ready to send again right away, regardless of whether the join was successful or not.

That said, I might be misunderstanding how this behavior is expected to work. If this is the intended functionality, I'd appreciate any clarification on how to properly handle the timing of OTAA join retries.

using: esp32, radiolib7.0.0, platformio, rfm95

StevenCellist, 2024-10-04:
Fair shout, the time-on-air of a JoinRequest is not tracked. I will pick this up.

HeadBoffin, 2024-10-05:
> If this is the intended functionality, I'd appreciate any clarification on how to properly handle the timing of OTAA join retries.

As clarified above, it's not intended functionality and you can just stick to the duty cycle limitations, but the "proper way" is outlined in TR007 which you can find on the LoRa Alliance website.

The real world answer is that much of it is a bit academic and a whole chunk of error handling that needs to be tested & debugged when a carefully provisioned device in an area with adequate gateway coverage will join once. So don't get too tied up with backoff - keep the calculations simple and move on.

closed 2024-10-07 by StevenCellist

StevenCellist, 2024-10-07:
@MortenGuldager should be resolved in upstream. Please let me know if the problem persists.

commit 9db98f2a (2024-10-07): [LoRaWAN] Collection of fixes (also fixes #1254)