jgromes/RadioLib #1256: Failed to initialise when using ESP32-S3 Supermini and nRF24L01

RockLakeGrass, 2024-10-06:
**Describe the bug**
MCU: ESP32-S3 Supermini
Module: nRF24L01
Platform: PlatformIO with env: lolin_s3_mini

<details><summary>Debug mode output</summary>
<p>

```
[nRF24] Initializing ... 
RLB_DBG: 
RadioLib Info
Version:  "7.0.2.0"
Platform: "ESP32"
Compiled: "Oct  6 2024" "14:48:58"
RLB_SPI: R 3 3
RLB_DBG: M nRF24
RLB_SPI: R 0 C
RLB_SPI: W 0 C
RLB_SPI: R 0 C
RLB_SPI: R 4 55
RLB_SPI: W 4 55
RLB_SPI: R 4 55
RLB_SPI: R 1D 0
RLB_SPI: W 1D 4
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: R 1D 0
RLB_SPI: 
RLB_SPI: address: 0x1D
RLB_SPI: bits: 2 0
RLB_SPI: value: 0x4
RLB_SPI: current: 0x0
RLB_SPI: mask: 0x7
RLB_SPI: new: 0x4
RLB_SPI: read: 0x0
RLB_DBG: -16 at .pio/libdeps/lolin_s3_mini/RadioLib/src/modules/nRF24/nRF24.cpp:582
RLB_DBG: -16 at .pio/libdeps/lolin_s3_mini/RadioLib/src/modules/nRF24/nRF24.cpp:34
failed, code -16
```

</p>
</details> 

**To Reproduce**

Hookup: 

nRF24L01 Pins | ESP32-S3 Mini Pins
-- | --
VCC | 3.3V
GND | GND
CE | GPIO 3
CSN | GPIO 1
SCK | GPIO 12
MOSI | GPIO 11
MISO | GPIO 13
IRQ | GPIO 5

<details><summary>Sketch that is causing the module fail</summary>
<p>

```c++
#include <Arduino.h>
#include <RadioLib.h>

nRF24 radio = new Module(1, 5, 3); // CS: 1, IRQ: 5, CE: 3
int transmissionState = RADIOLIB_ERR_NONE;

void sendPacket(const char* message) {
  Serial.println("[nRF24] Sending packet ... ");
  transmissionState = radio.startTransmit(message);
}

volatile bool transmittedFlag = false;

void setFlag(void) {
  transmittedFlag = true;
}

void setup() {
  Serial.begin(115200);
  Serial.println("[nRF24] Initializing ... ");
  int state = radio.begin();
  if (state != RADIOLIB_ERR_NONE) {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  byte addr[] = {0x01, 0x23, 0x45, 0x67, 0x89};
  state = radio.setTransmitPipe(addr);
  if (state != RADIOLIB_ERR_NONE) {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  radio.setPacketSentAction(setFlag);
  sendPacket("Hello World!");
}

void loop() {
  if (transmittedFlag) {
    transmittedFlag = false;
    if (transmissionState == RADIOLIB_ERR_NONE) {
      Serial.println("transmission finished!");
    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);
    }
    radio.finishTransmit();
    delay(1000);
    sendPacket("Hello World!");
  }
}

```

</p>
</details> 

**Expected behavior**
Successfully initilised.

**Additional info (please complete):**
See beginning

HeadBoffin, 2024-10-06:
If you look up the error code in TypeDef.h you'll see that it says: "RADIOLIB_ERR_SPI_WRITE_FAILED" with a comment of Real value in SPI register does not match the expected one. This can be caused by faulty SPI wiring.

So worth checking for dry joints or iffy Dupont cables. And try a different nRF module just in case the fault is on the board.

RockLakeGrass, 2024-10-13:
I tried another module with different manufacturer, it worked. Thank you.

closed 2024-10-13 by RockLakeGrass