jgromes/RadioLib #1257: CC1101 PAGER

Szetya, 2024-10-06:
Hi! I have several questions about using ESP32 CC1101 PAGER.
-I will start with the less problematic one. For some reason, above version 5.7.0 the reception does not work. Using the example code I have stepped through the versions so I suspect there is some kind of breakage there.
-Other: Since direct mode reception is using is there any way to stop it once it has been started by pager.startReceive()? So you can't switch between transmit and receive. The goal would be to listen continuously and if needed to stop reception for the duration of the transmission.
-Third, which may not be a radiolib problem. I have found that with Skyper Rubic receive, ESP crashes. Skyper uses ROT-1 encoding but only for the printable characters, but the rubic index can be included in the non-printable range.(?)

A case where this happens:
[image]

LOG:
[Pager] Address: 666411 [Pager] Data: Uptime Warning
[Pager] Received pager data, decoding ... success!
[Pager] Address: 4512   [Pager] Data:   (Decoded) > 1*DAPNET NEWS
Failed to obtain time
[Pager] Received pager data, decoding ... Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered ()
Core  0 register dump:
PC      : 0x4008eb77  PS      : 0x00060036  A0      : 0x800ebd62  A1      : 0x3ffbc990
A2      : 0x9c400000  A3      : 0x00038000  A4      : 0x9c400000  A5      : 0x00000008
A6      : 0x00000004  A7      : 0x00000000  A8      : 0x3ff5f000  A9      : 0x00000008
A10     : 0x3ffbc9c0  A11     : 0x00000000  A12     : 0x00000008  A13     : 0x3ffbc9c8  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000008  LBEG    : 0x400869e8  LEND    : 0x400869f3  LCOUNT  : 0xffffffff

Backtrace: 0x4008eb74:0x3ffbc990 0x400ebd5f:0x3ffbc9c0 0x400ebda5:0x3ffbc9f0 0x400ec1e7:0x3ffbca60 0x4008395d:0x3ffbcab0 0x40084940:0x3ffbcad0 0x4008d8fa:0x3ffbcb90 0x4008c5aa:0x3ffbcbb0 0x4008aa88:0x3ffbcbe0 0x4008aa38:0xa5a5a5a5 |<-CORRUPTED

jgromes, 2024-10-06:
> above version 5.7.0 the reception does not work

That is interesting, could you try to narrow down where exactly this breaks? Even between 5.7.0 and the next release (6.0.0) there's quite a few changes: https://github.com/jgromes/RadioLib/compare/5.7.0...6.0.0

> Other: Since direct mode reception is using is there any way to stop it once it has been started by pager.startReceive()?

If you want to go packet mode then you can call `radio.variablePacketLengthMode()`, that's what gets called during `radio.begin()`.

> I have found that with Skyper Rubic receive, ESP crashes. Skyper uses ROT-1 encoding but only for the printable characters

I don't know what these things are. It definitely does not look like a problem in this library. Moreover, attempting to print characters in a wrong encoding really shouldn't cause an exception. It says `Stack canary watchpoint triggered`, so it seems the program is doing something really wrong with the stack (e.g. accessing outside of array bounds).

Szetya, 2024-10-07:
> That is interesting, could you try to narrow down where exactly this breaks? Even between 5.7.0 and the next release (6.0.0) there's quite a few changes: [5.7.0...6.0.0](https://github.com/jgromes/RadioLib/compare/5.7.0...6.0.0)

Thank you very much for your quick reply!
OK. I have tried several times and got the following results:
    ;https://github.com/jgromes/RadioLib.git#27575b8c49528d35750476c88aa1c13678caf5ba ;RX OK with pocsagStartRx()
    ;https://github.com/jgromes/RadioLib.git#45e9fd44bb533c99f627c89455d0ebc9c62ffe0c ;undefined reference to `PagerClient::available()' and more similar warnings
    ;https://github.com/jgromes/RadioLib.git#88549baf8326ee9f679e09e04f7b3c1a0e8d7a2f ;RX OK with pocsagStartRx()
    ;https://github.com/jgromes/RadioLib.git#ec3d4eaf202fe9c88466de4dd3eb19ae0c0f79d5 ;In member function 'virtual void ArduinoHal::noTone(uint8_t)': return-statement with a value, in function returning 'void'
    https://github.com/jgromes/RadioLib.git#d8f0d36407ae0e9d9c85350577fb3b23e96f2236 ;RX Fail with pocsagStartRx()
    ;https://github.com/jgromes/RadioLib.git#3d43a618e36e4f90d5ab28da7da994d609f5d5b0 ;RX Fail with pocsagStartRx()

> If you want to go packet mode then you can call `radio.variablePacketLengthMode()`, that's what gets called during `radio.begin()`.

Is this supposed to be done without using the PagerClient pager(&radio); and instead manually setting everything (frequency, syncwold, RxBandwidth, PreambleLength, Dio2Action, etc?) Like in the interrupt examples, I need a void and a flag? It could be a big task for me. :)
 
> I don't know what these things are. It definitely does not look like a problem in this library. Moreover, attempting to print characters in a wrong encoding really shouldn't cause an exception. It [...]

I have turned off everything (led blinking with delay, E-ink display which is slow, and saving incoming RIC matching messages to SPIFFS), I will test in the days to see if the error comes without them.  For the moment, let's forget that it is listed as a bug. :)

Szetya, 2024-10-08:
Hi!

Today I was able to spend some time on the test.
I noticed the following RadioLibTime_t is undefined (photo):
[image]
In the left window is version 7.0.2 and in the right window is 5.7.0 or https://github.com/jgromes/RadioLib.git#88549baf8326ee9f679e09e04f7b3c1a0e8d7a2f (I can't remember as much as I tried)
surely the problem is similar to the one here: https://github.com/jgromes/RadioLib/issues/1261
-On the second point, I would like to ask for your help with an example.
-And the third one was solved by turning off the above mentioned things so I will need to further filter my code to see what is causing it. The Skyper bulk package is OK (two are missing I guess due to processing time):
[image]

Maybe later on if you want to add Skyper ROT-1 decoding:
String  decodeSkyperRubric(String rubric) {
    int ric = rubric.charAt(1) - 31; // Rubik-szám számítása (a második karakterből)
    
    // A harmadik karaktertől kezdve visszakódoljuk az üzenetet (ROT-1)
    String decodedMessage = "";
    for (int i = 3; i < rubric.length(); i++) {
        char currentChar = rubric.charAt(i);
        
        // Csak az olvasható karaktereket forgatjuk vissza (ASCII 33-126 között)
        if (currentChar > 32 && currentChar <= 126) {
            decodedMessage += (char)(currentChar - 1); // ROT-1
        } else {
            decodedMessage += currentChar; // Speciális karaktereket hagyjuk
        }
    }
    
    // A visszatérési formátum: "rubik_szam*dekódolt_üzenet"
    return String(ric) + "*" + decodedMessage;
}

jgromes, 2024-10-08:
> I noticed the following RadioLibTime_t is undefined 

That is not the issue, `RadioLibTime_t` is just a typedef for `unsigned long`. But the fact your IDE thinks its undefined doesn't mean it actually is. The code wouldn't compile if it were the case.

I'll see if I can put together a testing setup in the next few days ...

Szetya, 2024-10-08:
> That is not the issue, `RadioLibTime_t` is just a typedef for `unsigned long`. But the fact your IDE thinks its undefined doesn't mean it actually is. The code wouldn't compile if it were the case. [...]

Thank you very much on behalf of everyone else for your help! Unfortunately I am not that professional so I can only approximate the errors. :)
No rush, I have brought up the POCSAG topic many times, sometimes just in thought. Now I'm a bit further along with it like saving and reading back messages, TCP protocol to dapnet server... Maybe later it can become a TCP-RF transmitter. :)

Szetya, 2024-10-15:
OK.
I have tried many intermediate versions.
5.7.0 was the last one where RX worked but TX did not. With a small modification 5.7.0 can send and receive in Pager mode.
Somewhere above 5.7.0 the RX capability disappears and never works.
Currently this is the case with the released versions:

; jgromes/RadioLib @ 7.0.2 ; TX OK, RX FAIL
; jgromes/RadioLib @ 6.6.0 ; TX OK, RX FAIL
; jgromes/RadioLib @ 6.5.0 ; TX OK, RX FAIL
; jgromes/RadioLib @ 6.4.0 ; TX OK, RX FAIL
; jgromes/RadioLib @ 6.3.0 ; TX FAIL, RX FAIL
; jgromes/RadioLib @ 6.2.0 ; TX FAIL, RX FAIL
; jgromes/RadioLib @ 6.1.0 ; TX FAIL, RX FAIL
; jgromes/RadioLib @ 6.0.0 ; TX FAIL, RX FAIL
jgromes/RadioLib @ 5.7.0 ; TX FAIL, RX OK

```
int16_t CC1101::transmitDirect(bool sync, uint32_t frf) {
  // set RF switch (if present)
  _mod->setRfSwitchState(Module::MODE_TX);

  // user requested to start transmitting immediately (required for RTTY)
  if(frf != 0) {
    SPIwriteRegister(RADIOLIB_CC1101_REG_FREQ2, (frf & 0xFF0000) >> 16);
    SPIwriteRegister(RADIOLIB_CC1101_REG_FREQ1, (frf & 0x00FF00) >> 8);
    SPIwriteRegister(RADIOLIB_CC1101_REG_FREQ0, frf & 0x0000FF);

    SPIsendCommand(RADIOLIB_CC1101_CMD_TX);
    return(RADIOLIB_ERR_NONE); // Added to version 5.7.0 TX/RX works with CC1101

  }
```

Szetya, 2024-10-16:
This is the last one where RX works: https://github.com/jgromes/RadioLib.git#61f6da0cb2a41e88bc13494c67a4ef01f6ed546c ; TX FAIL, RX OK committed on Apr 22, 2023
Not in the next one: 
https://github.com/jgromes/RadioLib.git#767a2b006a0349c683624b33d3be3e8a9d5330c3 ; TX FAIL, RX FAIL

Environment: 
platform = espressif32
board = esp32dev
framework = arduino

Radio: CC1101/PAGER protocol

Szetya, 2024-10-16:
@jgromes https://github.com/jgromes/RadioLib/blob/710a1540b58e491172497c3c69c8f58fc2ca7e65/src/protocols/Pager/Pager.cpp#L17
With ICACHE_RAM_ATTR  
TX and RX works with version 7.0.2. :)

jgromes, 2024-10-17:
@Szetya that's interesting - that attribute was changed in 767a2b006a0349c683624b33d3be3e8a9d5330c3 to fix a deprecation warning from ESP-IDF after we have updated it in our CI. Which version of ESP-IDF have you been using?

Szetya, 2024-10-17:
> @Szetya that's interesting - that attribute was changed in [767a2b0](https://github.com/jgromes/RadioLib/commit/767a2b006a0349c683624b33d3be3e8a9d5330c3) to fix a deprecation warning from ESP-IDF after we have updated it in our CI. Which version of ESP-IDF have you been using?

Hi! 
I did not write a version number, so the last release is valid. 
I mentioned the environment in a previous post.
I swear I tried a lot of commits in the last days 😉 .
The test code is one example, which sends a message and then switches to receive (I waited for MMDVM afterwards, because I knew for sure it would receive the message sent. I noticed that the portapack was a bit unstable at first).
So I was able to test both RX and TX operation with one code.

jgromes, 2024-10-18:
@Szetya `ICACHE_RAM_ATTR` is just a redefinition of `IRAM_ATTR` in Arduino ESP32 core. It's used for compatibility between ESP8266, as seen [here](https://github.com/espressif/arduino-esp32/blob/ee8931d151665ffc6bb52e3edb9509a6d8e5c401/cores/esp32/esp8266-compat.h#L21) and [here](https://github.com/espressif/arduino-esp32/blob/ee8931d151665ffc6bb52e3edb9509a6d8e5c401/cores/esp32/esp32-hal.h#L49). I don't think that's the root cause.

Szetya, 2024-10-19:
> @Szetya `ICACHE_RAM_ATTR` is just a redefinition of `IRAM_ATTR` in Arduino ESP32 core. It's used for compatibility between ESP8266, as seen [...]

Hi! 
It really does not solve the problem, I have tried again and again and again. I just can't find where the breakpoint is. Sorry. :(

jgromes, 2024-10-22:
@Szetya I was able to replicate and subsequently fix the issue. The root cause was that when starting reception on CC1101, filtering mechanisms like preamble and sync word remained enabled, which prevented it from being able to receive the pager transmission. However, as far as I can tell this was also the case in 5.7.0. So it is possible the intermittent successes you were seeing were not due to a difference in version, but just due to random chance of the radio "seeing" a preamble and sync word pattern in the noise or the pager signal. Below you can see output from PDW and ESP32 with CC1101 receiving the same data.

[image]

I pushed a commit that fixes this, thank you for reporting!

closed 2024-10-22 by jgromes