jgromes/RadioLib #1260: SX1262 with FSK modem reiceves empty packets on Heltec wifi lora v3

Xiro28, 2024-10-08:
<details><summary>Sketch that is causing the module fail</summary>
<p>

```c++
/*
   RadioLib SX126x Ping-Pong Example

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// uncomment the following only on one
// of the nodes to initiate the pings
#define INITIATING_NODE

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
//SX1262 radio = new Module(10, 2, 3, 9);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

#define DIO1      GPIO_NUM_14
#define RST_LoRa  GPIO_NUM_12
#define BUSY_LoRa GPIO_NUM_13

#define SS        GPIO_NUM_8
#define MOSI      GPIO_NUM_10
#define MISO      GPIO_NUM_11
#define SCK       GPIO_NUM_9

#include <SPI.h>
SPIClass* hspi = new SPIClass(HSPI);
SX1262 radio = new Module(SS, DIO1, RST_LoRa, BUSY_LoRa, *hspi);

// save transmission states between loops
int transmissionState = RADIOLIB_ERR_NONE;

// flag to indicate transmission or reception state
bool transmitFlag = false;

// flag to indicate that a packet was sent or received
volatile bool operationDone = false;

// this function is called when a complete packet
// is transmitted or received by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
#if defined(ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
  // we sent or received a packet, set the flag
  operationDone = true;
}
int addr = 0;

void setup() {
  hspi->begin(SCK, MISO, MOSI, SS);
  Serial.begin(115200);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.beginFSK();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  // set the function that will be called
  // when new packet is received
  radio.setDio1Action(setFlag);

  #if defined(INITIATING_NODE)
    // send the first packet on this node
    transmissionState = radio.transmit("Hello World!", 0x3);
  #else
    // start listening for LoRa packets on this node
    Serial.print(F("[SX1262] Starting to listen ... "));
    state = radio.setNodeAddress(0x03);
    state = radio.startReceive();
    if (state == RADIOLIB_ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while (true) { delay(10); }
    }
  #endif
}

void loop() {
  #if defined(INITIATING_NODE)
    // send the first packet on this node
    radio.transmit("Hello World!", 0x3);
  #else
    // the previous operation was reception
      // print data and send another packet
      String str;
      int state = radio.readData(str);

      if (state == RADIOLIB_ERR_NONE) {
        // packet was successfully received
        Serial.println(F("[SX1262] Received packet!"));

        // print data of the packet
        Serial.print(F("[SX1262] Data:\t\t"));
        Serial.println(str);

        // print RSSI (Received Signal Strength Indicator)
        Serial.print(F("[SX1262] RSSI:\t\t"));
        Serial.print(radio.getRSSI());
        Serial.println(F(" dBm"));

        // print SNR (Signal-to-Noise Ratio)
        Serial.print(F("[SX1262] SNR:\t\t"));
        Serial.print(radio.getSNR());
        Serial.println(F(" dB"));
      }
  #endif
}
```

</p>
</details> 

**Additional info (please complete):**
This is my platformio setup:

[env:heltec_wifi_lora_32_V3]
platform = espressif32
board = heltec_wifi_lora_32_V3
framework = arduino

lib_deps = jgromes/RadioLib@^7.0.2, SPI

monitor_speed = 115200

It's my first time with the FSK modulation, so I don't know how to fix it. The output is empty or random values and I get 0 at RSSI and -20 at SNR. Thanks in advance!

jgromes, 2024-10-08:
You seem to be missing most of the logic from `loop` that checks the state of the flags that was in the example, I don't understand why you would edit in this way. Now your code essentially just transmits nonstop (that's a jammer!) or just calls `readData` in loop with no regard whether packet was actually received or not, depending on whether `INITIATING_NODE` is defined or not.

Xiro28, 2024-10-08:
It's a test code I've made since I cannot share the real one. The setup is likely the same, the only thing is that it isn't in a loop but it's based on button press. I was wondering if the setnodeaddress is working correctly since with it enabled I only get random data

jgromes, 2024-10-08:
> It's a test code I've made

Well, it's wrong. As I wrote above, you are completely missing the logic that would only call `readData` when there's actual data available - as shown in interrupt examples. Please try it with the example code, it is very easy to modify it to use FSK and a fixed node address.

Xiro28, 2024-10-08:
Here's the original code:

```c++
#if defined(__LORA__)
#if defined(HELTEC_PATCH)
SPIClass* hspi = new SPIClass(1);
SX1262 radio = new Module(8, 14, 12, 13, *hspi);
#else
SX1262 radio = new Module(10, 2, 3, 9);
#endif

volatile bool dataReady;
volatile bool isReceivingMode;

struct packet {
uint8_t length;
uint8_t data[255];
};
const uint8_t sz_buffer = 32;
packet ringbuffer[sz_buffer];

volatile uint16_t tx_idx = 0;
volatile uint16_t rx_idx = 0;

void onPacketReceived() {
Serial0.println("Interrupt triggered");

if (isReceivingMode) {
	Serial0.println("Packet received");
	dataReady = true;
	radio.clearIrqFlags(0xffff);
	
	radio.readData(ringbuffer[tx_idx].data, radio.getPacketLength());
	ringbuffer[tx_idx].length = radio.getPacketLength();

	tx_idx = (tx_idx + 1) % sz_buffer;
}

// this is also triggered when a packet is sent
}

int init(URI* lUri)
{

	Serial0.println("LoRa driver init");

	#if defined(__ESP32__)
		// Initialize LoRa module

		#ifdef HELTEC_PATCH
			hspi->begin(SCK, MISO, MOSI, SS);
		#endif

		if (radio.beginFSK() != RADIOLIB_ERR_NONE) return -1;
		if (radio.setNodeAddress(((Uri_LoRa*)lUri->urn)->addr & 0xff) != RADIOLIB_ERR_NONE) return -1;
		if (radio.setBroadcastAddress(0xFF) != RADIOLIB_ERR_NONE) return -1;
	#endif

	_state = _READY;

	return 0;
}

int enable()
{
  #if defined(__ESP32__)
	  // Setup the receive interupt 
	  radio.setPacketReceivedAction(&onPacketReceived);
	  
	  // Set the LoRa module to listening mode
	  radio.startReceive();
  
	  isReceivingMode = true;
	  dataReady = false;
  #endif
  return 0;
}

int read(unsigned char* buff, int sz)
{
	#if defined(__ESP32__)
		if (rx_idx >= tx_idx) return -1; 

		dataReady = false;
		packet &pkt = ringbuffer[rx_idx];
		int numBytes = sz > pkt.length ? pkt.length : sz;
		memcpy(buff, pkt.data, numBytes);

		rx_idx = (rx_idx + 1) % sz_buffer;
		return numBytes;
	#endif

	return 0;
}

int write(int addr, const unsigned char* buff, int size)
{
	#if defined(__ESP32__)
		// LoRa can only listen or transmit, not both
		// So what we do is we transmit the data and then go back to listening
		const bool wasListeningMode = isReceivingMode;

		// Stop listening
		isReceivingMode = false;

		int status = radio.transmit(buff, size, addr);
		if ( status == RADIOLIB_ERR_NONE){
			Serial0.print("LoRa driver write success. Bytes written: ");
			Serial0.println(size);

			delay(100);

			// Before we go back to listening, we need to tell the hardware that we 
			// are done transmitting
			radio.finishTransmit();

			
			// Since the trasmit function is blocking, we can go back to listening here
			// without any issues
			if (wasListeningMode) {
				isReceivingMode = true;
				radio.startReceive();
			}

			return size;
		}
	#endif

	return (-1);
}

int checkDataSz()
{
	#if defined(__ESP32__)
		if (rx_idx >= tx_idx) return -1;

		uint8_t sz = ringbuffer[rx_idx].length;
		if (dataReady && sz >= 0) return sz;
	#endif

	return -1;
}

#endif

```

jgromes, 2024-10-08:
I will repeat myself once again: please try it with the example code. You seem to be stitching together pieces if different examples without understanding how they differ. For example using a blocking method `transmit` alongside interrupt-driven reception. That will not work. Until this issue can be replicated with the examples modified only to use FSK and set a node address, I cannot help you.

closed 2024-10-08 by jgromes

Xiro28, 2024-10-10:
Here's with the example, I get CRC error

```cpp
/*
   RadioLib SX126x Ping-Pong Example

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// uncomment the following only on one
// of the nodes to initiate the pings
#define INITIATING_NODE

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
//SX1262 radio = new Module(10, 2, 3, 9);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

#define DIO1      GPIO_NUM_14
#define RST_LoRa  GPIO_NUM_12
#define BUSY_LoRa GPIO_NUM_13

#define SS        GPIO_NUM_8
#define MOSI      GPIO_NUM_10
#define MISO      GPIO_NUM_11
#define SCK       GPIO_NUM_9

#include <SPI.h>
SPIClass* hspi = new SPIClass(HSPI);
SX1262 radio = new Module(SS, DIO1, RST_LoRa, BUSY_LoRa, *hspi);

#define Sender

void setup() {
  hspi->begin(SCK, MISO, MOSI, SS);
  Serial.begin(115200);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.beginFSK();

  state = radio.setFrequency(433.5);
  state = radio.setBitRate(100.0);
  state = radio.setFrequencyDeviation(10.0);
  state = radio.setRxBandwidth(250.0);
  state = radio.setOutputPower(10.0);
  state = radio.setCurrentLimit(100.0);
  state = radio.setDataShaping(RADIOLIB_SHAPING_1_0);
  uint8_t syncWord[] = {0x01, 0x23, 0x45, 0x67,
                        0x89, 0xAB, 0xCD, 0xEF};
  state = radio.setSyncWord(syncWord, 8);

  state = radio.setCRC(2, 0xFFFF, 0x8005, false);

   #if defined(Sender)
    // set node address to 0x03
    state = radio.setNodeAddress(0x03);
    // set broadcast address to 0xFF
    state = radio.setBroadcastAddress(0xFF);
  #else
      // set node address to 0x02
    state = radio.setNodeAddress(0x02);
    // set broadcast address to 0xFF
    state = radio.setBroadcastAddress(0xFF);
  #endif

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

}

void loop() {
  #if defined(Sender)
    int state = radio.transmit("Hello World!", 0x02);

    if (state == RADIOLIB_ERR_NONE) {
      Serial.println(F("[SX1262] Packet transmitted successfully!"));
    } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
      Serial.println(F("[SX1262] Packet too long!"));
    } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
      Serial.println(F("[SX1262] Timed out while transmitting!"));
    } else {
      Serial.println(F("[SX1262] Failed to transmit packet, code "));
      Serial.println(state);
    }

    delay(1000);
  #else
  Serial.print(F("[SX1262] Waiting for incoming transmission ... "));

  // you can receive data as an Arduino String
  String str;
  int state = radio.receive(str);

  // you can also receive data as byte array
  /*
    byte byteArr[8];
    int state = radio.receive(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[SX1262] Data:\t\t"));
    Serial.println(str);

    // print the RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("[SX1262] RSSI:\t\t"));
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));

    // print the SNR (Signal-to-Noise Ratio)
    // of the last received packet
    Serial.print(F("[SX1262] SNR:\t\t"));
    Serial.print(radio.getSNR());
    Serial.println(F(" dB"));

    // print frequency error
    Serial.print(F("[SX1262] Frequency error:\t"));
    Serial.print(radio.getFrequencyError());
    Serial.println(F(" Hz"));

  } else if (state == RADIOLIB_ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
  #endif
}
```

jgromes, 2024-10-12:
Now we're talking ;) With this I was actually able to replicate the issue - it looks enabling FSK address on SX126x does indeed break CRC. However, I was not able to find the root cause of this behavior.

I will track this in a separate issue (#1268) - the only workaround I was able to find is to not use the built-in address filtering on SX126x and implement it in your software, by adding and checking the address byte yourself.