jgromes/RadioLib #1274: Getting Error -1106 during LoRaWAN Join on hardware that has already worked

rtamse12, 2024-10-16:
Hello,

i am using RadioLib to communicate via LoRaWAN with two Heltec Wifi Lora 32 (V3). I am using LoRaWAN 1.1.0 with radiolib-persistance. I am basically running the example from radiolib-persistance. This has already functioned perfectly on both devices, though on one of my devices I now always get the Error -1106 (There are no channels available for the requested datarate.). 

I tried clearing all nonces and sessions from my persistance as well as clearing nonces on the network server but the error persists. I can upload exactly the same code while also clearing all sessions to my other Heltec board and it works. Does the SX1262 have some kind of internal state that is messed up, so that it doesn't have any available channels? I tried running radio.reset(), though that doesn't seem to do anything.

Kind regards

HeadBoffin, 2024-10-16:
Please can you complete the issue template so we get the details which are going to be so totally vital to understand the basic how's & why's before we get in to the details. Also, highlight any differences between the configuration of the two devices.

FWIW, the Heltec WiFi LoRa 32 v3 is the primary development & test device - that is, the developer has some to develop on and the tester has one (and a dozen other boards) for testing. So the behaviour you are describing is unusual.

StevenCellist, 2024-10-27:
@rtamse12 any update on this?

StevenCellist, 2024-10-30:
Closing due to inactivity. Feel free to re-open if the problem persists (or open a Discussion).

closed 2024-10-30 by StevenCellist