jgromes/RadioLib #1283: Two devices to one SPI on ESP32 S3, program halted

tonbor, 2024-10-21:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Troubleshooting Guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide) Wiki page and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Describe the bug**
A Two devices to one SPI on ESP32 S3, program halted. Init TFT ili9488 init radiolib CC1101. Same spi, different cs/ss. TFT starts, get output to screen. cc1101 get input, working ok. Then screenoutput to TFT program halted. Just nothing.

. description of the problem and the conditions that cause it to occur
IDE (e.g. Arduino or PlatformIO) Arduino 2.3
TFT_eSPI library version latest, the problem may have been resolved!) from the Manage Libraries... menu
Board package version (e.g. 3.0.5) available from the Boards Manager... menu
Procesor, ESP32 S3 etc
TFT driver (e.g. ILI9488), a link to the vendors product web page is useful too.
Interface type (SPI )
<details><summary>Debug mode output</summary>
<p>

```
paste the debug output here none
```

</p>
</details> 

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please use Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

<details><summary>Sketch that is causing the module fail</summary>
#define USER_SETUP_ID 111
// See SetupX_Template.h for all options available
#define ILI9488_DRIVER
 ****_#define USE_HSPI_PORT // HSPI pins 10-13_****
                    // Typical board default pins - change to match your board
#define TFT_CS   10 //     10 or 34 (FSPI CS0) 
#define TFT_MOSI 11 //     11 or 35 (FSPI D)
#define TFT_SCLK 12 //     12 or 36 (FSPI CLK)
#define TFT_MISO -1 //     13 or 37 (FSPI Q)

// Use pins in range 0-31
#define TFT_DC    14
#define TFT_RST   18

cannot find where RadioLib initialize the SPI, is that the same as  USE_HSPI_PORT or VSPI?????

HeadBoffin, 2024-10-21:
There are two discussions in the last couple of days on more than one item on SPI you may want to check out.

Editing the template is good, but it's useful if you can fill in the debug output and entire sketch please. Plus formatting the code.

jgromes, 2024-10-21:
> cannot find where RadioLib initialize the SPI, is that the same as USE_HSPI_PORT or VSPI?????

By default, RadioLib uses the default `SPI` object, whatever your platform defines that to be. See https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup

tonbor, 2024-10-21:
> There are two discussions in the last couple of days on more than one item on SPI you may want to check out.
> 
> Editing the template is good, but it's useful if you can fill in the debug output and entire sketch please. Plus formatting the code.

Where are the two discussions?

jgromes, 2024-11-03:
Closed due to inactivity.

closed 2024-11-03 by jgromes