jgromes/RadioLib #1284: LoRaWAN: maxPayLen calculation for downlink is incorrect

tve, 2024-10-22:
The calculation of the maxPayLen https://github.com/jgromes/RadioLib/blob/master/src/protocols/LoRaWAN/LoRaWAN.cpp#L1401 is incorrect: it uses the uplink DR instead of the downlink DR. This makes a big difference in the US915 region where, for example, a JOIN uplink at DR0 with max 19 bytes is followed by a JOIN ACCEPT at DR10 with max 250 bytes payload. This was preventing my device from joining 'cause the join accept was 33 bytes long and maxPayLen was set to 32 (19+13).
The correct calculation as far as I can tell is:
```
  uint8_t maxPayLen = this->band->payloadLenMax[dlChannels[window].dr];
```

StevenCellist, 2024-10-22:
Hi @tve, that is indeed a problem, thanks for reporting and finding the actual problem! Just this weekend, I finally got my hands on a 915-gateway which means that I will now be able to verify stuff much easier (instead of going through a remote shell, flashing someone else's device without physical access and with resetting connections). I hope to do a proper testing round this or next week, and will start by fixing this. If you don't mind, I'll leave this issue open until then.

StevenCellist, 2024-10-22:
Had a couple of hours spare tonight, thanks and fixed @tve.

closed 2024-10-22 by StevenCellist

VolkerChristian, 2024-10-25:
I'm very sorry, but this commit introduces a failure at least in EU868. After a view uplinks (the number of uplinks working is not always the same - ranges from four to seven or eight)  _node.sendReceive_ returns with error _-12_.
```
[LoRaWAN] Constructing uplink
[LoRaWAN] Sending: 48.264775,14.441480,263.100000,2.890000
Error in sendReceive(-12)
```
Going back one commit and everything is fine!

Currently, I didn't have enough resources to debug this issue myself, thus I stay at 33ab117.

commit 61494b82 (2024-10-22): [LoRaWAN] Fix #1284 (max length) and channel struct for fixed bands