jgromes/RadioLib #1288: RFM95 / Chirpstack no join acks

tebrown, 2024-10-24:
End user devices cannot join network with sample code.  I tried two different gateways: ChirpStack and RAK7268V2 connecting to TTN.  I see the join acks coming back from the gateway, and it should be transmitting. The devices are about 1m apart. 

Chirpstack Logs:

```
Thu Oct 24 17:23:55 2024 user.info chirpstack-concentratord-sx1302[5175]: Frame received, uplink_id: 951884156, count_us: 1665329509, freq: 903900000, bw: 125000, mod: LoRa, dr: SF10, ftime_received: false, ftime_ns: 0
Thu Oct 24 17:23:55 2024 user.info chirpstack-mqtt-forwarder[6613]: Received uplink frame, uplink_id: 951884156
Thu Oct 24 17:23:55 2024 user.info chirpstack-mqtt-forwarder[6613]: Sending uplink event, uplink_id: 951884156, topic: us915_1/gateway/0016c001f160f3c5/event/up
Thu Oct 24 17:23:55 2024 user.info chirpstack-mqtt-forwarder[6613]: Received message, topic: us915_1/gateway/0016c001f160f3c5/command/down, qos: AtMostOnce
Thu Oct 24 17:23:55 2024 user.info chirpstack-mqtt-forwarder[6613]: Received downlink command, downlink_id: 212132108, topic: us915_1/gateway/0016c001f160f3c5/command/down
Thu Oct 24 17:23:55 2024 user.info chirpstack-mqtt-forwarder[6613]: Sending downlink frame, downlink_id: 212132108
Thu Oct 24 17:23:55 2024 user.info chirpstack-concentratord-sx1302[5175]: Enqueueing timestamped packet, downlink_id: 212132108, counter_us: 1670329509, current_counter_us: 1665572177
Thu Oct 24 17:23:55 2024 user.info chirpstack-mqtt-forwarder[6613]: Received ack, items: ["OK", "IGNORED"], downlink_id: 212132108
Thu Oct 24 17:23:55 2024 user.info chirpstack-mqtt-forwarder[6613]: Sending ack event, downlink_id: 212132108, topic: us915_1/gateway/0016c001f160f3c5/event/ack
Thu Oct 24 17:24:00 2024 user.info chirpstack-concentratord-sx1302[5175]: Scheduled packet for TX, downlink_id: 212132108, count_us: 1670329509, freq: 923300000, bw: 500000, mod: LoRa, dr: SF10
Thu Oct 24 17:24:10 2024 user.info chirpstack-concentratord-sx1302[5175]: Publishing stats event, rx_received: 1, rx_received_ok: 1, tx_received: 1, tx_emitted: 2
```

Device Logs:
```
17:23:54.649 > 
17:23:54.649 > Setup
17:23:54.649 > Wake not caused by deep sleep: 0
17:23:54.649 > Boot count: 1
17:23:54.649 > Initalise the radio
17:23:54.650 > RLB_DBG: 
17:23:54.650 > RadioLib Info
17:23:54.650 > Version:  "7.0.2.0"
17:23:54.650 > Platform: "ESP32"
17:23:54.650 > Compiled: "Oct 24 2024" "09:35:29"
17:23:54.656 > RLB_DBG: M       SX127x
17:23:54.662 > Recalling LoRaWAN nonces & session
17:23:54.664 > RLB_DBG: -1101 at .pio/libdeps/adafruit_feather_esp32s3_reversetft/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:473
17:23:54.664 > Join ('login') to the LoRaWAN Network
17:23:54.665 > RLB_PRO: Setting up fixed channels (subband 2)
17:23:54.665 > RLB_PRO: UL:   8 1 903.900 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.665 > RLB_PRO: UL:   9 1 904.100 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.665 > RLB_PRO: UL:  10 1 904.300 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.665 > RLB_PRO: UL:  11 1 904.500 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.666 > RLB_PRO: UL:  12 1 904.700 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  13 1 904.900 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  14 1 905.100 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  15 1 905.300 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:   8 1 903.900 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:   9 1 904.100 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  10 1 904.300 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  11 1 904.500 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  12 1 904.700 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  13 1 904.900 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  14 1 905.100 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.668 > RLB_PRO: UL:  15 1 905.300 (0 - 3) | DL:   0 0   0.000 (0 - 0)
17:23:54.711 > RLB_PRO: [MAC] 0x03
17:23:54.711 > RLB_PRO: 00000000: 00                                               .                 
17:23:54.711 > RLB_PRO: LinkAdrReq: dataRate = 0, txSteps = 0, nbTrans = 0
17:23:54.711 > RLB_PRO: LinkAdrAns: 07
17:23:54.711 > RLB_PRO: [MAC] 0x04
17:23:54.712 > RLB_PRO: 00000000: 00                                               .                 
17:23:54.712 > RLB_PRO: DutyCycleReq: max duty cycle = 1/2^0
17:23:54.712 > RLB_PRO: [MAC] 0x05
17:23:54.712 > RLB_PRO: 00000000: 08 68 e2 8c                                      .h..              
17:23:54.712 > RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 8, freq = 923.300
17:23:54.714 > RLB_PRO: [MAC] 0x08
17:23:54.714 > RLB_PRO: 00000000: 01                                               .                 
17:23:54.714 > RLB_PRO: RXTimingSetupReq: delay = 1 sec
17:23:54.714 > RLB_PRO: [MAC] 0x09
17:23:54.714 > RLB_PRO: 00000000: 1d                                               .                 
17:23:54.714 > RLB_PRO: [MAC] 0x0c
17:23:54.714 > RLB_PRO: 00000000: 65                                               e                 
17:23:54.714 > RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
17:23:54.714 > RLB_PRO: [MAC] 0x0f
[... 374 lines cut ...]
  }
}

// Helper function to display a byte array
void arrayDump(uint8_t *buffer, uint16_t len) {
  for (uint16_t c = 0; c < len; c++) {
    Serial.printf("%02X", buffer[c]);
  }
  Serial.println();
}

#endif
```

</p>
</details> 

**Expected behavior**
It looks like ChirpStack is transmitting, and I have a good antenna on both devices.    I am betting it's a problem with config or something, but I have been banging my head against this for the last 10 hours and any help would be appreciated.

**Additional info (please complete):**
 - MCU: ESP32S3 with Adafruit RFM95 hat
 - Wireless module type: RFM95
 - Platform IO: 6.1.16
 -  Library version: 7.0.2

StevenCellist, 2024-10-24:
"Good antenna" and "1m apart" are things that should never appear in the same sentence. My devices with a terrible Heltec antenna cannot receive SF11+ messages from a gateway 1m away _without any antenna_, not even a dummy load. So first advice would be to get 10m & a brick wall between devices and gateway, and if possible a terrible antenna / dummy load (don't be like me without antenna).
The rest of the log looks decent to me on first glance, so I'd start with some separation and obstruction.

I would also suggest moving to upstream RadioLib as I've very recently pushed a fix for fixed bands (such as US915) in combination with persistence which you are using - it'll save you some frustration once the join is working.

HeadBoffin, 2024-10-25:
As this isn't a RadioLib issue (RFM95 with about 5 different MCUs talking to Chirpstack & TTS is well exercised for many years), converting to discussion.