jgromes/RadioLib #1290: When using the `radio.startWifiScan` function with the LR1121 module, an exception occurs, returning "failed, code -706".

Llgok, 2024-10-25:
### Control MCU in use:
ESP32S3 chip

jgromes, 2024-10-25:
As far as I can tell, LR1121 does not support WiFi or GNSS scanning. Unlike LR1110 or LR1120, its datasheet does not mention it.

Llgok, 2024-10-25:
> As far as I can tell, LR1121 does not support WiFi or GNSS scanning. Unlike LR1110 or LR1120, its datasheet does not mention it.

Thank you for your response. I reviewed the datasheet and confirmed that the Wi-Fi functionality is indeed not supported. Additionally, I have another question. I am the designer of the LR1121, and due to design constraints, I have insufficient MCU pins. Can I use an external pull-up resistor of 10K ohms combined with a 10µF filter capacitor for the reset pin of the LR1121 in my hardware design? Would this approach negatively impact any of its functions?

jgromes, 2024-10-25:
> Can I use an external pull-up resistor of 10K ohms combined with a 10µF filter capacitor for the reset pin of the LR1121 in my hardware design? Would this approach negatively impact any of its functions?

A good question to ask Semtech, unfortunately I can't help much there. For what it's worth, the LR11xx do have a reboot command, which we *could* use if `RADIOLIB_NC` is provided instead of a reset pin. Though personally I always prefer to use a dedicated reset pin.

closed 2024-10-25 by jgromes