jgromes/RadioLib #1295: LR1110: setRfSwitchTable is problematic

tve, 2024-10-27:
Issue1: in most of the LR11x0 examples setRfSwitchTable is called before `begin()` but it has to be called after. Example: https://github.com/jgromes/RadioLib/blob/master/examples/LR11x0/LR11x0_Transmit_Blocking/LR11x0_Transmit_Blocking.ino#L66

Issue2: The table used for setRfSwitchTable for the LR1110 must have all possible state rows and all 5 columns for all 5 possible pins according to https://github.com/jgromes/RadioLib/blob/master/src/modules/LR11x0/LR11x0.cpp#L1456-L1460

I had something like:

```
static const uint32_t rfswitch_dio_pins[] = {
  // DIO5:LNAEN, DIO6:VRFSW, DIO8:RFSW
  RADIOLIB_LR11X0_DIO5, RADIOLIB_LR11X0_DIO6, RADIOLIB_LR11X0_DIO8,
  RADIOLIB_NC, RADIOLIB_NC
};

static const Module::RfSwitchMode_t rfswitch_table[] = {
  // mode                  DIO5  DIO6  DIO8
  { LR11x0::MODE_STBY,   { LOW,  LOW,  LOW  } },
  { LR11x0::MODE_RX,     { LOW,  HIGH, LOW  } },
  { LR11x0::MODE_TX,     { LOW,  HIGH, HIGH } },
  { LR11x0::MODE_GNSS,   { HIGH, LOW,  LOW  } },
  END_OF_MODE_TABLE,
};
```
and this does not work at all.

(On issue 2 I'm not arguing how it should work except that if it's completely different then the docs for the method should mention it.)

jgromes, 2024-10-27:
> Issue1: in most of the LR11x0 examples setRfSwitchTable is called before begin() but it has to be called after.

You're right, LR11x0 is a bit different from the other radio modules in terms of the RF switch. We are basically hijacking the RF switching interface (which was originally intended to just configure GPIO high/low behavior) to perform a specific SPI transaction, and that obviously has to be called after begin, so I will fix the examples.

> Issue2: The table used for setRfSwitchTable for the LR1110 must have all possible state rows and all 5 columns for all 5 possible pins

Hmmm, that's trickier - there's also the fact that there can be a mix of LR11x0 DIO pins, as well as the host MCU pins in the switch table (for example, the GNSS LNA enable pin on the WM1110). I will try to figure something out here.

tve, 2024-10-27:
> the fact that there can be a mix of LR11x0 DIO pins, as well as the host MCU pins in the switch table (for example, the GNSS LNA enable pin on the WM1110). I will try to figure something out here.

you're right. note that the LR1110 command expects a 5-bit bitmask for each state fix its fixed bit assignments (e.g. bit 0 is always DIO5 even if DIO5 is globally disabled). Right now the loop sets the bit corresponding to the position in `rfswitch_table.values` which is incorrect.

jgromes, 2024-10-27:
I just pushed a couple of commits which should resolve this. It will now be able to distinguish between DIO and GPIO pins, and it should handle gaps in the modes/pins table correctly.

I have tested your particular configuration in gdb (it's a good test case as it contains both a skipped DIO pin, as well as some skipped modes) and it looks like it configures the RF switch correctly.

closed 2024-10-27 by jgromes