jgromes/RadioLib #1296: No communication range on Ebyte E220-900m22s, LLCC68 LoRa module for 868Mhz frequency range

codev123, 2024-10-29:
Thank you sir for resolving my previous issue, currently i am working with LLCC68 LoRa module at 868Mhz frequency, 
when i initialize it like
`LLCC68 radio = new Module(20, 22, 24, 23);  // Initialize LLCC68 radio`
it results in error -2, chip not found
so when i initialise my same llcc68 868Mhz module like
`SX1262 radio = new Module(20, 22, 24, 23);  // Initialize SX1262 radio`
it works correctly, it transmits and receive data without any issue. but somehow it results in very less communication range, even it full power of 22dbm, it gives on 30-50 meters communication range.

so i bought some LLCC68 module of 433Mhz, and tested my code, it works with this initialization
`LLCC68 radio = new Module(20, 22, 24, 23);  // Initialize LLCC68 radio`
and gives commmunication range of 2km at 22dbm power.

what could be issue with LLCC68 868Mhz LoRa module code or initialization and how to resolve it. hank you sir

codev123, 2024-10-29:
****************************update**********************************
sorry sir, it was error on my part, the antenna length was 17.27cm for both 433mhz, 868Mhz, that why 433mHz module was working great i long range, but 868Mhz one did not had any range. i had cut antenna to lamda/4 i.e 8.65cm, not its working ok and have good comunication range.

but i had to initialize my llcc68 868Mhz module as
`SX1262 radio = new Module(20, 22, 24, 23);  // Initialize SX1262 radio`
and in setup
`int state = radio.begin(868.0, 125.0, 9, 5, 0x34, 22, 8, 0, false);  //LLCC68 module` 
but its working.....

jgromes, 2024-10-29:
> but i had to initialize my llcc68 868Mhz module as

You should probably enable debug mode and check whether the module you think is LLCC68 actually is LLCC68. RadioLib checks this by reading the version register. If it's reporting -2 for the LLCC68 constructor, it's probably not an LLCC68.

closed 2024-10-29 by jgromes