jgromes/RadioLib #1297: Unable to send multiple LoRaWAN packets

alistair23, 2024-10-30:
**Describe the bug**
Running using the TockHal on the Apollo3 with a SX126x the first LoRa transmission works, but all subsequent transmits don't work, wither either error -5 (`RADIOLIB_ERR_TX_TIMEOUT`) or -1108 (`RADIOLIB_ERR_UPLINK_UNAVAILABLE`).

I am confident it's not a pin out issue as the first operation works and multiple non-LoRaWAN operations also work

<details><summary>Debug mode output</summary>
<p>

```
[2024-10-30 15:27:37.525] [SX1261] Initialising Radio ... 
[2024-10-30 15:27:37.528] RLB_DBG: 
[2024-10-30 15:27:37.528] RadioLib Info
[2024-10-30 15:27:37.528] Version:  "7.1.0.0"
[2024-10-30 15:27:37.531] Platform: "Generic"
[2024-10-30 15:27:37.534] Compiled: "Oct 30 2024" "15:26:05"
[2024-10-30 15:27:37.547] RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
[2024-10-30 15:27:37.552] RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
[2024-10-30 15:27:37.560] RLB_DBG: 
[2024-10-30 15:27:37.560] RLB_DBG: M    SX126x
[2024-10-30 15:27:37.677] RLB_PRO: Setting up fixed channels (subband 2)
[2024-10-30 15:27:37.680] RLB_PRO: UL:   8 1  (1717986918 - 1082959462) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:37.686] RLB_PRO: UL:   9 1  (0 - 1082959872) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:37.692] RLB_PRO: UL:  10 1  (-1717986918 - 1082960281) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:37.698] RLB_PRO: UL:  11 1  (858993459 - 1082960691) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:37.704] RLB_PRO: UL:  12 1  (-858993459 - 1082961100) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:37.711] RLB_PRO: UL:  13 1  (1717986918 - 1082961510) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:37.717] RLB_PRO: UL:  14 1  (0 - 1082961920) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:37.723] RLB_PRO: UL:  15 1  (-1717986918 - 1082962329) | DL:   0 5  (0 - 0)
[2024-10-30 15:27:38.038] RLB_PRO: [MAC] 0x03
[2024-10-30 15:27:38.038] RLB_PRO: 00000000: 20                                                                 
[2024-10-30 15:27:38.046] RLB_PRO: LinkAdrReq: dataRate = 2, txSteps = 0, nbTrans = 0
[2024-10-30 15:27:38.059] RLB_PRO: LinkAdrAns: 07
[2024-10-30 15:27:38.059] RLB_PRO: [MAC] 0x04
[2024-10-30 15:27:38.062] RLB_PRO: 00000000: 00                                               .                 
[2024-10-30 15:27:38.071] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^0
[2024-10-30 15:27:38.074] RLB_PRO: [MAC] 0x05
[2024-10-30 15:27:38.076] RLB_PRO: 00000000: 08 68 e2 8c                                      .h..              
[2024-10-30 15:27:38.082] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 8, freq = 
[2024-10-30 15:27:38.106] RLB_PRO: [MAC] 0x08
[2024-10-30 15:27:38.106] RLB_PRO: 00000000: 01                                               .                 
[2024-10-30 15:27:38.114] RLB_PRO: RXTimingSetupReq: delay = 1 sec
[2024-10-30 15:27:38.117] RLB_PRO: [MAC] 0x09
[2024-10-30 15:27:38.120] RLB_PRO: 00000000: 1d                                               .                 
[2024-10-30 15:27:38.126] RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
[2024-10-30 15:27:38.134] RLB_PRO: [MAC] 0x0c
[2024-10-30 15:27:38.134] RLB_PRO: 00000000: 65                                               e                 
[2024-10-30 15:27:38.143] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
[2024-10-30 15:27:38.149] RLB_PRO: [MAC] 0x0f
[2024-10-30 15:27:38.149] RLB_PRO: 00000000: fa                                               .                 
[2024-10-30 15:27:38.157] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
[2024-10-30 15:27:38.497] RLB_PRO: 
[2024-10-30 15:27:38.497] RLB_PRO: PHY:  Frequency =  MHz, TX = 858993459 dBm
[2024-10-30 15:27:38.538] RLB_PRO: LoRa: SF = 10, BW =  kHz, CR = 4/335452, IQ: 
[2024-10-30 15:27:38.569] RLB_DBG: Timeout in 1853 ms
[2024-10-30 15:27:38.977] RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
[2024-10-30 15:27:38.981] RLB_PRO: 00000000: 00 00 00 00 00 00 00 00 00 59 8a 06 d0 7e d5 b3  .........Y...~..
[2024-10-30 15:27:38.991] RLB_PRO: 00000010: 70 00 00 fa 00 48 33                             p....H3           
[2024-10-30 15:27:39.020] RLB_PRO: 
[2024-10-30 15:27:39.020] RLB_PRO: PHY:  Frequency =  MHz, TX = -858993459 dBm
[2024-10-30 15:27:39.061] RLB_PRO: LoRa: SF = 10, BW =  kHz, CR = 4/335452, IQ: 
[2024-10-30 15:27:43.967] RLB_PRO: Opening Rx1 window (126 ms timeout)... <-- Rx Delay end 
[2024-10-30 15:27:44.100] RLB_PRO: Closing Rx1 window
[2024-10-30 15:27:44.136] RLB_PRO: JoinAccept (JoinNonce = 175, previously 0):
[2024-10-30 15:27:44.141] RLB_PRO: 00000000: 20 af 00 00 13 00 00 4c 80 0d 26 88 05 00 ff 00   ......L..&.....
[2024-10-30 15:27:44.151] RLB_PRO: 00000010: 00 00 00 00 00 02 00 00 00 00 00 00 01 49 43 c8  .............IC.
[2024-10-30 15:27:44.157] RLB_PRO: 00000020: 9e                                               .                 
[2024-10-30 15:27:44.165] RLB_PRO: LoRaWAN revision: 1.1
[2024-10-30 15:27:44.187] RLB_PRO: [MAC] 0x05
[2024-10-30 15:27:44.187] RLB_PRO: 00000000: 08 68 e2 8c                                      .h..              
[2024-10-30 15:27:44.196] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 8, freq = 
[... 164 lines cut ...]
      hal->delay(60 * 1000);
    }
  }

  return 0;
}
```

</p>
</details> 

**Expected behavior**
I expect the transmission to complete every 60 minutes, with either no errors or something like `-1116`.

I also expect to see all of the traffic in TTN, but I'm currently only seeing the first message.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info (please complete):**
 - MCU: Apollo3
 - Link to Arduino core: N/A
 - Wireless module type: sx1262
 - Arduino IDE version: N/A
 - Library version: 7.2.0

StevenCellist, 2024-10-30:
> Library version: 7.2.0

Faster than light?

Does this problem also occur on 7.0.2 and if yes, on the last 6.x version?

alistair23, 2024-10-30:
Ha!

It happens on 7.0.2, I'll have to test on the 6.x version

StevenCellist, 2024-10-30:
To be very clear: it feels like you think you used 7.0.2 for this issue, but your debug output shows 7.1.0 being the version used. Please confirm if the issue exists on all of 7.1.0 (yes, given the log), and 7.0.2, and the last of 6.x.

alistair23, 2024-10-30:
Same thing on 6.6.0-ish (fffb1fae9f6c43997c7ee0993e68ec7cd80ebb17).

I was origianlly using fffb1fae9f6c43997c7ee0993e68ec7cd80ebb17 as it include some LoRaWAN fixes for some other problems, specifically a custom `scanGuard`.

```
[2024-10-30 21:32:27.915] [SX1261] Initialising Radio ... 
[2024-10-30 21:32:27.918] RLB_DBG: 
[2024-10-30 21:32:27.918] RadioLib Info
[2024-10-30 21:32:27.918] Version:  "6.6.0.0"
[2024-10-30 21:32:27.921] Platform: "Generic"
[2024-10-30 21:32:27.924] Compiled: "Oct 30 2024" "21:31:51"
[2024-10-30 21:32:27.938] RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
[2024-10-30 21:32:27.942] RLB_DBG: 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
[2024-10-30 21:32:27.950] RLB_DBG: 
[2024-10-30 21:32:27.950] RLB_DBG: M    SX126x
[2024-10-30 21:32:28.067] RLB_PRO: Setting up fixed channels (subband 2)
[2024-10-30 21:32:28.070] RLB_PRO: mask[7] = 0x0002
[2024-10-30 21:32:28.072] RLB_PRO: UL:  65 1  (0 - 1082960896) | DL:   6 6  (0 - 0)
[2024-10-30 21:32:28.078] RLB_PRO: mask[0] = 0xff00
[2024-10-30 21:32:28.078] RLB_PRO: UL:  65 1  (0 - 1082960896) | DL:   6 6  (0 - 0)
[2024-10-30 21:32:28.084] RLB_PRO: UL:   8 1  (1610612736 - 1082959462) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.090] RLB_PRO: UL:   9 1  (0 - 1082959872) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.096] RLB_PRO: UL:  10 1  (-1610612736 - 1082960281) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.102] RLB_PRO: UL:  11 1  (1073741824 - 1082960691) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.108] RLB_PRO: UL:  12 1  (-536870912 - 1082961100) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.115] RLB_PRO: UL:  13 1  (1610612736 - 1082961510) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.123] RLB_PRO: UL:  14 1  (0 - 1082961920) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.127] RLB_PRO: UL:  15 1  (-1610612736 - 1082962329) | DL:   0 5  (0 - 0)
[2024-10-30 21:32:28.442] RLB_PRO: [MAC] 0x03
[2024-10-30 21:32:28.442] RLB_PRO: 0000000 20 00 00 80                                     |  ...              
[2024-10-30 21:32:28.450] RLB_PRO: LinkADRReq: dataRate = 2, txSteps = 0, chMask = 0x0000, chMaskCntl = 0, nbTrans = 0
[2024-10-30 21:32:28.462] RLB_PRO: LinkADRAns: status = 0x07
[2024-10-30 21:32:28.465] RLB_PRO: [MAC] 0x04
[2024-10-30 21:32:28.465] RLB_PRO: 0000000 00                                              | .                 
[2024-10-30 21:32:28.473] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^0
[2024-10-30 21:32:28.476] RLB_PRO: [MAC] 0x05
[2024-10-30 21:32:28.479] RLB_PRO: 0000000 08 68 e2 8c                                     | .h..              
[2024-10-30 21:32:28.487] RLB_PRO: RXParamSetupReq: rx1DrOffset = 0, rx2DataRate = 8, freq = 
[2024-10-30 21:32:28.499] RLB_PRO: RXParamSetupAns: status = 0x07
[2024-10-30 21:32:28.502] RLB_PRO: [MAC] 0x08
[2024-10-30 21:32:28.502] RLB_PRO: 0000000 01                                              | .                 
[2024-10-30 21:32:28.510] RLB_PRO: RXTimingSetupReq: delay = 1 sec
[2024-10-30 21:32:28.513] RLB_PRO: [MAC] 0x09
[2024-10-30 21:32:28.516] RLB_PRO: 0000000 1d                                              | .                 
[2024-10-30 21:32:28.525] RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
[2024-10-30 21:32:28.530] RLB_PRO: [MAC] 0x0c
[2024-10-30 21:32:28.530] RLB_PRO: 0000000 65                                              | e                 
[2024-10-30 21:32:28.539] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
[2024-10-30 21:32:28.545] RLB_PRO: [MAC] 0x0f
[2024-10-30 21:32:28.545] RLB_PRO: 0000000 fa                                              | .                 
[2024-10-30 21:32:28.554] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
[2024-10-30 21:32:28.559] RLB_PRO: RejoinParamSetupAns: status = 0x03
[2024-10-30 21:32:28.871] RLB_PRO: 
[2024-10-30 21:32:28.871] RLB_PRO: PHY: Frequency UL =  MHz
[2024-10-30 21:32:28.908] RLB_PRO: PHY: SF = 10, TX = 22 dBm, BW =  kHz, CR = 4/0
[2024-10-30 21:32:28.950] RLB_DBG: Timeout in 1853 ms
[2024-10-30 21:32:29.356] RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
[2024-10-30 21:32:29.359] RLB_PRO: 
[2024-10-30 21:32:29.361] RLB_PRO: PHY: Frequency DL =  MHz
[2024-10-30 21:32:29.398] RLB_PRO: PHY: SF = 10, TX = 22 dBm, BW =  kHz, CR = 4/0
[2024-10-30 21:32:34.344] RLB_PRO: Opening Rx1 window (126 ms timeout)... <-- Rx Delay end 
[2024-10-30 21:32:34.477] RLB_PRO: Closing Rx1 window
[2024-10-30 21:32:34.523] RLB_PRO: JoinAccept (JoinNonce = 177, previously 0):
[2024-10-30 21:32:34.527] RLB_PRO: 0000000 20 b1 00 00 13 00 00 79 95 0d 26 88 05 00 ff 00 |  ......y..&.....
[2024-10-30 21:32:34.537] RLB_PRO: 0000010 00 00 00 00 00 02 00 00 00 00 00 00 01 64 e6 72 | .............d.r
[2024-10-30 21:32:34.543] RLB_PRO: 0000020 e2                                              | .                 
[2024-10-30 21:32:34.552] RLB_PRO: LoRaWAN revision: 1.1
[2024-10-30 21:32:34.572] RLB_PRO: [MAC] 0x05
[2024-10-30 21:32:34.572] RLB_PRO: 0000000 08 68 e2 8c                                     | .h..              
[2024-10-30 21:32:34.581] RLB_PRO: RXParamSetupReq: rx1DrOffset = 0, rx2DataRate = 8, freq = 
[... 146 lines cut ...]
[2024-10-31 03:32:41.190] RLB_PRO: 0000000 11 00 00 00 0c 00 00 00 a2 a2 01 00 fc ff ff ff | ................
[2024-10-31 03:32:41.198] RLB_PRO: 0000010 40 79 95 0d 26 80 01 00 01 38 20 15 ae 74 25 4d | @y..&....8 ..t%M
[... cut ...]
[2024-10-31 04:32:41.257] RLB_PRO: 0000020 97 63 48 34                                     | .cH4              
[2024-10-31 04:32:41.294] RLB_DBG: Timeout in 282 ms
[2024-10-31 04:32:41.608] RLB_PRO: Uplink sent <-- Rx Delay start
[2024-10-31 04:32:41.614] failed, code -5
[2024-10-31 04:32:41.614] Waiting 60 minutes before transmitting again
[2024-10-31 05:32:40.913] [SX1261] Transmitting
[2024-10-31 05:32:40.913] failed, code -1108
[2024-10-31 05:32:40.916] Waiting 60 minutes before transmitting again
[2024-10-31 06:32:40.215] [SX1261] Transmitting
[2024-10-31 06:32:40.215] failed, code -1108
```

alistair23, 2024-10-31:
Same with `7.0.2`

```
[2024-10-31 08:52:41.902] [SX1261] Initialising Radio ... 
[2024-10-31 08:52:41.906] RLB_DBG: 
[2024-10-31 08:52:41.906] RadioLib Info
[2024-10-31 08:52:41.906] Version:  "7.0.2.0"
[2024-10-31 08:52:41.908] Platform: "Generic"
[2024-10-31 08:52:41.911] Compiled: "Oct 31 2024" "08:51:34"
[2024-10-31 08:52:41.925] RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
[2024-10-31 08:52:41.930] RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
[2024-10-31 08:52:41.938] RLB_DBG: 
[2024-10-31 08:52:41.938] RLB_DBG: M    SX126x
[2024-10-31 08:52:42.054] RLB_PRO: Setting up fixed channels (subband 2)
[2024-10-31 08:52:42.057] RLB_PRO: UL:   8 1  (1717986918 - 1082959462) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.063] RLB_PRO: UL:   9 1  (0 - 1082959872) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.069] RLB_PRO: UL:  10 1  (-1717986918 - 1082960281) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.076] RLB_PRO: UL:  11 1  (858993459 - 1082960691) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.082] RLB_PRO: UL:  12 1  (-858993459 - 1082961100) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.088] RLB_PRO: UL:  13 1  (1717986918 - 1082961510) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.094] RLB_PRO: UL:  14 1  (0 - 1082961920) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.100] RLB_PRO: UL:  15 1  (-1717986918 - 1082962329) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.106] RLB_PRO: UL:   8 1  (1717986918 - 1082959462) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.112] RLB_PRO: UL:   9 1  (0 - 1082959872) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.118] RLB_PRO: UL:  10 1  (-1717986918 - 1082960281) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.125] RLB_PRO: UL:  11 1  (858993459 - 1082960691) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.131] RLB_PRO: UL:  12 1  (-858993459 - 1082961100) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.137] RLB_PRO: UL:  13 1  (1717986918 - 1082961510) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.143] RLB_PRO: UL:  14 1  (0 - 1082961920) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.149] RLB_PRO: UL:  15 1  (-1717986918 - 1082962329) | DL:   0 5  (0 - 0)
[2024-10-31 08:52:42.464] RLB_PRO: [MAC] 0x03
[2024-10-31 08:52:42.464] RLB_PRO: 00000000: 20                                                                 
[2024-10-31 08:52:42.472] RLB_PRO: LinkAdrReq: dataRate = 2, txSteps = 0, nbTrans = 0
[2024-10-31 08:52:42.483] RLB_PRO: LinkAdrAns: 07
[2024-10-31 08:52:42.483] RLB_PRO: [MAC] 0x04
[2024-10-31 08:52:42.486] RLB_PRO: 00000000: 00                                               .                 
[2024-10-31 08:52:42.495] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^0
[2024-10-31 08:52:42.498] RLB_PRO: [MAC] 0x05
[2024-10-31 08:52:42.501] RLB_PRO: 00000000: 08 68 e2 8c                                      .h..              
[2024-10-31 08:52:42.507] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 8, freq = 
[2024-10-31 08:52:42.528] RLB_PRO: [MAC] 0x08
[2024-10-31 08:52:42.528] RLB_PRO: 00000000: 01                                               .                 
[2024-10-31 08:52:42.536] RLB_PRO: RXTimingSetupReq: delay = 1 sec
[2024-10-31 08:52:42.539] RLB_PRO: [MAC] 0x09
[2024-10-31 08:52:42.542] RLB_PRO: 00000000: 1d                                               .                 
[2024-10-31 08:52:42.548] RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
[2024-10-31 08:52:42.557] RLB_PRO: [MAC] 0x0c
[2024-10-31 08:52:42.557] RLB_PRO: 00000000: 65                                               e                 
[2024-10-31 08:52:42.565] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
[2024-10-31 08:52:42.571] RLB_PRO: [MAC] 0x0f
[2024-10-31 08:52:42.571] RLB_PRO: 00000000: fa                                               .                 
[2024-10-31 08:52:42.579] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
[2024-10-31 08:52:42.917] RLB_PRO: 
[2024-10-31 08:52:42.917] RLB_PRO: PHY:  Frequency UL =  MHz
[2024-10-31 08:52:42.957] RLB_PRO: LoRa: SF = 10, TX = 22 dBm, BW =  kHz, CR = 4/0
[2024-10-31 08:52:42.987] RLB_DBG: Timeout in 1853 ms
[2024-10-31 08:52:43.393] RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
[2024-10-31 08:52:43.397] RLB_PRO: 00000000: 00 00 00 00 00 00 00 00 00 59 8a 06 d0 7e d5 b3  .........Y...~..
[2024-10-31 08:52:43.407] RLB_PRO: 00000010: 70 00 00 fa 00 48 33                             p....H3           
[2024-10-31 08:52:43.431] RLB_PRO: 
[2024-10-31 08:52:43.431] RLB_PRO: PHY:  Frequency DL =  MHz
[2024-10-31 08:52:43.471] RLB_PRO: LoRa: SF = 10, TX = 22 dBm, BW =  kHz, CR = 4/0
[2024-10-31 08:52:48.381] RLB_PRO: Opening Rx1 window (126 ms timeout)... <-- Rx Delay end 
[2024-10-31 08:52:48.514] RLB_PRO: Closing Rx1 window
[2024-10-31 08:52:48.550] RLB_PRO: JoinAccept (JoinNonce = 179, previously 0):
[2024-10-31 08:52:48.555] RLB_PRO: 00000000: 20 b3 00 00 13 00 00 23 f6 0d 26 88 05 00 ff 00   ......#..&.....
[2024-10-31 08:52:48.565] RLB_PRO: 00000010: 00 00 00 00 00 02 00 00 00 00 00 00 01 77 46 3c  .............wF<
[2024-10-31 08:52:48.571] RLB_PRO: 00000020: cf                                               .                 
[2024-10-31 08:52:48.579] RLB_PRO: LoRaWAN revision: 1.1
[2024-10-31 08:52:48.601] RLB_PRO: [MAC] 0x05
[... 65 lines cut ...]
[2024-10-31 09:52:54.525] RLB_PRO: Uplink (FCntUp = 1) decoded:
[2024-10-31 09:52:54.529] RLB_PRO: 00000000: 49 00 00 00 00 01 23 f6 0d 26 00 00 00 00 00 0f  I.....#..&......
[... cut ...]
[2024-10-31 11:02:51.894] RLB_PRO: Uplink (FCntUp = 1) decoded:
[2024-10-31 11:02:51.899] RLB_PRO: 00000000: 49 00 00 00 00 00 23 f6 0d 26 01 00 00 00 00 12  I.....#..&......
[2024-10-31 11:02:51.906] RLB_PRO: 00000010: 40 23 f6 0d 26 82 01 00 db aa 01 05 c8 7c 09 b1  @#..&........|..
[2024-10-31 11:02:51.915] RLB_PRO: 00000020: ac 13 10 e1 2f b6                                ..../.            
[2024-10-31 11:02:51.954] failed, code -1108
[2024-10-31 11:02:51.954] Waiting 60 minutes before transmitting again
```

StevenCellist, 2024-10-31:
Given that this problem persists across all three versions, and 6.6.0 and 7.0.2 haven't been reported by anyone else to result in this behaviour, I'm very much tempted to say this is a Tock issue. My primary suspect would be your HAL, more specifically the clock-stuff, as I see this line: 
```
[2024-10-31 09:52:54.660] RLB_PRO: Delaying transmission by 597393 ms
[2024-10-31 10:02:51.946] RLB_DBG: Timeout in 282 ms
```
Maybe overflow of values? If that is the case, we might get away with a different way of handling clock values within the LoRaWAN stack, but before that, you'll need to investigate the cause of these weird values.

alistair23, 2024-10-31:
I suspect you are right about it being a timing issue. It seems like the TX window for the subsequent transmits are too low.

What is a usual value of `Delaying transmission`?

It looks like the `getTimeOnAir()` values could be too low

For the working case in `getTimeOnAir()` I am seeing values like this

```
spreadingFactor: 10
symbolLength_us: 8192
N_symbol_header: 20
```

while the second attempt is

```
spreadingFactor: 7
symbolLength_us: 1024
N_symbol_header: 20
```

which results in a much shorter window

StevenCellist, 2024-10-31:
Let's take RadioLib 7.1 as the go-to version as that is the currently active one.

Transmission delays should not occur under normal circumstances - it can be used by `scheduleTransmission()` which pretty much no-one should use. The reason why that text would occur is that `this->tUplink` is apparently larger than `tNow`:

https://github.com/jgromes/RadioLib/blob/4564d87721997917c567f75c2078106ee92e757a/src/protocols/LoRaWAN/LoRaWAN.cpp#L925-L931

This would suggest to me that somehow `tNow` has dropped to a lower value than `tUplink` because of an overflow or something along those lines. So please investigate that before going into Time-on-Air stuff - RadioLib does a clean 5x ToA timeout which has never been a problem.

Also worth investigating is the use of interrupts - notably the STM32Cube I've been working on doesn't handle that very well. IRQs are weird.

alistair23, 2024-11-01:
The `Delaying transmission` is occurring because the timer is wrapping. I can look at fixing that, but from what I can tell that shouldn't cause issues as the delay occurs before the transmit starts.

As for interrupts, they seem to be working, but are only used for the receive, but I don't get that far as the transmit times out

StevenCellist, 2024-11-01:
I can tell you, LoRaWAN relies on the timer much more often than just for the transmission delay... pretty much each, every and any action relies on the internal clock / timer. So far, it looks like a `uint32_t`-sized clock runs perfectly fine (at least, there have not been reports of problems), but smaller-sized clocks that roll over very fast (looking at you, CubeCell) are highly likely to result in all sorts of weird stuff including what you see here. 

As this appears to be an issue with the Tock platform instead of RadioLib, and there are (as far as I'm aware) only very few Tock users around here, I'll revert this issue to a discussion. I'm happy to provide pointers there to help you, but it's not something we can fix here, really.