jgromes/RadioLib #1299: Issue with Lorawan, I can send JoinRequest and receive JoinAccept. But when I transmit a message it is not being received by the Helium network.

predictitai, 2024-10-31:
After a JoinRequest and JoinAccept I transmit "hello world". I see that via an rtl sdr that the message has actually been send; however I don't see the uplink message frame in my Helium-IoT Console. Note that transmit and sendReceive function do not result in an uplink frame.
scanguard=50 and added enabled debugging; .

<details><summary>Debug mode output</summary>
<p>

Setup ... 
Initialise the radio
RLB_DBG: 
RadioLib Info
Version:  "7.1.0.0"
Platform: "ESP32"
Compiled: "Oct 31 2024" "13:29:29"
RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
RLB_DBG: 
RLB_DBG: M      SX126x
Join ('login') the LoRaWAN Network
RLB_PRO: Setting up dynamic channels
RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
RLB_PRO: [MAC] 0x03
RLB_PRO: 00000000: 20                                                                 
RLB_PRO: LinkAdrReq: dataRate = 2, txSteps = 0, nbTrans = 0
RLB_PRO: LinkAdrAns: 07
RLB_PRO: [MAC] 0x04
RLB_PRO: 00000000: 07                                               .                 
RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
RLB_PRO: [MAC] 0x05
RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
RLB_PRO: [MAC] 0x08
RLB_PRO: 00000000: 01                                               .                 
RLB_PRO: RXTimingSetupReq: delay = 1 sec
RLB_PRO: [MAC] 0x09
RLB_PRO: 00000000: 05                                               .                 
RLB_PRO: [MAC] 0x0c
RLB_PRO: 00000000: 65                                               e                 
RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
RLB_PRO: [MAC] 0x0f
RLB_PRO: 00000000: fa                                               .                 
RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
RLB_PRO: 
RLB_PRO: PHY:  Frequency = 868.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: U
RLB_DBG: Timeout in 1853 ms
RLB_PRO: JoinRequest sent (DevNonce = 39783) <-- Rx Delay start
RLB_PRO: 00000000: 00 64 bf 97 76 8d 2a 9b b0 e7 16 5d a4 c2 ba e7  .d..v.*....]....
RLB_PRO: 00000010: ba 67 9b ba 8b 29 78                             .g...)x           
RLB_PRO: 
RLB_PRO: PHY:  Frequency = 868.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx1 window (331 ms timeout)... <-- Rx Delay end 
RLB_PRO: Closing Rx1 window
RLB_PRO: JoinAccept (JoinNonce = 49, previously 0):
RLB_PRO: 00000000: 20 31 00 00 24 00 00 1b 08 00 48 80 01 18 4f 84   1..$.....H...O.
RLB_PRO: 00000010: e8 56 84 b8 5e 84 88 66 84 58 6e 84 00 93 84 d2  .V..^..f.Xn.....
RLB_PRO: 00000020: 1c                                               .                 
RLB_PRO: LoRaWAN revision: 1.1
RLB_PRO: Setting up dynamic channels
RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
RLB_PRO: [MAC] 0x05
RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
RLB_PRO: [MAC] 0x08
RLB_PRO: 00000000: 01                                               .                 
RLB_PRO: RXTimingSetupReq: delay = 1 sec
[... 131 lines cut ...]
    if (state > 0) {
        Serial.println(F("Received a downlink"));
    } else {
        Serial.println(F("No downlink received"));
    }

    Serial.print(F("Next uplink in "));
    Serial.print(uplinkIntervalSeconds);
    Serial.println(F(" seconds\n"));

    // Wait until next uplink - observing legal & TTN FUP constraints
    delay(uplinkIntervalSeconds * 1000UL);  // delay needs milli-seconds
}

</p>
</details> 

**Expected behavior**
I expect an uplink message in Helium-IoT console.

**Additional info (please complete):**
 - MCU: esp32
 - Wireless module type SX1262
 - Arduino IDE version  I use visual studio code with Platform IO
 - Library version 7.0.2

predictitai, 2024-10-31:
Thank you for your quick response! Just to be sure I will send you the complete log; joinrequest and joinaccept i see in the log. But no uplink message. When i use my frequency analyzer i notice that the message actually been send by the device. I get "No downlink received", which probably makes sense as the message has not been received by Helium IOT.
Hope you can advice me!

<begin log>
Initialise the radio
RLB_DBG: 
RadioLib Info
Version:  "7.1.0.0"
Platform: "ESP32"
Compiled: "Oct 31 2024" "13:29:29"
RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
RLB_DBG: 
RLB_DBG: M      SX126x
Join ('login') the LoRaWAN Network
RLB_PRO: Setting up dynamic channels
RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
RLB_PRO: [MAC] 0x03
RLB_PRO: 00000000: 20                                                                 
RLB_PRO: LinkAdrReq: dataRate = 2, txSteps = 0, nbTrans = 0
RLB_PRO: LinkAdrAns: 07
RLB_PRO: [MAC] 0x04
RLB_PRO: 00000000: 07                                               .                 
RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
RLB_PRO: [MAC] 0x05
RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
RLB_PRO: [MAC] 0x08
RLB_PRO: 00000000: 01                                               .                 
RLB_PRO: RXTimingSetupReq: delay = 1 sec
RLB_PRO: [MAC] 0x09
RLB_PRO: 00000000: 05                                               .                 
RLB_PRO: [MAC] 0x0c
RLB_PRO: 00000000: 65                                               e                 
RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
RLB_PRO: [MAC] 0x0f
RLB_PRO: 00000000: fa                                               .                 
RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
RLB_PRO: 
RLB_PRO: PHY:  Frequency = 868.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: U
RLB_DBG: Timeout in 1853 ms
RLB_PRO: JoinRequest sent (DevNonce = 17386) <-- Rx Delay start
RLB_PRO: 00000000: 00 64 bf 97 76 8d 2a 9b b0 e7 16 5d a4 c2 ba e7  .d..v.*....]....
RLB_PRO: 00000010: ba ea 43 44 51 a8 e3                             ..CDQ..           
RLB_PRO: 
RLB_PRO: PHY:  Frequency = 868.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx1 window (331 ms timeout)... <-- Rx Delay end 
RLB_PRO: Closing Rx1 window
RLB_PRO: JoinAccept (JoinNonce = 50, previously 0):
RLB_PRO: 00000000: 20 32 00 00 24 00 00 1d 08 00 48 80 01 18 4f 84   2..$.....H...O.
RLB_PRO: 00000010: e8 56 84 b8 5e 84 88 66 84 58 6e 84 00 b5 21 2a  .V..^..f.Xn...!*
RLB_PRO: 00000020: bb                                               .                 
RLB_PRO: LoRaWAN revision: 1.1
RLB_PRO: Setting up dynamic channels
RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
RLB_PRO: [MAC] 0x05
RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
RLB_PRO: [MAC] 0x08
RLB_PRO: 00000000: 01                                               .                 
RLB_PRO: RXTimingSetupReq: delay = 1 sec
RLB_PRO: Processing CFList
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 03 18 4f 84 50                                   ..O.P             
RLB_PRO: UL:   3 1 867.100 (0 - 5) | DL:   3 1 867.100 (0 - 5)
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 04 e8 56 84 50                                   ..V.P             
RLB_PRO: UL:   4 1 867.300 (0 - 5) | DL:   4 1 867.300 (0 - 5)
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 05 b8 5e 84 50                                   ..^.P             
RLB_PRO: UL:   5 1 867.500 (0 - 5) | DL:   5 1 867.500 (0 - 5)
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 06 88 66 84 50                                   ..f.P             
RLB_PRO: UL:   6 1 867.700 (0 - 5) | DL:   6 1 867.700 (0 - 5)
RLB_PRO: [MAC] 0x07
RLB_PRO: 00000000: 07 58 6e 84 50                                   .Xn.P             
RLB_PRO: UL:   7 1 867.900 (0 - 5) | DL:   7 1 867.900 (0 - 5)
Connected to the network!
Sending uplink
RLB_PRO: Uplink MAC payload:
RLB_PRO: 00000000: 0b 01                                            ..                
RLB_PRO: Uplink (FCntUp = 0) decoded:
RLB_PRO: 00000000: 1d 08 e5 00 00 78 56 ad 4c c3 ce 3f 17 00 00 00  .....xV.L..?....
RLB_PRO: 00000010: 40 1d 08 00 48 82 00 00 3b ff 01 35 22 55 56 ad  @...H...;..5"UV.
RLB_PRO: 00000020: 4c c3                                            L.                
RLB_PRO: 
RLB_PRO: PHY:  Frequency = 867.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: U
RLB_DBG: Timeout in 1648 ms
RLB_PRO: Uplink sent <-- Rx Delay start
RLB_PRO: 
RLB_PRO: PHY:  Frequency = 867.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx1 window (331 ms timeout)... <-- Rx Delay end 
RLB_PRO: Closing Rx1 window
RLB_PRO: 
RLB_PRO: PHY:  Frequency = 869.525 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 12, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx2 window (788 ms timeout)... <-- Rx Delay end 
RLB_PRO: Closing Rx2 window
No downlink received
Next uplink in 300 seconds 
<end log>

predictitai, 2024-10-31:
sorry it should be kept open!

predictitai, 2024-10-31:
Further packets i see the FCntUp increasing; but in the console the number of uplinks remains 0:

### Additional LoRaWAN Debug Log

```plaintext
Sending uplink
RLB_PRO: Uplink (FCntUp = 1) decoded:
RLB_PRO: 00000000: 1d 08 e5 00 00 78 56 ad 4c c3 ce 3f 17 00 00 00  .....xV.L..?....
RLB_PRO: 00000010: 40 1d 08 00 48 80 01 00 01 53 43 68 21 78 56 ad  @...H....SCh!xV.

RLB_PRO: PHY: Frequency = 868.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: U
RLB_DBG: Timeout in 1648 ms

RLB_PRO: Uplink sent <-- Rx Delay start

RLB_PRO: PHY: Frequency = 868.100 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx1 window (331 ms timeout)... <-- Rx Delay end
RLB_PRO: Closing Rx1 window

RLB_PRO: PHY: Frequency = 869.525 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 12, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx2 window (788 ms timeout)... <-- Rx Delay end
RLB_PRO: Closing Rx2 window

No downlink received
Next uplink in 300 seconds

Sending uplink
RLB_PRO: Uplink (FCntUp = 2) decoded:
RLB_PRO: 00000000: 1d 08 e5 00 00 78 56 ad 4c c3 ce 3f 17 00 00 00  .....xV.L..?....
RLB_PRO: 00000010: 40 1d 08 00 48 80 02 00 01 16 36 3d 21 78 56 ad  @...H.....6=!xV.

RLB_PRO: PHY: Frequency = 867.300 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: U
RLB_DBG: Timeout in 1648 ms

RLB_PRO: Uplink sent <-- Rx Delay start

RLB_PRO: PHY: Frequency = 867.300 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx1 window (331 ms timeout)... <-- Rx Delay end
RLB_PRO: Closing Rx1 window

RLB_PRO: PHY: Frequency = 869.525 MHz, TX = 16 dBm
RLB_PRO: LoRa: SF = 12, BW = 125.0 kHz, CR = 4/5, IQ: D
RLB_PRO: Opening Rx2 window (788 ms timeout)... <-- Rx Delay end
RLB_PRO: Closing Rx2 window

No downlink received
Next uplink in 300 seconds

predictitai, 2024-11-01:
i have 196 data credits but I am not sure whether that is sufficient for uplinks. I wrote an email to Helium IOT about this issue. Thanks and i will keep you informed.

disk91, 2024-11-01:
A much I can see (on the operator side) , the device uplink is not received and does not generate error. The credits are OK, the join request seems to be a  success according to the log. Can you see in the log the devaddr used and the nework key to verify it ?

predictitai, 2024-11-01:
For security reasons i am going to change dev id, app key and nwkkey later but just for your information:
I have set this in config.h
// joinEUI - previous versions of LoRaWAN called this AppEUI
// for development purposes you can use all zeros - see wiki for details
#define RADIOLIB_LORAWAN_JOIN_EUI  0xb09b2a8d7697bf64

// the Device EUI & two keys can be generated on the TTN console
#ifndef RADIOLIB_LORAWAN_DEV_EUI   // Replace with your Device EUI
[image]

#define RADIOLIB_LORAWAN_DEV_EUI   0xbae7bac2a45d16e7
#endif
#ifndef RADIOLIB_LORAWAN_APP_KEY   // Replace with your App Key 
#define RADIOLIB_LORAWAN_APP_KEY  0x09, 0x70, 0x2B, 0x09, 0x40, 0x15, 0x03, 0xA0, 0xB2, 0x07, 0x50, 0x79, 0x05, 0x70, 0x5A, 0x07
#endif
#ifndef RADIOLIB_LORAWAN_NWK_KEY   // Put your Nwk Key here
#define RADIOLIB_LORAWAN_NWK_KEY   0x0C, 0x70, 0x37, 0x04, 0x40, 0x92, 0x0B, 0x70, 0x0A, 0x05, 0x00, 0x47, 0x08, 0xD0, 0x99, 0x00
#endif

[image]

In the attachment you can see a screendump of the settings from Helium console. And yes I also asked my friend chatgpt to double confirm the settings are correct.

disk91, 2024-11-01:
Your deveui / appeui / appkey is ok, or you won't get a join accept
This is why I'm asking to control the devaddr and the nwkSkey used by the device to make sure it corresponds to what you see in chirpsatck in the Ativation Session.

Note: I can see you detail myself in the console so don't need to copy here. But I can't see if this is matching the device one.

StevenCellist, 2024-11-01:
According to the logs above, the DevAddr of that session is (was) 0x4800081D. The NwkSEncKey is not available from the logs.

StevenCellist, 2024-11-01:
@predictitai it might be useful if you can provide logs with millisecond-timestamps enabled in your terminal. That way we can see if the timing of the internals is as we'd expect. I do not see a real reason why it would be off, but who knows.

predictitai, 2024-11-01:
@StevenCellist I added the following lines in my code:   RADIOLIB_DEBUG_PROTOCOL_PRINTLN("devAddr: %d", this->devAddr );
  RADIOLIB_DEBUG_PROTOCOL_PRINTLN("appSKey: %d", this->appSKey );
  RADIOLIB_DEBUG_PROTOCOL_PRINTLN("nwkSEncKey: %d", this->nwkSEncKey );
  RADIOLIB_DEBUG_PROTOCOL_PRINTLN("fNwkSIntKey: %d", this->fNwkSIntKey );
  RADIOLIB_DEBUG_PROTOCOL_PRINTLN("sNwkSIntKey: %d", this->sNwkSIntKey ); 

Just to be certain that the session data matches. I am currently at a location without helium reach so I will have to do the actual tests tonight. Thanks for your support so far!

predictitai, 2024-11-02:
@StevenCellist so with your code you provided this is the log i got:

RLB_PRO: DevAddr: 4800096a
RLB_PRO: AppSKey, NwkSEncKey, FNwkSIntKey, SNwkSIntKey:
RLB_PRO: 00000000: 4d d1 c7 ec 79 be 6d d3 90 ba e2 b7 bc 4b 2f 20  M...y.m......K/ 
RLB_PRO: 00000000: d0 68 56 7a 5d 6f 46 2f 57 57 8a ed 31 51 08 38  .hVz]oF/WW..1Q.8
RLB_PRO: 00000000: 72 2e 7e af 17 cf 5a 00 a7 53 96 c6 0b c6 24 99  r.~...Z..S....$.
RLB_PRO: 00000000: a9 c0 1a 25 14 42 f3 97 44 90 7b 63 0f 50 7b f0  ...%.B..D.{c.P{.

And so if i go to helium console, activation tab i can confirm I see exactly the same data here!

StevenCellist, 2024-11-02:
Not much of a surprise, but confirmation is good. Next useful thing is to enable timestamps in your terminal and have your device do a join + one or two uplinks and paste the output here. That way we can verify whether the timing also makes sense.

predictitai, 2024-11-03:
@StevenCellist So i kind of fixed the problem by accident. I used Lorawan 1.1.0 but if i switch back to 1.0.4 i do get up- and downlinks. Confusing part is that in the documentation it is recommended to set it to 1.1.0 https://github.com/jgromes/RadioLib/wiki/LoRaWAN:-Device-setup-on-TTN.
So i am not sure if it was intended or a glitch but 1.1.0 doesn't seem to be working yet....

StevenCellist, 2024-11-03:
Interesting, so it looks like the new Helium server still is unable to handle LoRaWAN 1.1.
@disk91 can you brief us on this?

disk91, 2024-11-03:
You need to setup chirpstack device profile and device to the same version. I don't know what was your previous test versions. It's good to know a such difference can bring no error message and no messages received.

Can you please details what was the device + device profile version setup when not working and what is once working please ?

predictitai, 2024-11-03:
I got it to work with lorawan 1.0.4 revision B and then it works (though the nwkkey seem to  have disappeared).
Previously i setup lorawan 1.1.0 revision A (including the nwkkey) but no uplink/downlink messages.
I use the T-Deck plus (which is esp32) and noticed that in Radiolib you can set the version to 1.0 or 1.1 (uint8_t rev = 0;) but you cannot set the revision anywhere on the device.

I tried all combinations of 1.1.0 and all revisions but none seems to generate an uplink or downlink (or error for that matter). The only version that works find for me is 1.0.4 revision B.

Please let me know if you need any additional information or further testing.

StevenCellist, 2024-11-03:
> I noticed that in Radiolib you can set the version to 1.0 or 1.1 (uint8_t rev = 0;)

Did you really modify this yourself?

predictitai, 2024-11-03:
@StevenCellist Yes I see now that it is supposed to be set by the network. So this basically means that you only have to set version and revision in Helium settings and that you don't have to do anything on the device am i correct?

StevenCellist, 2024-11-03:
That is correct. As you noticed, v1.1 has a NwkKey while 1.0.4 doesn't. So that, along with some magic from the JoinAccept, is enough information for RadioLib to figure out what version and encryption it should use.

So, while you re-test 1.1.0 without any modifications to RadioLib (although maybe keep the key dumps in there) and provide us your NwkKey alongside, please tell us why you thought you'd had to modify the internals - it's useful information so we can hopefully prevent others from doing so.

predictitai, 2024-11-04:
@StevenCellist I updated the code (see previous mail); i received a downlink at my location here. If you need any further info please let me know.

StevenCellist, 2024-11-04:
Given the poor reach of your device apparently, as it often fails to join, have you considered the possibility that none of the uplinks is heard due to signal strength? Do you have access to a Helium gateway yourself so you can check what happens on the gateway?

Also, there's an oddity that occurs when you don't receive the JoinAccept, as it looks like a DIO is not fired correctly / at the right time. Please double-check your radio wiring. And what happens when you don't modify the scanGuard?

As I don't have any Helium credits, don't aspire to buy any, and I also don't know if I even have Helium coverage here, there is no possibility for me to replicate this with the Helium server. The best action you can take is to try and have your device join The Things Network (now known as Stack Sandbox instead of Network).

disk91, 2024-11-06:
@StevenCellist As much as I've seen the signal is good for transmitting the Join Request, I can't say if the join accept is received. We can easily investigate the Join Request signal if required. (sometime, problem on join accept is also to to signal streingh too high)

If you want to do some trial on helium-iot you have 500 credit for free, and I'll be happy to credit you more for this kind of debugging purpose, just let me know with a DM / Service request on the platform.

StevenCellist, 2024-11-06:
@disk91 where can I check if there's supposed to be coverage around for this helium-iot network? Honestly no clue which website to look at as there are multiple now, it appears.

disk91, 2024-11-07:
have a look at mapper : https://mappers.helium.com/
if the area is not mapped you can have a look at https://explorer.moken.io to see where the hotspot are

predictitai, 2024-11-07:
@StevenCellist I just tested it with the standard scanguard of 10 which also seem to be working fine so i set it back to its default value. I actually wanted to buy a gateway but for some reason they are quite expensive and i will only do that once i am 100 procent confident helium is suitable for the applications i have in mind. 

As for the DIO issue i forwarded this to Lilygo as i don't see anything that could cause any issues as i am using basic examples.

At my home my helium coverage but around the city i have good network. So i am pretty confident it has nothing to do with the coverage cause where ever i go i don't see any uplinks and/or downlinks.

I will test it out on the TTN platform this evening.

disk91, 2024-11-07:
@predictitai you can use a standard gateway with helium, connecting it to a gateway-rs service or enabling local gateway-rs is you select a helium data only hardware, usually at the same price of a classical gateway. Look at Seed and Rak light gateway if you want to mine at a low cost or data only gateway if you don't want no mine.

StevenCellist, 2024-12-09:
@predictitai @disk91 sorry for the long delay, but this appears to be reproducible. I do not have a Helium gateway on my desk but appear to be just enough in range of one to get somewhat reliable results. As RSSI -130 / SNR -15 at SF12 is stretching it significantly though, I'd like to have a gateway on my desk to check the logs. I definitely do not have the time to play around, so if either one of you can point me at a blatantly obvious guide to use a Semtech UDP gateway on Helium (no mining), I will try and see if I can find where it goes wrong.

StevenCellist, 2024-12-09:
First attempt using 1.0.4 sees uplink come through, so that narrows stuff down significantly. I myself will not (be able to) investigate any further until I have a gateway on my desk.

StevenCellist, 2024-12-10:
@clavisound this is completely beside what I want to achieve - I am not buying anything, I will not create a wallet, I will not do any mining. That is why I said I want to use a Semtech UDP gateway without mining. If that is not possible to achieve (although it looks like an earlier answer by @disk91 might be useful), I will not investigate this any further.

jthiller, 2024-12-14:
It is possible to provision a gateway on Helium without tokens/wallets/etc. The tokens are only used to create an on-chain representation of the gateway, but are not required for data transfer (the network won't know the location of the gateway or issue rewards to the gateway).

The simple architecture is [semtech packet forwarder] -> [helium gateway client]. The gateway client can be run locally on the gateway, (pointing to 127.0.0.1) or remotely. verbose guide here, just ignore the onboarding steps: https://docs.helium.com/iot/data-only-hotspots-onboarding

Overall, this is perhaps a bit off-topic for this thread, so please feel free to email me if you'd like any further tips. joey at helium.foundation. I can also connect you with some non-crypto hardware if it'd be interesting for you to experiment with.

disk91, 2024-12-15:
> @clavisound this is completely beside what I want to achieve - I am not buying anything, I will not create a wallet, I will not do any mining. That is why I said I want to use a Semtech UDP gateway [...]

Sorry, long to answer, If you want to setup your desk gateway for a test, you can use one of my spare data-only gateway. just give me a location (no need to be exact but need to know the region) and I'll give you an IP and port for the semtech protocol.

Just need a way to establish a direct contact to exchange these information, you can find me on twitter, bluesky, discord or use the contact form on my blog disk91.com.

StevenCellist, 2024-12-15:
@disk91 thank you for your offer! I sent you a friend request on Discord. I also got some additional information from @jthiller, so it looks like I'll be able to get going with the help of the both of you :)

StevenCellist, 2024-12-16:
Helium data-only gateway `real-navy-tortoise` reporting for duty. Thanks for the instructions @jthiller and credits @disk91!
Using the all-mighty powers from @disk91, we have now confirmed that the problem lies in the Helium Packet Router, as it does not handle LoRaWAN v1.1, because the MIC encryption keys are different. Investigation is ongoing - will keep it open for a bit for further updates, or close if the Helium update cycle is going to take a while. 

Resolved from RadioLib's perspective.

StevenCellist, 2024-12-17:
The problem has been recognised and is being worked on. Given the time of the year, chances are high it's going to take a bit before Helium Packet Router is actually updated, but rest assured that it will be fixed. For now, simply use LoRaWAN v1.0.4 (for which see the readme in examples/LoRaWAN). Closing.

closed 2024-12-17 by StevenCellist

predictitai, 2025-05-01:
Hey hello @StevenCellist,

This ticket has closed as I can use Version:  "7.1.0.0" with lorawan 1.0.4 just fine. After downloading the latest version of Radiolib; i noticed that I don't receive any uplinks in Helium console at all anymore with any lorawan version. I confirmed that (just in previous code) network key, application session key and device id are ok which they are; I also receive join accept although less reliable than in the older Radiolib version. It appears that helium just drops the payload. So even though everything works fine for me as I reverted back to the old radiolib I wanted you to inform that. Do you need any further information for me to test or provide? Kind regards, Anouar Manders

StevenCellist, 2025-05-01:
Hi Anouar, you could try and inspect the gateway logs if available to you. Then you could verify whether the uplinks are actually received by the gateway. If that is the case, the issue would likely be somewhere up the Helium ladder.
Regarding the reliability of the JoinAccept: we've been tuning the accuracy of the Rx windows the last few months. You could try downloading the current code from GitHub, which may be more accurate than 7.1.2 is now. Otherwise any logs from device and gateway could help.

Note that Helium is used much less than other network servers and as their support at times is lacking, we may not be able to go too deep down the rabbit hole. But a quick gander wouldn't hurt.

predictitai, 2025-05-13:
Thanks for your reply and suggestions. This week I will pull the latest code from github and see if it addresses my current issues. 

Situation: I can join the Helium network and i can uplink messages to a Helium gateway. When I prepare for a downlink I notice that the gateway has actually send the (confirmed) downlink packet but I receive the message rarely. I notice the Helium network always transfers the packet with SF12. Why is that? I mean if it could be send in RX1 mode (which is set to SF10) then the transfer would be more efficient. Sometimes Radiolib detects that a downlink has been received; however without payload as if only the header information has been received.

So what would you advise me to check? I played already a bit with the timeouts but without good results. 
Do you think I should use the current code from Github as starting point or stay with the version that at least works? 

Should I provide you with logs from the Helium platform and logs from Radiolib from the different Radiolib versions (each with different issues)? Hope you can give me some guidance,

Kind regards,

Anouar Manders

clavisound, 2025-05-13:
> I notice the Helium network always transfers the packet with SF12. Why is that?

Off topic: I have the [same question](https://forum.chirpstack.io/t/txpower-29-and-rx2-window/22560). It's unanswered. Although my GW is idling (only receiving uplinks, not sending downlinks) it always sends RX2 and not only that, but full power at 29dB even the device is close to GW `> -100 dB`

For the question about RadioLib I don't know. In my short tests I failed to work with Helium and RadioLib (1.0.4). I don't know why - maybe I made some silly error. But it's ok with TTN.

jthiller, 2025-05-13:
Helium is just the "pipe" and doesn't do anything to instruct the device or change payloads. That responsibility lies with the LNS or device. 
Since you're running the same device in TTN and Helium, it may be worth taking a closer look at the LNS. If using the deprecated console.helium.com, move to a Chirpstack-based provider. 

Given this is off-topic for this thread I won't respond further here. Feel free to open a new issue and tag me.