jgromes/RadioLib #1304: help to init nicerf Lora1278F30_150 module, please

whoim2, 2024-11-01:
Hello.
Im have trouble with use this module:

my init:
```
SPISettings spiSettings(10000000, MSBFIRST, SPI_MODE0);
SX1278 radio = new Module(CS_PIN, DIO0_PIN, RST_PIN, DIO1_PIN, spi2, spiSettings);
radio.begin(freq, 125, 7, 5, LORA_SYNC_WORD, 15, 8, 0);
```
the module works and the power meter shows the power declared by the manufacturer, and the spectrum analyzer shows peaks at the specified frequency. Everything looks functional, but in fact the connection disappears in a couple of hundred meters, which is surprising - a similar module from ebyte (e32-400M) works at 15 kilometers.
I think that I am initializing the module incorrectly. As far as I understand, there is a RFO pin and PA_BOOST and you should switch between them via SetOutputPower (pwr, RFO_if_true).
Since the module frequency is lower than that of the e32, then the RFO pin is probably used, I thought. However, when you specify true in SetOutputPower, the spectrum analyzer shows nothing.

I am completely confused and ask for help in correctly configuring the module based on the configuration example from the manufacturer.

```
bool LORA::config()
{
	// In setting mode, RF module should turn to sleep mode
	// low frequency modeЈ¬sleep mode
	SPIWriteReg(LR_RegOpMode,LR_Mode_SLEEP|LORA_FREQUENCY_BAND);	
	// wait for steady
	delay(5);								

	// external Crystal
	SPIWriteReg(LR_RegTCXO,LR_EXT_CRYSTAL|LR_REGTCXO_RESERVED);	
	// set lora mode
	SPIWriteReg(LR_RegOpMode,LR_LongRangeMode_LORA|LORA_FREQUENCY_BAND);
	// RF frequency 433.5M
	setFrequency(433500000);	
	// max power ,20dB
	setTxPower(0x0f);
	// close ocp
	SPIWriteReg(LR_RegOcp,LR_OCPON_OFF|0x0B);	
	// enable LNA
	SPIWriteReg(LR_RegLna,LR_LNAGAIN_G1|LR_LNABOOSTHF_1);		
	
	_headerMode=LR_EXPLICIT_HEADER_MODE;
	
	// bandwidth = 62.5Hz, spreading factor=9,
	// coding rate = 4/5,implict header mode 
	setHeaderMode(_headerMode);
	setRFpara(LR_BW_62p5k,LR_CODINGRATE_1p25,LR_SPREADING_FACTOR_9,LR_PAYLOAD_CRC_ON);
	// LNA
	SPIWriteReg(LR_RegModemConfig3,LR_LOWDATARATEOPTIMIZE_DISABLED);	
	// max rx time out
	setRxTimeOut(0x3ff);
	// preamble 12+4.25 bytes  	
	setPreambleLen(12);

	// 20dBm on PA_BOOST pin
	SPIWriteReg(LR_RegPADAC,LR_REGPADAC_RESERVED|LR_20DB_OUTPUT_ON);  
	 // no hopping
	SPIWriteReg(LR_RegHopPeriod,0x00);     

	// DIO5=ModeReady,DIO4=CadDetected
	SPIWriteReg(LR_RegDIOMAPPING2,LR_DIO4_CADDETECTED|LR_DIO5_MODEREADY);   
	 // standby mode
	SPIWriteReg(LR_RegOpMode,LR_Mode_STBY|LORA_FREQUENCY_BAND); 
	
	// default payload length is 10 bytes in implicit mode
	setPayloadLength(10);

}
bool LORA::setTxPower(uint8_t power)
{
	if(power>0x0f) //15
		return false;
	SPIWriteReg(LR_RegPaConfig,LR_PASELECT_PA_POOST|0x70|power);
}
bool LORA::setFrequency(uint32_t freq)
{
	uint32_t frf;
	uint32_t temp1;
	uint32_t temp2;
	uint8_t reg[3];

	// Frf(23:0)=frequency/(XOSC/2^19)
	
	temp1=freq/1000000;
	temp2=LORA_XOSC/1000000;
	frf=temp1*524288/temp2;

	temp1=freq%1000000/1000;
	temp2=LORA_XOSC/1000;
	frf=frf+temp1*524288/temp2;

	temp1=freq%1000;
	temp2=LORA_XOSC;
	frf=frf+temp1*524288/temp2;
	
	reg[0]=frf>>16&0xff;
	reg[1]=frf>>8&0xff;
	reg[2]=frf&0xff;
	
	SPIWriteReg(LR_RegFrMsb,reg[0]);
	SPIWriteReg(LR_RegFrMid,reg[1]);
	SPIWriteReg(LR_RegFrLsb,reg[2]);	
	
	// read if the value has been in register
	if((reg[0]!=SPIReadReg(LR_RegFrMsb))||(reg[1]!=SPIReadReg(LR_RegFrMid))||(reg[2]!=SPIReadReg(LR_RegFrLsb)))
		return false;
}
```

In this case, the code example is given for frequency 433, that is, another module from this manufacturer, although on the same chip 1278. So I can't be sure that this setting is exactly correct. But I could use your hints to try.

whoim2, 2024-11-01:
```SPIWriteReg(LR_RegPaConfig,LR_PASELECT_PA_POOST|0x70|power)```
PA_POOST is correct settings?

jgromes, 2024-11-01:
> the module works and the power meter shows the power declared by the manufacturer

And what is this declared value, in dBm? I also assume your power meter is set up for the correct frequency, with correct impedance matching etc.

> but in fact the connection disappears in a couple of hundred meters, which is surprising - a similar module from ebyte (e32-400M) works at 15 kilometers

That suggests an antenna issue. If the power the module is outputting is as you expect, then the library is working correctly. I have previously verified the output power of SX1278, so I'm fairly confident this is correct in the library. As such, I will convert this to a discussion.

I also assume you have all the necessary paperwork you probably need to be transmitting at high power (30 dBm?) in some 150 MHz band.