jgromes/RadioLib #1314: setPacketSentAction(); Boot loop esp32-s3 psram

PTT3108, 2024-11-07:
**IMPORTANT: Check the docs**
Before submitting new issue, please check the [Troubleshooting Guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide) Wiki page and the [API documentation](https://jgromes.github.io/RadioLib/). If you are seeing an error code, we have [online status code decoder](https://radiolib-org.github.io/status_decoder/decode.html).

**Describe the bug**
A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode) **using the appropriate debug mode**.

<details><summary>Debug mode output</summary>
<p>

```
paste the debug output here
```

</p>
</details> 

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please use Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

<details><summary>Sketch that is causing the module fail</summary>
<p>
/*
  RadioLib SX128x Transmit with Interrupts Example

  This example transmits LoRa packets with one second delays
  between them. Each packet contains up to 256 bytes
  of data, in the form of:
  - Arduino String
  - null-terminated char array (C-string)
  - arbitrary binary data (byte array)

  Other modules from SX128x family can also be used.

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx128x---lora-modem

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>
// SX1280 has the following connections:
#define Serial Serial0
// SX1280 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
// SX1280 radio = new Module(10, 2, 3, 9);
SX1280 radio = new Module(15, 16, 17, 18);
// or detect the pinout automatically using RadioBoards
// https://github.com/radiolib-org/RadioBoards
/*
#define RADIO_BOARD_AUTO
#include <RadioBoards.h>
Radio radio = new RadioModule();
*/
// flag to indicate that a packet was sent
volatile bool transmittedFlag = false;

void IRAM_ATTR setFlag(void) {
  // we sent a packet, set the flag
  transmittedFlag = true;
}

// counter to keep track of transmitted packets
int count = 0;

// save transmission state between loops
int transmissionState = RADIOLIB_ERR_NONE;

void setup() {
  Serial.begin(115200);
    SPI.begin(37, 35, 36, 15);

  // initialize SX1280 with default settings
  Serial.print(F("[SX1280] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  // set the function that will be called
  // when packet transmission is finished
  radio.setPacketSentAction(setFlag);

  // start transmitting the first packet
  Serial.print(F("[SX1280] Sending first packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  transmissionState = radio.startTransmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                      0x89, 0xAB, 0xCD, 0xEF};
    state = radio.startTransmit(byteArr, 8);
  */
}

// this function is called when a complete packet
// is transmitted by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!

void loop() {
  // check if the previous transmission finished
  if(transmittedFlag) {
    // reset flag
    transmittedFlag = false;

    if (transmissionState == RADIOLIB_ERR_NONE) {
      // packet was successfully sent
      Serial.println(F("transmission finished!"));

    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);

    }

    // clean up after transmission is finished
    // this will ensure transmitter is disabled,
    // RF switch is powered down etc.
    radio.finishTransmit();

    // wait a second before transmitting again
    delay(1000);

    // send another one
    Serial.print(F("[SX1280] Sending another packet ... "));

    // you can transmit C-string or Arduino string up to
    // 256 characters long
    String str = "Hello World! #" + String(count++);
    transmissionState = radio.startTransmit(str);

    // you can also transmit byte array up to 256 bytes long
    /*
      byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                        0x89, 0xAB, 0xCD, 0xEF};
      transmissionState = radio.startTransmit(byteArr, 8);
    */
  }
}

</p>
</details> 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Additional info (please complete):**
 - MCU: ESP32S3 N16R8
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type SX1280
 - Platformio with arduino code
 - Library version 7.1.0

jgromes, 2024-11-07:
Seems like either a problem in the ESP32, or with the hardware. Why do you think this is an issue in RadioLib?

PTT3108, 2024-11-09:
Do you have any ideas to solve this problem?

jgromes, 2024-11-09:
Not really, because I don't what the problem is. I'm just pointing out that this does not look like an issue with RadioLib. 

>  This could be due to memory conflicts or improper handling of PSRAM.

These are unlikely to be caused by the library. First of all we have CI that performs multiple checks to ensure that common memory handling issues are not present. We also compile and run this library on many different platforms, none of which is reporting this issue. The library is also agnostic about how it handles memory, that's up to your build environment (e.g. the linker script). In terms of `setPacketSentAction`, it really just calls Arduino `attachInterrupt`

Closing until this can be shown to be caused by the library, and not an unrelated issue (e.g. bad build setup).

closed 2024-11-09 by jgromes

PTT3108, 2024-11-10:
thankss

SleepInfinity, 2024-11-11:
It happend to me only when i use esp adf

PTTEDU31, 2025-08-20:
I use the pins of lora to match the pins of esp psram.^^

ezefferth, 2025-08-20:
My log is just the same, but, always in `node.activateOTAA`. I dont know what to do. I try many forms, change IDE version, change version of RadioLib, i got the logs from debug. Its something like -707. I change the `radio.begin(915.0, 125.0, 9, 7, 0x12, 10, 8, 1.8, false);`, force to inicialize with 10dBm, but aways start with 30, and the device reset in step _activate_.. So i open discucion [1572](https://github.com/jgromes/RadioLib/discussions/1572), If anyone help-me to understanding.

```ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377176
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1180
load:0x403c8700,len:0xc2c
load:0x403cb700,len:0x311c
entry 0x403c88b8
Initialise the radio
[2.925611] RLB_DBG: 
RadioLib Info
Version:  "7.2.1.0"
Platform: "ESP32"
"Aug 20 2025" "15:12:55"
[2.929000] RLB_DBG: -707 at c:\Users\ezeff\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
[2.939487] RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
[2.939730] RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
[2.939815] RLB_DBG: 
[2.939839] RLB_DBG: M	SX126x
[2.942961] RLB_DBG: -707 at c:\Users\ezeff\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
Initialise begin OTAA
[4.978959] RLB_PRO: UL:  65 1 917.500 (6 - 6) | DL:   0 0   0.000 (0 - 0)
[5.046688] RLB_PRO: [MAC] 0x03
[5.046828] RLB_PRO: 00000000: 60                                               `                 
[5.046942] RLB_PRO: LinkAdrReq: dataRate = 6, txSteps = 0, nbTrans = 0
[5.047236] RLB_PRO: LinkAdrAns: 07
[5.047325] RLB_PRO: [MAC] 0x04
[5.047367] RLB_PRO: 00000000: 00                                               .                 
[5.047451] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^0
[5.047491] RLB_PRO: [MAC] 0x05
[5.047554] RLB_PRO: 00000000: 08 68 e2 8c                                      .h..              
[5.047641] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 8, freq = 923.300
[5.048406] RLB_PRO: [MAC] 0x08
[5.048531] RLB_PRO: 00000000: 01                                               .                 
[5.048622] RLB_PRO: RXTimingSetupReq: delay = 1 sec
[5.048690] RLB_PRO: [MAC] 0x09
[5.048744] RLB_PRO: 00000000: 1d                                               .                 
[5.048831] RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
[5.048916] RLB_PRO: [MAC] 0x0c
[5.048969] RLB_PRO: 00000000: 65                                               e                 
[5.049052] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
[5.049131] RLB_PRO: [MAC] 0x0f
[5.049184] RLB_PRO: 00000000: fa                                               .                 
[5.049270] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
[5.118687] RLB_PRO: JoinRequest (DevNonce = 0):
[5.118933] RLB_PRO: 00000000: 00 00 00 00 00 00 00 00 00 ec f6 06 d0 7e d5 b3  .............~..
[5.119074] RLB_PRO: 00000010: 70 00 00 0b c8 8b 5d                             p.....]           
[5.119616] RLB_PRO: 
[5.119668] RLB_PRO: PHY:  Frequency = 917.500 MHz, TX = 30 dBm
[5.121267] RLB_PRO: LoRa: SF = 8, BW = 500.0 kHz, CR = 4/5, IQ: U```