jgromes/RadioLib #1322: Only supports ESP32 but this is not specified in the ESP-IDF manifest file

IanBurwell, 2024-11-18:
In the ESP components registry RadioLib is listed as "supports all targets" but the code only [compiles for a basic esp32](https://github.com/jgromes/RadioLib/blob/c7d6ad3e951300cd37871728e40894fe01eebc2a/src/hal/ESP-IDF/EspHal.h#L10).  The following should be added to the `idf_component.yml` file:
```yml
targets:
  - esp32
```

Esp component registry screenshot:
[image]

jgromes, 2024-11-18:
I'm a bit conflicted here, because the fact that this only compiles for an ESP32 is intentional by that macro check you have pointed out. The reasoning is that the ESP-IDF HAL is pretty much just an example how to implement RadioLib HAL in ESP-IDF environment. It's not really the intention to have a fully compatible HAL for all ESP targets - at that point we are basically re-inventing Arduino.

Bulk of this library is platform-agnostic, all the platform-dependent code is in the HAL. But I do get that this can cause some confusion. There are a few options how to proceed:

1. Do what you suggest and mark this as only supporting the base ESP32. I don't like that approach because it hides the fact that porting to a new ESP device is as simple as we could make it. You don't have to do any modifications to the library core, you just have to provide basic stuff like SPI reading and writing.
2. Move the current ESP-IDF HAL back just to examples and make it clear in the readme that porting to something like ESP32-S3 is up to the user.
3. Make the ESP-IDF HAL compatible with all ESP32 targets. This is by far my least preferred option since my experience with ESP-IDF is limited, and testing this would be challenging to say the very least.

IanBurwell, 2024-11-18:
I think the confusion here is that _as an esp-idf component_ it currently only supports the ESP32 out of the box. But I agree that saying it only supports the esp32 does hide the fact that it should be relativity easy to support other variants (and this is just something the user needs to do). Maybe the error/comment should more specifically call out that other targets (etc esp32s3) should be implemented and tested by the user, using the esp32 hal as an example?

Regardless I may take a deeper look myself at the current example HAL and see what changes one would need to make to have it support the ESP32S3 (which I have).

jgromes, 2024-11-30:
For the time being, I went with option 2 - that is to move the HAL back to examples. I have also slightly updated the errro message to hoefully make it clearer that this is just an example, and that support other targets requires some porting from the user.

closed 2024-11-30 by jgromes

IanBurwell, 2024-12-02:
I think that is a good way of handling it for now. I got things running on the ESP32S3 by changing the SPI functions you defined in that example HAL to use the ESP-IDF [SPI master driver](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/spi_master.html). I could honestly see Option 3 being implemented easily with the [ESP-IDF's HAL drivers](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/hardware-abstraction.html), but they are not technically stable yet so it would have to pin the supported ESP-IDF version. I will continue with my modified HAL for now as I am pinning my own ESP-IDF version.

timoehl, 2025-05-30:
> I think that is a good way of handling it for now. I got things running on the ESP32S3 by changing the SPI functions you defined in that example HAL to use the ESP-IDF [SPI master [...]

@IanBurwell 
I also attempted to change the example HAL to use the SPI master driver but was mostly unsuccessful (see #1130) . Could you post your HAL using the SPI master driver or give me some pointers what my issue could be?

IanBurwell, 2025-06-01:
@timoehl I have a HAL that I wrote [here](https://github.com/IanBurwell/DynamicLRS/blob/main/components/DLRS_LoRadio/radiolib_esp32s3_hal.hpp) that works with my esp32s3 and has a few small improvements over the one included as an example with radiolib. It should work with other ESP variants that support the ESP-IDF SPI driver, but I've only tested it on my esp32s3.

CharlesMcBaker, 2025-06-29:
> [@timoehl](https://github.com/timoehl) I have a HAL that I wrote [here](https://github.com/IanBurwell/DynamicLRS/blob/main/components/DLRS_LoRadio/radiolib_esp32s3_hal.hpp) that works with my [...]

Does your hall fix the issue where if Radiolib debug is off, you get weird ISR issues? If I build without RadioLib debug options I can't get a stable connection.