jgromes/RadioLib #1324: lora sometime stuck after dio1 is triggered, but getPacketLength is 0

trustmiao, 2024-11-18:
in case of receiving(no transmitting job), when dio1 interrupt triggered, there are chances getPacketLength() return 0. everytime this happened the lora will stop working, no more interrupt will be triggered unless it will be reset. 

MCU: esp32 idf
Wireless module type: SX1262
Library version 7.0.2

StevenCellist, 2024-11-18:
Hi @trustmiao, there is a reason that there is a template when you file an issue. That reason is that we can't help you unless you fill out the whole template. So please do that first (and always)!

trustmiao, 2024-11-20:
I will try to upload the debug info. will RADIOLIB_DEBUG_BASIC level be enough?

StevenCellist, 2024-11-20:
Basic is good, and please also supply all other information requested by the template - not just the debug info.

trustmiao, 2024-11-25:
I edited original post, wish this will explain what happened. however, there are not much debug info when issue happens.

jgromes, 2024-12-15:
Closed due to inactivity, we can't do much without the debug output.

closed 2024-12-15 by jgromes