jgromes/RadioLib #1327: RADIOLIB_DEBUG_PROTOCOL on ST nucleo-WL55JC1 explodes in cascades of garbled output

MortenGuldager, 2024-11-21:
Trying to use `RADIOLIB_DEBUG_PROTOCOL` to get a understanding of why my device works better with some LNS than others.

When I enable `RADIOLIB_DEBUG_PROTOCOL` in `BuildOptUser.h` the console just print a garbled mess. Seems to begin imediatly the moment where the first debug is to be printed.
`RADIOLIB_DEBUG_BASIC` is working as expected. 

- LoRaWAN
- PlatformIO (in vscode)
- ST Nucleo WL55JC1

The program is as simple as I can make it.

```
#include <Arduino.h>
#include <RadioLib.h>

typedef uint64_t lorawan_eui_t;
typedef uint8_t lorawan_key_t[16];

uint64_t joinEUI = 0x0000000000000000;
uint64_t devEUI  = 0xbaddecafcafedeed;
lorawan_key_t nwkKey = {
    0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 
    0x88, 0x77, 0x66, 0x55, 0x44, 0x33, 0x22, 0x99
};
lorawan_key_t appKey = {
    0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 
    0x88, 0x77, 0x66, 0x55, 0x44, 0x33, 0x22, 0x99
};

STM32WLx radio = new STM32WLx_Module();
const LoRaWANBand_t Region = EU868;
const uint8_t subBand = 0;
LoRaWANNode node(&radio, &Region, subBand);

static const uint32_t rfswitch_pins[] =
    {PC3, PC4, PC5, RADIOLIB_NC, RADIOLIB_NC};

static const Module::RfSwitchMode_t rfswitch_table[] = {
    {STM32WLx::MODE_IDLE, {LOW, LOW, LOW}},
    {STM32WLx::MODE_RX, {HIGH, HIGH, LOW}},
    {STM32WLx::MODE_TX_LP, {HIGH, HIGH, HIGH}},
    {STM32WLx::MODE_TX_HP, {HIGH, LOW, HIGH}},
    END_OF_MODE_TABLE,
};

void setup()
{
    Serial.begin(115200);

    radio.setRfSwitchTable(rfswitch_pins, rfswitch_table);
    Serial.printf("radio.begin: %d\r\n", radio.begin());                                
    Serial.printf("beginOTAA %d\r\n", node.beginOTAA(joinEUI, devEUI, nwkKey, appKey)); 
    Serial.printf("activateOTAA %d\r\n", node.activateOTAA());                         
}

void loop()
{
    static uint8_t i = 0;
    uint8_t data[] = "ABC";
    data[0] = i++; // make first byte increment for tracking on LNS
    Serial.printf("sendReceive: %d\r\n", node.sendReceive(data, sizeof(data), 22)); 
    delay(60 * 1000);
}

```

HeadBoffin, 2024-11-21:
There is a reasonable chance that the garbled mess is in fact information, so before we disappear down too many rabbit holes, can you post some of the output. You didn't say if it still worked even if the output looks wrong.

The latest version has passed all the pre-compliance checks for LoRa so if you could tell us which LNS's are causing issue & what those issues are that would be useful - it is tested against TTS by default and then Chirpstack occasionally.

MortenGuldager, 2024-11-21:
Hmm, I got it wrong, it actually do get a few lines of normal output before it turns crazy. I test against a completely default configured chirpstack. I dont see any traffic from the device if I enable the flag.
The output looks like: (and it runs really fast, what you see below is approx 0.5 second)
```
radio.begin: 0
beginOTAA 0
RLB_PRO: Setting up dynamic channels
RLB_PRO: UL:   0 1  (0 - 5) | DL:   0 1  (0 - 5)
␀BRLB_PRO: UL:   1 1  (0 - 5) | DL:   1 1  (0 - 5)
␀BRLB_PRO: UL:   2 1  (0 - 5) | DL:   2 1  (0 - 5)
␀BRLB_PRO: [MAC] 0x03
RLB_PRO: 00000000: 20                                                                 
RLB_PRO: LinkAdrReq: dataRate = 2, txSteps = 0, nbTrans = 0
RLB_PRO: LinkAdrAns: 07
RLB_PRO: [MAC] 0x04
RLB_PRO: 00000000: 07                                               .                 
RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
RLB_PRO: [MAC] 0x05
RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 
␀RLB_PRO: [MAC] 0x08
RLB_PRO: 00000000: 01                                               .                 
RLB_PRO: RXTimingSetupReq: delay = 1 sec
RLB_PRO: [MAC] 0x09
RLB_PRO: 00000000: 05                                               .                 
RLB_PRO: [MAC] 0x0c
RLB_PRO: 00000000: 65                                               e                 
RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
RLB_PRO: [MAC] 0x0f
RLB_PRO: 00000000: fa                                               .                 
RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
RLB_PRO: 
?��!?%?9?M?�������������P�P�P�����D������␀␀␀␀␀␀␀␀������������������������������␀␀␀␀␀␀␀␀��␀A␂�␀���␀C0�O�A␄O�C␅��␅���␂␏␟�T�␀␌U�␂␌␡�d\␡�e\␀��O�TT��UU��␂�␎�"�␅�␘A�␀␁␃�␎�␘C�␅�YA␎�� ␅␎� ␎␁*␃�␎�(�L�␂␌C�␅��␘Q��q␁�␀E␇�O�␀␎��␀␌~�␀␀n�␁␁���␟␛ӱ�␀␟␌�_�0␀O�<␌␄�␁␄O�DR␒��␏�𚀼�␀�_�P␌P�␀␀A�␄QA�␅␁0�_�L␌@AA�␁␁␁<(����␟�ґ�␀␏␄�␁F␀ ���� 3��␋␃�� ␂␌�␌�␂�␔␌��␌␂␁�␌�!�␂�␌�␂�␔␂ؿ�� ␌␁�␂� �␌�ܿA�␌␁�@�␚��␁�␄Q)C0�o�␄␄␟<␜�␌4␎�␄�␔␄�� ␂ �␄�␁�␂�@�␃␀!�␄�E�␃␁0���␌␄�� ␂ �␂�␁�␄�@�␃␀)F0�!�␄�)F0���␀␏��␓␆���␑␁4␁=N�␡�d\␘�␡�e\)Д�␅���␂␏␅�T�␀␌␄�␙F␐F0���␃␏␞�␀!␀ 0�_�T\␅�@␀IA(�A�␀A0�␔��␄<�␁��␑0�␁�␀EE��AA�p␁O�␀␀0�␡�d\␚�␙F␐F␡�e\␜�␋F␂FP�␁4␆�R�␃5��␃␏A�␀!0�␀���␀␏␄�␀!pG0�O�d␄�2␄O�␀␅O�␀␁P�␀���␀␏␄�␀!pG0�O�d␄�2␄␐�␀EH�@BO�␀␁>�␀�B␀O��␁O�1␁O�␂p␟�␒�␡C��␡O��`QpG2�␡�pG��␡O␄�A�␀!pG0�O�`t␁�␀E!�␀A␜�␀�P�␁�pG0�O�␀␅
�P�␁�pG0�␑�␀E␂�@Ba�A␁O�d␄�2␄_�\?�خO�␃␂_��␌␘�␃2_��␌␘�␃2␂��␂�� ␃␀�␃� �␂�␁�␃�@�␎␀!�␂�␔D��␀�p�O��␌L��l␜�␑T␝�␜�␓U��␌␏��␌␏␀���,D��␃␆!�LQ#�LSP�␁5␘�R�␃5A�␑C�␓8Р�␂�O�␀␅��␂�␆�␀B��␃�O�␀␆��␃V��␀␏␘�N�␁␎���␄��␀␡d�@t␄�_�N␎mAF�␆␆B��!A�UQO�� @�^PO��.���␌�����o␞ؾ�␀�_�P␎P�␀␀A�␄Qp�␆�␀FF�␁␁@�␂␀��␃␁��\␄¿��␌␅A�␄Qp�A�␑O�␀␎␁<␀�␔�6␏޿␀ ␁�␀Ap���␀␄ <5�␌4␛�␄�␔␄�� ␅␀�␅� �␄�␁�␅�@�␂␀␁�␀B!�␀A␐��p!�␄�B�␆␁^�C� ��pp���␌␄�� ␅␀�␄� �␅�␁�␄�@�␂␀␁�␀A␐��pA�␀␁^�C� ��pp��� ␅␀�␅�N�␂␎ �␄�␁�␅�C�␂␃!�␄�␁�␀A!�␄� �␂␀␀��p^�C� ��pp���␀␏␏�␁�␀F@␀A�␁␁␑��␁<��A�␆␁��␀␏␘�pG␃�␀FR␀C�␃␃␓��␁=��C�␆␃pG��␌␏␌�␓U␘���␌␏␌�P�A␆␘�R�C␆�с�␃␁␁�␀AO�␀␀p�P�A␆␆�␐F␙FR�C␆␙Д�␌␏␂�P�␁6␓ѕ�␌␏␅�R�␃6␜�␐F␙F
с�␃␁␁�␀AA��AA�p␁O�␀␀p�A��AA�x␁p�p�O��␌L��l␜�␑T␝�␜�␓U��␌␏��␌␏␀����␅␄��␃␎R�␃5O�␁1␀���O�␃3O��UE�␓␓C�␒cO�␂"E�␑␕E�␐eO�␀&␎�␀A���BD��␄␄�@t␂�O�2␂�␚e�␃␅O�2␂O�␐O�␀,��␂␎u�␃␎"��␚uF@�␌␀O�2␂��␂␎u�␃␎"��␚uF@�\␀O�2␂��␂␎u�␃␎"��␚uF@�␀O�2␂��␂␎u�␃␎"��␚uF@��␀U�␆␎␘�O�␅␕E�␖uO�␆␖O��␃C�RsO��␂_�␜␜��␑�␟␋�A�␀␁O�␀␀O�␀L��␑�␟␄�␁C␀ ���␌�����o?�����␃␌␄���␂␌_�P␌P�␀␀A�␄Qp�␎�␀NN�␑1␔�\␄¿��␌␅A�␄Qp�A�␑O�␀␎␁<��E�␆␎��␌�␓U��␌���␌␏?�;���␌␏
�R�C␅␡��␒�O��<␆�␀�O�␁␌␂�␀�O�␁␌M�␄�O�A␌␡�l\O�C␌␘�␡�l\␛�␁�P�A␌␌�R�C␌��␃␏␂���␂␏␀ pG␐�␀␏��␃␏X����B,��␗o��p@�␁␀pGO�A␌␡�l\␂�P�␁<␇�O�C␌␡�l\��R�␃<��]�␄␋pG␀�␁�␀LO��␂L�Pp��␀O@�� �␁␀pG␑��O!�␓�8r��␁�␀@pGA�␑O�RR��␘␂�� ␌␐�␌� �␂�␘�@�␁␀O��#O��#␃�␌�@�␌␀#�␂�O�C␃��␡�bS␇�P�␁3␞�O��@@�@␀pG␁�␀@@��@@�␀␀pG␀���␀@␂�␀���␀AB␀␟�_�A␃��␃␏␡�"l␡�#lj�O�␒b��␓c���␘A@H@A@��[B␙+��pG␐�␀O@�␀␀ �␡@␘�@B␑�␀OA�␀␁!�␡A␘�IB��␃␏?Т�␁␂A�␃�␐�␌␀�� ␃␁�␃�␀�␀C␂�IB`�@␀��␀␏␓Ӱ�␡␆�O�1␁␂�␁␂�*Qұ�␀O@��� �␁␀@�␃␀pGI␀@�␀␀␁:(���␀␏�Ұ����␌��␌␂␀�␌�␀��PRB␘C���@␘CpG��␀␏��␀␁␆���␀␀␁2␁;��O�A␃␡�"l␘�␡�#l!В�␃␏␄В�␀FpG��␁␏␜�␀ pG␒�␡O␄�@␀(�@�␀@pG␒�␀r<�␀�␀␀pG␀�␀CC��@@�␀␀pG␡�"b␖F␡�#c␁FB␂␆�_�A#��␁␏@�␀pGO�␀␃␄�␀�␐�␀CH�@B_�␀�pGC�C␁FO�␀␀␜�P�␁�pGO�␀␃
�␀�P�␁�pG␑�␀C␂�@Ba�A␁_�␁␌␂��F␁F␀ C����S��␀␃���:���S␐�␁�␂�cD␀�␂��� ␂��␀O �␂�C�␂� �␁␀pG␂� ␂␁�␂��� ␂P�L␀!�␂�C�␂� ��ppGO��␌␜��R␞�␜��S��␌␏��␌␏o�␚D��␁␌@␂␘�_�A!␞�O�␀cC�P␐C�Q␑��␁1␌�␀@��␀␏>�I␀A��q[␀@�␁␀b�␡␂�*␝س�␀O@��� �␁␀pG��␀␏␌�␀�I␂L�P @�Q ␡:¿���␃@��PpG@�␀␀O�␀␃␁:]�␒�␙␏ܿ␀�␀@pG��␀␂A␀!�␂��� ␂␀�␂�_�1␀@�␀␀S�L� ��ppG��␀␏␀�␀L␂�@␀␐�␀␏␁:��@�␌␀��␀␏␁�␀L␂�I␀␑�␀␏␁;��A�␌␁��␌��S��␌␏␘���␌␏
�0�␀L␘�1�␀L�р�␁␀␀�␀@pG��␀␏␗���␀F��␀␏��␀O␔В�␌␏␁�B␂␏ѓ�␌␏␃�K␂␘р�␁␀␀�␀@@��@@�␀␀pG@��@@�@␀pGO��␌␜��R␞�␜��S��␌␏��␌␏iТ�␃␂��␁␌I␂O�@ 7�O��SC�␑␑C�␐␓␌�␀@�B8�[␀B�}␂O�␀␌�B$�[␚@�␌␀��Q␏$���Q␃@�\␀��␏$���␃@�␀���␏$����␃@��␀␛␁␘�_�␜␜���*?�P��B@��� �␁␀pG␌�␀LL�P ␡2¿���␃@��PpG@�␀␀O�␀␃␁:7��␀␏␀�␀L␂�@␀␐�␀␏␁:��@�␌␀��␀␏␁�␀L␂�I␀␑�␀␏␁;��A�␌␁��␌��S��␌�B␂␡�}���␌␏␡�pFv��␌␏␄�K␂?�LFn�0�␀L␘�1�␀L��0�␀B␡�\�1�␀C␡�<�_�␀�O��<␆�␀�O�␁␌␂�␀�O�␁␌M�␄�O�@␂O�A␃␡�"l␘�␡�#l␑�␁�R�S␌␘���␁␏X���␃␀���␗8�o��p␘�@�␁␀pG␡�"l␂�_�@,␅�␡�#l��_�A,��]�␄␋pG␀��FaF��␏�����␀(H�␐�␀␏␏�M�����␌�␁ ␀ ]�␀�M�����4�␁ ␀ ]�␀�M�������␁ ␀ ]�␀�M�������␁ ␀ ]�␀�M������O�␀#C�␀C␐�␀O#�␂�␘�@BpGO�␀␀pG␒�a␏␁�B␂␅�␐�␀�o�␀@pGO�␀␀pGB␀␎Ҳ��O␋�O�␃��␒b �O�␀#C�␀C#�␂�pGO�␀␀pG␒�a␏␁�B␂␂�O��0pGO�␀␀pG␀�S�J�␀�␀(␜�O��1O��0␀�␌��␌m�␄�␌����␄���␂#␄�pGpG␀�␃F␓�␁+␀*��␘␚␁8pG␀␀␀␀␀␀␀␀␀␀␀␀␁��␁␐*+�␐�␇�␐�␁;␁:�B-�␐�␇␏B����A�␁!A�␁A"�␇␄␡�␀␇␀#��␂<��␁␅��␁␆��G�������G����������␁��␁␂�␇␂2�␐�␁;␁:��␁␃�S�␀���O�B␁ ���0#�␘␃���0��␀4#��cC�␀s��␀4��␀4␃��c␁�␁�$��#␋�␀#��␐␃����␒#␀��

```

MortenGuldager, 2024-11-21:
I have now tried the program on a ATSAMD21+RFM95 combo. Changed a few lines for the new radio:

```
#define SPI_CS_PIN 0         // grey
#define RFM95_DIO0_IRQ_PIN 1 // purple
#define RFM95_RST_PIN 2      // orange
Module module(SPI_CS_PIN, RFM95_DIO0_IRQ_PIN, RFM95_RST_PIN);
RFM95 radio(&module);
```
and removed the `setRfSwitchTable` thing.

Prints the debug just perfect, and connect to my LNS too. Some time ago I did the same with an ESP, no problems with the `RADIOLIB_DEBUG_PROTOCOL` either.

jgromes, 2024-11-21:
Couple of things to check would be whether the other debug flags also cause this problem, and how much the compiled binary size changes when you enable them.

Which version of the library are you running?

Garbled UART output is usually one of the following:

1. Something else besides the Serial pulling the Tx pin - seems unlikely
2. Incorrect baudrate - also unlikely, would be strange to change in the middle of the program and the library definitely does not touch it
3. Program fault. If this is only being seen with enabled protocol debug, it could be some issue in the printing, so it could go away with different debug level.

StevenCellist, 2024-11-21:
I assume you're using STM32duino, just to be clear? Haven't had any issues with it through the Arduino IDE, didn't test PIO.

HeadBoffin, 2024-11-22:
Are you using the actual Nucleo-WL55JC1 board - if so, I can give it a try and then we can debug from there.

MortenGuldager, 2024-11-22:
correct, a Nucleo-WL55CJ1

MortenGuldager, 2024-11-22:
And I can add that I have just repeated the test, this time building on a windows machine. Same result. The previous tests as all been built on Linux

lyusupov, 2024-11-22:
@MortenGuldager 

Not sure that this is 100% good hint for you
but Arduino Core for STM32 was (is ?) known to have very small default size of UART RX and RX buffers.

Try to set 
```
-DSERIAL_TX_BUFFER_SIZE=512 -DSERIAL_RX_BUFFER_SIZE=512
```
in your sketch build environment ( `build_opt.h` ) .

Read this wiki page for details:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/Custom-definitions

MortenGuldager, 2024-11-22:
> @MortenGuldager
>
> Try to set
> ```
> -DSERIAL_TX_BUFFER_SIZE=512 -DSERIAL_RX_BUFFER_SIZE=512
> ``` 
> in your sketch build environment ( `build_opt.h` ) .

I might need a little help on this. I added the two compile flags to build_flags in my `platformio.ini`. That didn't help. 
Then there is the `build_opt.h`  where your link says I could try to add some higher values for `SERIAL_TX_BUFFER_SIZE` and `SERIAL_RX_BUFFER_SIZE`, But it does not tell where I should place the file. 

I even went to the extreme and put this in the file while moving it around:
```
#define SERIAL_RX_BUFFER_SIZE CrazyHorse
```
rationale would be that if the compiler actually pick up the file and tries to use the crazy value I would at least get a build error. But that never happened....

StevenCellist, 2024-11-27:
@MortenGuldager could you please try the fix proposed by @lyusupov? Please note that the referenced `build_opt.h` file is a different one than the one used inside the RadioLib library.

jgromes, 2024-12-15:
Closed due to inactivity. We can reopen this if there are more reports of STM32WL devices misbhaving.

closed 2024-12-15 by jgromes

goshawk22, 2024-12-27:
Was just about to make a comment saying I had the same problem, but luckily noticed some funny behaviour with some of my own printf statements. Try using newlib standard instead of newlib nano (default on stm32duino). Fixed the issue for me and all the debug messages are now printing normally.

MortenGuldager, 2024-12-28:
> Was just about to make a comment saying I had the same problem, but luckily noticed some funny behaviour with some of my own printf statements. Try using newlib standard instead of newlib nano (default on stm32duino). Fixed the issue for me and all the debug messages are now printing normally.

How do I change the library?

goshawk22, 2024-12-28:
In the drop-down settings in Arduino ide there is an option to change it. I can't remember exactly how to do it in platformio but I know that you can. It will probably be in the stm32duino docs or in the platformio docs for the platform.

MortenGuldager, 2024-12-28:
And for future reference, here is how I did it in `platformio.ini`:
```
build_flags = 
    -D RADIOLIB_DEBUG_PROTOCOL
    -D PIO_FRAMEWORK_ARDUINO_STANDARD_LIB
```

No more cascades of giberish then enabling protocol debug. Fantastic. thanks!