jgromes/RadioLib #1329: Module LLCC68 does not initialize

jrhbcn, 2024-11-22:
I am using a [EBYTE E220-900M22S](https://www.cdebyte.com/products/E220-900M22S) module that comes with a LLCC68 chip connected to a ESP32 D1 mini. I am experimenting with the SX126x_Channel_Activity_Detection_Receive example. Using the LLCC86 module does not seem to work:

I have changed this part in the example:

```
LLCC68 radio = new Module(5, 32, 33, 25);
int state = radio.begin(BAND, SignalBandwidth, SpreadingFactor, CodingRate, SyncWord, TxPower, PreambleLength, 0, false);

```

And it gives the error:

```
[SX1262] Initializing ... RLB_DBG: 
RadioLib Info
Version:  "7.1.0.0"
Platform: "ESP32"
Compiled: "Nov 22 2024" "15:39:17"
RLB_DBG: SX126x not found! (1 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
RLB_DBG: Expected string: LLCC68
...
RLB_DBG: SX126x not found! (10 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
RLB_DBG: Expected string: LLCC68
RLB_DBG: No SX126x found!
failed, code -2

```

However, if I change to using the module SX1262 (LLCC68 is supposed to be pin compatible except that it does not support some configurations), then the module initializes, and it seems to receive messages from other LoRa modules:

```
[SX1262] Initializing ... RLB_DBG: 
RadioLib Info
Version:  "7.1.0.0"
Platform: "ESP32"
Compiled: "Nov 22 2024" "15:39:17"
RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
RLB_DBG: 
RLB_DBG: M      SX126x
success!
[SX1262] Starting to listen ... success!
success!
[SX1262] Data:          0xAAAA
[SX1262] RSSI:          -59.00 dBm
[SX1262] SNR:           9.00 dB
[SX1262] Frequency error:       -30079.69 Hz
```

What is happening? It seems the LLCC68 chip is actually returning the SX1261 string? Both SX1261 and SX1262 classes seem to work as intended (but not LLCC68)? Is this normal, or is it just a bogus chip?

jgromes, 2024-11-22:
There's been a couple other issues/discussions reporting this. The version string is stored in the module itself, so it looks like the module "thinks" it's an SX1261. So the question is whether there's an SX1261 inside that Ebyte module, or if the string is wrong and it really is LLCC68. I don't have that module, so I can't check this.

One way to figure this out by configuring a SF/BW configuration that the LLCC68 does not support (e.g. SF12, BW 125 kHz). If it is LLCC68, it should fail. If it is the more capable SX1261, it should have no issues and freely transmit. @jrhbcn if you could try taht, that would be great.

Since this came up a couple times already, I have added a note about this to LLCC68 examples, and [the troubleshooting guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide).

codev123, 2024-11-23:
i have also encountered the same issue, previously ordered ebyte 220- 400m22s LLCC68 modules were working with constructor LLCC68. but newly ordered ebyte 220 - 400m22s are not working with constructor llcc68, while its working  with constructor SX1262. so i ipened shield from module and checked the ic on both prev and new modules. the ic in both is LLCC68. so i think its supplying wrong string.
what should be the solution sir?

jgromes, 2024-11-23:
@codev123 it's a good question to ask Semtech ;) The best we can do is test it - would it be possible to perform the test I proposed above, i.e. configuring a LoRa setting not supported by the LLCC68 and trying if it works? If the test fails, then we know it's just the string in the device that is incorrect, and we can work around that. If it works, there might a different issue (wrong marking? though it seems unlikely ...)

Also, if you could take a photo of both ICs so that the markings are visible (the old and the new one) and post them, that would be very helpful. Maybe there's a specific batch of the ICs that has this problem.

codev123, 2024-11-23:
Thank you sir. i have already tested what you have mentioned, and when i initialise llcc68 with constructor SX1262, with bandwidth 125.0 and sf 12, it got initialized but i wasnt able to transmit or receive, but able to transmit and receive with 125.0, 9 values of bw and sf. so that confirms its LLCC68 module?

jrhbcn, 2024-11-23:
I can also confirm that my [EBYTE E220-900M22S](https://www.cdebyte.com/products/E220-900M22S) (brought less than 2-3 weeks ago) also works doing a pingpong example with BW=125 and SF=12. So it does not have a LLCC68 inside. 

@codev123 you can easily add images to github comments, just copy and paste and it will be included! 

I am going to investigate if there is a sx1261 or a sx1262 chip instead of the advertised LLCC68. The only difference I can see is that sx1261 supports +15dBm and sx1262 +22dBm. @jgromes, I see that in the library, sx1261 is capped to 14dBm not 15dBm. Is that intended?

jrhbcn, 2024-11-23:
Hello again, 

I have done some more tests with the PingPong example. PingPong sender is the E220-900M22S module with the "fake" LCC68 chip. PingPong receiver is a LILYGO T3_V1.6.1 integrated ESP32+LoRa (sx1276). I have configured the receiver with some fixed settings: BW=125, SF=9, CR=5, dBm=20 (max for sx1276). 

I compiled the PingPong sender (fake LLCC68) with the sx1261 and sx1262 classes at different dBm:

**1. Using sx1261 class with dBm=14** 

Receiver reports RSSI of around -125dBm. Really weird as both receiver and sender are face to face. This is the receiver log:
```
[SX1276] Data:          Hello World!
[SX1276] RSSI:          -124.50 dBm
[SX1276] SNR:           -6.50 dB
```

while the sender log reports much higher RSSI:
```
[SX1261] Data:          Hello World!
[SX1261] RSSI:          -42.00 dBm
[SX1261] SNR:           10.50 dB
```

Measuring the current consumption of the E220-900M22S shows values of 10mA (again too low which correlates with the low RSSI reported):
[image]

**2. Using sx1262 class with dBm=14** 
Receiver reports:
```
[SX1276] Data:          Hello World!
[SX1276] RSSI:          -68.00 dBm
[SX1276] SNR:           8.50 dB
```

Sender reports:
```
[SX1262] Data:          Hello World!
[SX1262] RSSI:          -46.00 dBm
[SX1262] SNR:           10.50 dB
```

Current consumption looks more in line with datasheets with 82mA when transmitting the message:
[image]

**3. Using sx1262 class with dBm=22** 
Receiver reports:
```
[SX1276] Data:          Hello World!
[SX1276] RSSI:          -67.00 dBm
[SX1276] SNR:           9.25 dB
```

Sender reports:
```
[SX1262] Data:          Hello World!
[SX1262] RSSI:          -45.00 dBm
[SX1262] SNR:           9.75 dB
```

Current consumption stays in 82mA when transmitting the message:
[image]

Sorry for the extra-long message but I am a bit loss here. It seems that using sx1262 class allows the module to transmit with more power. However, it does not seem to change transmission power (receiving RSSI and mA remain constant). So actually no idea what is going on in that chip. Also, compared with the LILYGO, it seems to transmit at a lower power even with the sx1262 class.

 @jgromes, @codev123. Any ideas? If you want me to make any other test to understand a bit better what is happening, let me know!

jgromes, 2024-11-24:
@jrhbcn thank you for the test!

>  I see that in the library, sx1261 is capped to 14dBm not 15dBm

That might be just a small mistake on my part; you are correct that it can go up to +15 dBm, so I will fix that.

[image]

Regarding the power, it seems that despite what it reports, the LLCC68 is actually based on SX1262, at least for the PA configuration. In the previous screenshot, the `deviceSel` parameter is fixed to 1 for SX1261. However, LLCC68 datasheet shows that is should be set to 0, which is the value SX1262 uses.

[image]

So to summarize, the LLCC68 you have reports being SX1261, but behaves as if it were SX1262 🙃 If you could confirm this still works with SF12 BW 125 then I'd say you lucked out, because from what I understood, LLCC68 was supposed to be a cheaper alternative to SX1262, with limited features. Now you got the full feature set ;)

jgromes, 2024-11-25:
@jrhbcn thank you for the additional test! Lacking any official information from the manufacturer, I think the best we can do for now is to accept the "SX1261" string as valid for LLCC68 and leave all the limits as the datasheet suggests.

jrhbcn, 2024-11-26:
Yes, I agree @jgromes, this is the best we can do. Based on my tests, I would recommend, if LLCC68 fails to initialize, to use the sx1262 class instead of the sx1261 (even if it reports the latter). I don't know if you want to change the comment in the examples to reflect this. In any case, I think we can close the issue now. Many thanks!

closed 2024-11-26 by jrhbcn

jgromes, 2024-11-26:
@jrhbcn I can't really "change the class if LLCC68 fails to initialize", that's up to the user. What I can do is allow them to use the device even if it has the SX1261 string. The mismatch will also be reported to the debug port.

jrhbcn, 2024-11-26:
> @jrhbcn I can't really "change the class if LLCC68 fails to initialize", that's up to the user. What I can do is allow them to use the device even if it has the SX1261 string. The mismatch will also be reported to the debug port.

I was just referring to the comment line in the LLCC68 example. However, I just saw you removed them. In any case, I think that commit to the LLCC68 is perfect! ;)

thanosbi, 2024-12-10:
Tested my E220-900 modules today. The chip inside is marked LLCC68 as advertized with markings 2401  66315. 
I can't tell if its a fake. Of course reports as sx1261...
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
So i configured it as SX1262 with xtal and works fine.
 First tested power output with a 5cm wire as antenna BW 125 ,SF 9 and currentlimit 140 ,RXEN and TXEN pins connected on MCU. Tested from 10 to 22db the module consumes  73 to 134mA  rssi readings also confirm the change in power. 
Then tried setting BW 125 and SF 12  works fine . Tx side datarate drops  to 80 bps and RX side occasionally gives crc errors (propably esp32 interrupt issue). So far checked and tested 4 modules with same results. 
I cant tell if these are fakes or some weird semtech ' s manufacturing-marketing process.