jgromes/RadioLib #1331: 3kHz receive packet frequency error on SX1262

jamcinnes2, 2024-11-25:
I'm creating a LoRa link using two SX1262 chips. The link is working fine. I'm seeing getFrequencyError() returning about 3000.0 Hz on every packet. (it varies from 2800 Hz to 3100 Hz). I did read in your documentation that this measurement is not supported on Sx126X. Should I try to tune it out or just ignore it?

lyusupov, 2024-11-25:
If your SX1262 design uses a regular crystal clock source - temperature and the crystal ageing are the factors of a frequency errors.

List of deviation measured for certain devices/modules:
https://github.com/lyusupov/SoftRF/wiki/Frequency-deviation

When the clock source is a TCXO with 0.2-2 PPM - the deviation typically stays within 1 kHz.

jgromes, 2024-11-25:
@jamcinnes2 

> I did read in your documentation that this measurement is not supported on Sx126X

Where exactly did you find this information? There is [this note/warning](https://jgromes.github.io/RadioLib/class_s_x126x.html#a068d709ffb79ae66d718ffb76ffae997) that this measurement is not a documented feature of the SX126x, is that what you meant?

> Should I try to tune it out or just ignore it?

Depends on your use case. If you are using LoRa, it can deal with frequency offset of up to 20% of the bandwidth. As long as your frequency offset is lower than that, you should have no issues - 3 kHz is safe. And if you want high accuracy (especially if you expect your device to work in large temperature ranges), you should use a TCXO as @lyusupov pointed out.

closed 2024-11-25 by jamcinnes2