jgromes/RadioLib #1334: Regional Parameters Typo in "LoRaWAN: Device setup on TTN" Wiki

Zep0, 2024-11-26:
Hi,
there is a newer version of the equivalent wiki page of the RadioLib, where the version number of the regional parameters is fixed. 

Instead of:

> RadioLib uses Regional Parameters 1.1 revision 1.1. 

it now says:

> RadioLib uses RP001 Regional Parameters 1.1 revision A.

Source: 
https://github.com/jgromes/RadioLib/wiki/LoRaWAN:-Device-setup-on-TTN
https://github.com/jgromes/RadioLib/blob/master/examples/LoRaWAN/LoRaWAN_Starter/notes.md

StevenCellist, 2024-11-26:
[Talking about consistency](https://github.com/TheThingsNetwork/lorawan-stack/issues/2784#issuecomment-648735723).... this is a proper mess. But at least I'll make these two match each other.

closed 2024-11-26 by StevenCellist

StevenCellist, 2024-11-26:
Actually, upon further inspection, I added a slight modification such that RadioLib properly corresponds to 1.1 revision B instead of revision A. All mentions in this repository have been updated to reflect this change, although it's practically identical to revision A.