jgromes/RadioLib #1344: Need method to get expected payload length from Reg 0x22 in SX1276 modem while in LoRa implicit header mode

snowrodeo, 2024-12-11:
There are two(?) ways to set the payload length in implicit header mode
setImplicit(length) and setReceived(timeout, flags, flags, length)

However there is no way to explicitly report on this length without using a SPI read from the register in God mode.

Request that a getImplicit method is added to return the number of payload bytes while in implicit more or 0 if not in implicit header mode

jgromes, 2024-12-14:
The same argument can be applied to any parameter that is being configured - that would be a very large amount of new methods to be added. There is currently an open issue for that (#622), though to be honest, I'm not sure that this is the right way forward.

Is there a specific scenario in which you would need to retrieve this value except for checking the radio settings when debugging?

jgromes, 2024-12-15:
It just opens a can of worms that I don't yet know how to resolve. Since this is only for debugging, we will stick with only having this accessible via godmode for now. Also #1345 has been resolved properly.

closed 2024-12-15 by jgromes