jgromes/RadioLib #1345: Conflicting ways to set payload length in implicit header mode with SX127X

snowrodeo, 2024-12-11:
To set an SX127x device to use implicit headers, you need to use
setImplicit(payloadLength)

However, when putting the device into receive mode, you need to also set the payload length or it will default to 0 and overwrite the previously set value

Request that setReceive use the value from setImplicit instead of requiring an arg

jgromes, 2024-12-14:
This is indeed a bit of an oversight - I think we can handle this similarly to how `startReceive` behaves in exaplicit mode: If the length is set to 0, it will use the previously configured implicit length. Otherwise, the length will be overwritten. Will this match your use case?

snowrodeo, 2024-12-15:
That would work great.  Thanks!

jgromes, 2024-12-15:
Implemented in the commit above.

closed 2024-12-15 by jgromes