jgromes/RadioLib #1349: LR1121 can not reach 22dBm output power

klabshowroom, 2024-12-14:
I use a LR1121 evaluate board from Semtech, it can work for 14 dBm power output.

When I set the power to be 22 dBm, the output power become 0 dBm.

The rf switch table is as:

static const Module::RfSwitchMode_t rfswitch_table[] = {
  // mode              DIO5(V1)  DIO6(V2) 
  { LR11x0::MODE_STBY,   { LOW,  LOW  } },  //shutdown
  { LR11x0::MODE_RX,     { HIGH, LOW  } },  //J2
  { LR11x0::MODE_TX,     { HIGH, HIGH } },  //J3, RFO_LP_LF
  { LR11x0::MODE_TX_HP,  { LOW,  HIGH } },  //J1, RFO_HP_LF
  { LR11x0::MODE_TX_HF,  { LOW,  LOW  } },
  { LR11x0::MODE_GNSS,   { LOW,  LOW  } },
  { LR11x0::MODE_WIFI,   { LOW,  LOW  } },
  END_OF_MODE_TABLE,
};

is there anything wrong for my setting?

[image]

lyusupov, 2024-12-14:
[image]

jgromes, 2024-12-14:
14 dBm is the upper limit of the low-power PA. RadioLib selects the PA based on the requested output power, or if you force it with the second argument of `setOutputPower(int8_t power, bool forceHighPower)`. If you go above 14 dBm, the lbirary will have no choice but to use the high-power PA. So from your description, I'm guessing that you have something wrong with the RF switch configuration for the high-power PA since it looks like the power doesn't make it to the antenna port (assuming that's where you are measuring it).

lyusupov, 2024-12-14:
@jgromes 

@klabshowroom uses LR1121 reference design board ( LR1121DVK1TxKS 868_915MHz DevKit ) from Semtech

[image]

<br>
<br>

Complete design files are available at this location:
https://www.semtech.com/products/wireless-rf/lora-connect/lr1121dvk1tbks#documentation

Not sure that RF switch table provided by @klabshowroom is 100% valid - but it looks quite realistic.

jgromes, 2024-12-15:
> Not sure that RF switch table provided by @klabshowroom is 100% valid - but it looks quite realistic.

I'm not sure either, because it wasn't fully posted - the pin definition table is missing. Since the reference design uses DIO5 and DIO6 as the switch pins, the `RADIOLIB_LR11X0_DIO5, RADIOLIB_LR11X0_DIO6` macros have to be used, as shown in the examples.

Other than that, the table looks OK.

klabshowroom, 2024-12-18:
Hi All, thanks for response, and I am new here, so don't know how will be better to post documents . I will try again, and report the  result later. 

Thanks again.

klabshowroom, 2024-12-18:
> I'm not sure either, because it wasn't fully posted - the pin definition table is missing. Since the reference design uses DIO5 and DIO6 as the switch pins, the `RADIOLIB_LR11X0_DIO5, [...]

in LR11x0.h, there is definition for these two pins:
#define RADIOLIB_LR11X0_DIO5                                    (0)
#define RADIOLIB_LR11X0_DIO6                                    (1)

I try to modify these pin number to be 
#define RADIOLIB_LR11X0_DIO5                                    (3)
#define RADIOLIB_LR11X0_DIO6                                    (4)

and find there is no change for the result. 

it seems the pin definition can not change the result, and how to make these two pins work?

jgromes, 2024-12-18:
> in LR11x0.h, there is definition for these two pins:

That is not "definition for these two pins", those are internal macros and must be left unchanged. Nothing in the lirbary is supposed to be modified by the user, except for some build options. I also don't know why would you choose to change it to 3 and 4.

Please read my previous post again. The switch table you have posted is incomplete. You also need to define the pins your switch is using, and provide that to the library. In the LR11x0_Transmit_Blocking example, this is shown as `rfswitch_dio_pins` array that contains the pin numbers and a call to `radio.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);` during `setup` which sets up the RF switch pins and the switch table. From what you've posted, I don't know if you are doing that. That's why the issue template asks for the *complete* code.

klabshowroom, 2024-12-19:
@jgromes Thanks for your information. 

Finally, I figure out the problem, but don't know the root cause. 

in the example, it show:
 // set RF switch control configuration
 // this has to be done prior to calling begin()
 radio.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);

So, I put the command before raiod.begin()
It will not let dio5, dio6 to work. 

I put this command after radio.begin() as:
int state = radio.begin(920.3, 125.0, 9, 7, RADIOLIB_LR11X0_LORA_SYNC_WORD_PRIVATE, 22, 8, 0);
radio.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);

and the output power could reach 22dBm now. 

I don't know why, just report my status.

klabshowroom, 2024-12-19:
By the way, 

void LR11x0::setRfSwitchTable(const uint32_t (&pins)[Module::RFSWITCH_MAX_PINS], const Module::RfSwitchMode_t table[]) {
  // find which pins are used
  uint8_t enable = 0;
  for(size_t i = 0; i < Module::RFSWITCH_MAX_PINS; i++) {
    if((pins[i] == RADIOLIB_NC) || (pins[i] > RADIOLIB_LR11X0_DIO10)) {
      continue;
    }
    enable |= 1UL << pins[i];
  }

this command 
"enable |= 1UL << pins[i]; "
seems not correct, since we need to use DIO5, DIO6 (bit0, bit1)
So, the enable should be 3, but the code result is 12
it seems the bit order is not correct.

jgromes, 2024-12-19:
This has been reported in  #1295 and fixed about 2 moths ago in 0156c9004efee62dca24aa5a4e012775f233d3d4. It is released in 7.1.0 so I'm guessing that you have some outdated version of the library. Keyword there being guessing since you never reported your version.

Had you filled in the issue template we would have had all the information we needed from the start ...

closed 2024-12-19 by jgromes

Llgok, 2025-04-22:
@jgromes This issue is indeed worth paying attention to because I encountered the same problem. Placing `radio.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);` before `radio.begin();` does not achieve the 22dBm power output. Only by placing `radio.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);` after `radio.begin();` does it work properly. The version of RadioLib I am using is 7.1.2.

[image]

[image]

jgromes, 2025-04-22:
@Llgok this issue has been resolved, see my previous comment.