jgromes/RadioLib #1353: SX126x in FSK works with v6.6.0, but not with v7.0.0 or v7.1.0

matthias-bs, 2024-12-16:
**Describe the bug**
I am using an SX1262 in FSK mode in a setup very close to [SX126x_Receive_Interrupt.ino](https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_Receive_Interrupt/SX126x_Receive_Interrupt.ino).

* 868.3 MHz
* Fixed packet length
* CRC filtering disabled
* Preamble: 32 bits
* Sync word: 2 bytes

## RadioLib v6.6.0

Works as expected, i.e. receives messages starting with `0xD4`:
```
08:49:55:644 -> [   285][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 =========== Before Setup Start ===========
08:49:55:644 -> Chip Info:
08:49:55:646 -> ------------------------------------------
08:49:55:646 ->   Model             : ESP32-S3
08:49:55:646 ->   Package           : 0
08:49:55:650 ->   Revision          : 0.02
08:49:55:651 ->   Cores             : 2
08:49:55:653 ->   CPU Frequency     : 240 MHz
08:49:55:656 ->   XTAL Frequency    : 40 MHz
08:49:55:659 ->   Features Bitfield : 0x00000012
08:49:55:661 ->   Embedded Flash    : No
08:49:55:663 ->   Embedded PSRAM    : No
08:49:55:666 ->   2.4GHz WiFi       : Yes
08:49:55:670 ->   BT Low Energy     : Yes
08:49:55:668 ->   Classic BT        : No
08:49:55:673 ->   IEEE 802.15.4     : No
08:49:55:676 -> ------------------------------------------
08:49:55:678 -> INTERNAL Memory Info:
08:49:55:682 -> ------------------------------------------
08:49:55:686 ->   Total Size        :   402284 B ( 392.9 KB)
08:49:55:690 ->   Free Bytes        :   372748 B ( 364.0 KB)
08:49:55:694 ->   Allocated Bytes   :    24488 B (  23.9 KB)
08:49:55:701 ->   Minimum Free Bytes:   367700 B ( 359.1 KB)
08:49:55:703 ->   Largest Free Block:   335860 B ( 328.0 KB)
08:49:55:706 -> ------------------------------------------
08:49:55:707 -> Flash Info:
08:49:55:711 -> ------------------------------------------
08:49:55:715 ->   Chip Size         :  4194304 B (4 MB)
08:49:55:719 ->   Block Size        :    65536 B (  64.0 KB)
08:49:55:723 ->   Sector Size       :     4096 B (   4.0 KB)
08:49:55:727 ->   Page Size         :      256 B (   0.2 KB)
08:49:55:729 ->   Bus Speed         : 80 MHz
08:49:55:732 ->   Bus Mode          : QIO
08:49:55:735 -> ------------------------------------------
08:49:55:737 -> Partitions Info:
08:49:55:741 -> ------------------------------------------
08:49:55:748 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
08:49:55:755 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
08:49:55:765 ->                app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
08:49:55:771 ->                app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
08:49:55:778 ->              spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
08:49:55:786 ->            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
08:49:55:789 -> ------------------------------------------
08:49:55:791 -> Software Info:
08:49:55:795 -> ------------------------------------------
08:49:55:798 ->   Compile Date/Time : Dec 16 2024 08:42:46
08:49:55:803 ->   Compile Host OS   : linux
08:49:55:806 ->   ESP-IDF Version   : v5.1.4-972-g632e0c2a9f-dirty
08:49:55:808 ->   Arduino Version   : 3.0.7
08:49:55:812 -> ------------------------------------------
08:49:55:813 -> Board Info:
08:49:55:817 -> ------------------------------------------
08:49:55:820 ->   Arduino Board     : LILYGO_T3S3_SX1262
08:49:55:824 ->   Arduino Variant   : lilygo_t3_s3_sx1262
08:49:55:835 ->   Arduino FQBN      : esp32:esp32:lilygo_t3s3:PSRAM=disabled,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=none,Revision=Radio_SX1262
08:49:55:855 -> ============ Before Setup End ============
08:49:55:955 -> [  1078][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 19 already has type USB_DM (45) with bus 0x3fc94ea8
[... 482 lines cut ...]
    return decode_res;
}

void loop()
{
    // Tries to receive radio message (non-blocking) and to decode it.
    // Timeout occurs after a small multiple of expected time-on-air.
    getMessage();

    delay(100);
} // loop()
```

</p>
</details> 

**Expected behavior**
The sketch should work with RadioLib > v6.6.0.

**Additional info (please complete):**
 - MCU: ESP32, LilyGo T3-S3 Radio-SX1262
 - Link to Arduino core: https://github.com/espressif/arduino-esp32 v3.0.7
 - Wireless module type SX1262
 - Arduino IDE version  2.3.4
 - Library version RadioLib 7.0.0, 7.1.0

matthias-bs, 2024-12-16:
[devttyACM0_2024_12_16.11.56.01.074_Radiolib_6.6.0_Debug.txt](https://github.com/user-attachments/files/18149284/devttyACM0_2024_12_16.11.56.01.074_Radiolib_6.6.0_Debug.txt)
[devttyACM0_2024_12_16.11.59.29.380_Radiolib_7.0.0_Debug.txt](https://github.com/user-attachments/files/18149288/devttyACM0_2024_12_16.11.59.29.380_Radiolib_7.0.0_Debug.txt)
[devttyACM0_2024_12_16.12.05.10.812_Radiolib_7.1.0_Debug.txt](https://github.com/user-attachments/files/18149292/devttyACM0_2024_12_16.12.05.10.812_Radiolib_7.1.0_Debug.txt)

jgromes, 2024-12-16:
See #1350 - this might be the same issue, though to solution to it is probably not complete yet.

StevenCellist, 2024-12-16:
As the attention-paying master @jgromes found the footnote showing why the configured 32-bit preamble of this setup is incompatible with the 2-byte syncword, I think we can close this one and track it in #1350 😉 
Either way, jinx on the problem and kudos for the well-filed issue.

closed 2024-12-16 by StevenCellist

tonbor, 2024-12-25:
With radiolib 7.1.1 all ok