jgromes/RadioLib #1362: [LR1121] LoRaWAN join fails with RADIOLIB_ERR_NO_JOIN_ACCEPT (-1116)

matthias-bs, 2024-12-25:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Troubleshooting Guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide) Wiki page and an extensive [API documentation](https://jgromes.github.io/RadioLib/). If you are seeing an error code, we have [online status code decoder](https://radiolib-org.github.io/status_decoder/decode.html).
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

<details><summary>Sketch that is causing the module fail</summary>
<p>

```c++
/*
  RadioLib LoRaWAN Starter Example

  ! Please refer to the included notes to get started !

  This example joins a LoRaWAN network and will send
  uplink packets. Before you start, you will have to
  register your device at https://www.thethingsnetwork.org/
  After your device is registered, you can run this example.
  The device will join the network and start uploading data.

  Running this examples REQUIRES you to check "Resets DevNonces"
  on your LoRaWAN dashboard. Refer to the network's 
  documentation on how to do this.

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/

  For LoRaWAN details, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/LoRaWAN

*/

#include "config.h"
SPIClass *spi;
//spi.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);
#if defined(ARDUINO_LILYGO_T3S3_LR1121)
LR1121 radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY);
#elif defined(ARDUINO_LILYGO_T3S3_SX1262)
SX1262 radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY);
#elif defined(ARDUINO_LILYGO_T3S3_SX1276)
SX1276 radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY);
#else
#pragma message("Unsupported board")
#endif
LoRaWANNode *node;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(1000);  // Give time to switch to the serial monitor
  Serial.println(F("\nSetup ... \n"));

  spi = new SPIClass(SPI);
  spi->begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);
  radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY, *spi);
  node = new LoRaWANNode(&radio, &Region, subBand);

  Serial.println(F("Initialise the radio"));
  int16_t state = radio.begin();
  debug(state != RADIOLIB_ERR_NONE, F("Initialise radio failed"), state, true);

  // Setup the OTAA session information
  state = node->beginOTAA(joinEUI, devEUI, nwkKey, appKey);
  debug(state != RADIOLIB_ERR_NONE, F("Initialise node failed"), state, true);

  Serial.println(F("Join ('login') the LoRaWAN Network"));
  state = node->activateOTAA();
  debug(state != RADIOLIB_LORAWAN_NEW_SESSION, F("Join failed"), state, true);

  Serial.println(F("Ready!\n"));
}

void loop() {
  Serial.println(F("Sending uplink"));

  // This is the place to gather the sensor inputs
  // Instead of reading any real sensor, we just generate some predictable values
  static uint8_t value1 = 0;
  static uint16_t value2 = 0;

  // Build payload byte array
  uint8_t uplinkPayload[3];
  uplinkPayload[0] = value1++;
  uplinkPayload[1] = highByte(value2);   // See notes for high/lowByte functions
  uplinkPayload[2] = lowByte(value2);
  value2 += 2;
  
  // Perform an uplink
  int16_t state = node->sendReceive(uplinkPayload, sizeof(uplinkPayload));    
  debug(state < RADIOLIB_ERR_NONE, F("Error in sendReceive"), state, false);

  // Check if a downlink was received 
  // (state 0 = no downlink, state 1/2 = downlink in window Rx1/Rx2)
  if(state > 0) {
    Serial.println(F("Received a downlink"));
  } else {
    Serial.println(F("No downlink received"));
  }

  Serial.print(F("Next uplink in "));
  Serial.print(uplinkIntervalSeconds);
  Serial.println(F(" seconds\n"));
  
  // Wait until next uplink - observing legal & TTN FUP constraints
  delay(uplinkIntervalSeconds * 1000UL);  // delay needs milli-seconds
}

```

</p>
</details> 

**Hardware setup**
I have got
1. [Lilygo T3 S3 SX1262](https://lilygo.cc/products/t3s3-v1-0?variant=42586879721653) - works as expected
2. [Lilygo T3 S3 LR1121](https://lilygo.cc/products/t3-s3-lr1121) - fails to join

According to [Lilygo](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/213) the schematic of the two boards is basically the same. Also the LoRa radio pin definitions in https://github.com/espressif/arduino-esp32/tree/master/variants/lilygo_t3_s3_sx1262 and https://github.com/espressif/arduino-esp32/tree/master/variants/lilygo_t3_s3_lr1121 are identical.

Furthermore

**Debug mode output**
Enable the appropriate [debug levels](https://github.com/jgromes/RadioLib/wiki/Debug-mode) and paste the Serial monitor output here. For debugging protocols, enable `RADIOLIB_DEBUG_PROTOCOL`. For debugging issues with the radio module itself, enable `RADIOLIB_DEBUG_SPI`.

<details><summary>Debug mode output</summary>
<p>

```
paste the debug output here
```

</p>
</details> 

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0 or git hash]

StevenCellist, 2024-12-25:
We're lacking the basic & protocol debug output here - and while you edit that into your post, please remove the unnecessary template items from your issue.

matthias-bs, 2024-12-25:
@StevenCellist You are too quick! ;-)

StevenCellist, 2024-12-25:
A couple of things:
1. Do you see any action on the (TTN) gateway console? Is the JoinRequest received, is a JoinAccept scheduled?
2. Have you tried LoRa P2P / pingpong between the LR1121 and SX1262 at somewhat similar-to-LW PHY settings? With what results?

matthias-bs, 2024-12-25:
[config.h.zip](https://github.com/user-attachments/files/18246907/config.h.zip)

matthias-bs, 2024-12-25:
1. I did not see any action on the TTN console.
2. No, not yet.

matthias-bs, 2024-12-25:
The pingpong example shows that the LR1121 module receives, but does not dend (even if it says so). I will provide an SPI log later on.

jgromes, 2024-12-26:
Just to check we didn't accidentally regress somewhere, I checked with a Seeed WM1110, and it does transmit LoRa (checked with an SDR). My guess is that this is caused by the RF switch, since it can be a lot more complicated on LR11xx than it is on SX126x.

matthias-bs, 2024-12-26:
I could not find any RF switch in the schematic. Did I miss anything?

jgromes, 2024-12-26:
There has to be one - LR1121 has a separate input, low power output and high power output RF ports as can be seen in its block diagram below (from LR1121 datasheet).

[image]

matthias-bs, 2024-12-26:
So it is probably inside the LoRa module!

Meanwhile I tried FSK mode: same result - receiving works but transmitting does not.

jgromes, 2024-12-26:
> So it is probably inside the LoRa module!

Yes, it needs to be there. I would also guess it uses some of the LR11xx DIO pins for control, but that's something for the manufacturer to confirm.

matthias-bs, 2024-12-27:
**Summary**

This issue was not caused by RadioLib, but by missing code for controlling the RF switch in my application.

The RF switch is not visible in the [LilyGo T3 S3 LR1121 schematic](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/213), because it is integrated in the LR1121 RF module.

Additionally, the LilyGo T3 S3 boards use non-standard SPI pins for controlling the radio chip.

<details><summary>Working LoRaWAN_Starter.ino example for the LilyGo T3S3 boards (variants SX1262, SX1276 and LR1121)</summary>

<p>
```c++
/*
  RadioLib LoRaWAN Starter Example

  ! Please refer to the included notes to get started !

  This example joins a LoRaWAN network and will send
  uplink packets. Before you start, you will have to
  register your device at https://www.thethingsnetwork.org/
  After your device is registered, you can run this example.
  The device will join the network and start uploading data.

  Running this examples REQUIRES you to check "Resets DevNonces"
  on your LoRaWAN dashboard. Refer to the network's 
  documentation on how to do this.

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/

  For LoRaWAN details, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/LoRaWAN

*/

#include "config.h"
SPIClass *spi;

#if defined(ARDUINO_LILYGO_T3S3_LR1121)
LR1121 radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY);
#elif defined(ARDUINO_LILYGO_T3S3_SX1262)
SX1262 radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY);
#elif defined(ARDUINO_LILYGO_T3S3_SX1276)
SX1276 radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY);
#else
#pragma message("Unsupported board")
#endif

#if defined(ARDUINO_LILYGO_T3S3_LR1121)
  // Note: The LR1121 has an RF switch, which in integrated in the LoRa module
  //
  // For schematic, see
  // https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/213
  // For code example, see
  // https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/T3S3Factory/T3S3Factory.ino#L381
  static const uint32_t rfswitch_dio_pins[] = {
    RADIOLIB_LR11X0_DIO5, RADIOLIB_LR11X0_DIO6,
    RADIOLIB_NC, RADIOLIB_NC, RADIOLIB_NC
  };

  static const Module::RfSwitchMode_t rfswitch_table[] = {
    // mode                  DIO5  DIO6
    { LR11x0::MODE_STBY,   { LOW,  LOW  } },
    { LR11x0::MODE_RX,     { HIGH, LOW  } },
    { LR11x0::MODE_TX,     { LOW,  HIGH } },
    { LR11x0::MODE_TX_HP,  { LOW,  HIGH } },
    { LR11x0::MODE_TX_HF,  { LOW,  LOW  } },
    { LR11x0::MODE_GNSS,   { LOW,  LOW  } },
    { LR11x0::MODE_WIFI,   { LOW,  LOW  } },
    END_OF_MODE_TABLE,
  };
#endif

LoRaWANNode *node;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(1000);  // Give time to switch to the serial monitor
  Serial.println(F("\nSetup ... \n"));

  spi = new SPIClass(SPI);
  spi->begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);
  radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_BUSY, *spi);
  node = new LoRaWANNode(&radio, &Region, subBand);

  Serial.println(F("Initialise the radio"));
  int16_t state = radio.begin();
  debug(state != RADIOLIB_ERR_NONE, F("Initialise radio failed"), state, true);

  #if defined(ARDUINO_LILYGO_T3S3_LR1121)
  radio.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);

    // LR1121 TCXO Voltage 2.85~3.15V
  radio.setTCXO(3.0);
  #endif

  // Setup the OTAA session information
  state = node->beginOTAA(joinEUI, devEUI, nwkKey, appKey);
  debug(state != RADIOLIB_ERR_NONE, F("Initialise node failed"), state, true);

  Serial.println(F("Join ('login') the LoRaWAN Network"));
  state = node->activateOTAA();
  debug(state != RADIOLIB_LORAWAN_NEW_SESSION, F("Join failed"), state, true);

  Serial.println(F("Ready!\n"));
}

void loop() {
  Serial.println(F("Sending uplink"));

  // This is the place to gather the sensor inputs
  // Instead of reading any real sensor, we just generate some predictable values
  static uint8_t value1 = 0;
  static uint16_t value2 = 0;

  // Build payload byte array
  uint8_t uplinkPayload[3];
  uplinkPayload[0] = value1++;
  uplinkPayload[1] = highByte(value2);   // See notes for high/lowByte functions
  uplinkPayload[2] = lowByte(value2);
  value2 += 2;
  
  // Perform an uplink
  int16_t state = node->sendReceive(uplinkPayload, sizeof(uplinkPayload));    
  debug(state < RADIOLIB_ERR_NONE, F("Error in sendReceive"), state, false);

  // Check if a downlink was received 
  // (state 0 = no downlink, state 1/2 = downlink in window Rx1/Rx2)
  if(state > 0) {
    Serial.println(F("Received a downlink"));
  } else {
    Serial.println(F("No downlink received"));
  }

  Serial.print(F("Next uplink in "));
  Serial.print(uplinkIntervalSeconds);
  Serial.println(F(" seconds\n"));
  
  // Wait until next uplink - observing legal & TTN FUP constraints
  delay(uplinkIntervalSeconds * 1000UL);  // delay needs milli-seconds
}
```
</p>
</details>

See [Xinyuan-LilyGO/LilyGo-LoRa-Series](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series) for further examples, also based on RadioLib.

Thanks @jgromes and @StevenCellist for pointing me in the right direction.

closed 2024-12-27 by matthias-bs

jgromes, 2024-12-27:
@matthias-bs glad you were able to get it working! 

On a side note, the definitions from the example would be better suited in the [RadioBoards repo](https://github.com/radiolib-org/RadioBoards) as a contributed board definition. It already supports [custom SPI and RF switch definition](https://github.com/radiolib-org/RadioBoards/blob/main/src/maintained/SeeedStudio/WM1110.h). PRs are always welcome ;)