jgromes/RadioLib #1363: Visionmaster t190 cannot get lora to init error code -2

jamesgbahr, 2024-12-25:
I cannot get the lora to init i always get error code -2 when trying to set any of the options, eg: freq, band, etc could my lora module itself be bad?

jamesgbahr, 2024-12-26:
#include <TFT_eSPI.h> // Include the TFT library
#include <RadioLib.h> // Include the RadioLib library for LoRa

// Support for 1.9" 320x170 Heltec T190 IPS TFT Display
#define USER_SETUP_ID 190

#define ST7789_DRIVER
#define USE_HSPI_PORT
#define TFT_RGB_ORDER TFT_BGR

#define TFT_WIDTH 170
#define TFT_HEIGHT 320

#define TFT_BL 17
#define TFT_BACKLIGHT_ON HIGH

#define TFT_MOSI 48
#define TFT_SCLK 38
#define TFT_CS 39
#define TFT_DC 47
#define TFT_RST 40

#define SPI_FREQUENCY 27000000

// LoRa Pin Definitions for SX1262
#define LORA_NSS 8    // SPI Chip Select for LoRa
#define LORA_SCK 9    // SPI Clock
#define LORA_MISO 11  // SPI MISO
#define LORA_MOSI 10  // SPI MOSI
#define LORA_RST 12   // LoRa Reset
#define LORA_BUSY 13  // LoRa Busy
#define LORA_DIO1 14  // LoRa DIO1 for interrupts

// Create LoRa and TFT objects
SX1262 radio = new Module(LORA_NSS, LORA_DIO1, LORA_RST, LORA_BUSY);
TFT_eSPI tft = TFT_eSPI();

void setup() {
    // Initialize serial communication for debugging
    Serial.begin(115200);
    Serial.println("Initializing SX1262 LoRa and TFT");

    // Initialize TFT Display
    pinMode(7, OUTPUT); // Power for the TFT display
    digitalWrite(7, LOW);

    pinMode(TFT_BL, OUTPUT); // Backlight control
    digitalWrite(TFT_BL, HIGH);

    tft.init();
    tft.setRotation(1);
    tft.fillScreen(TFT_BLACK);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.setTextSize(2);
    tft.setCursor(10, 10);
    tft.println("Initializing LoRa...");

    // Initialize LoRa
    if (!radio.begin()) {
        Serial.println("LoRa initialization failed!");
        tft.setCursor(10, 40);
        tft.println("LoRa Init Failed!");
        while (true); // Halt if initialization fails
    }

    // Configure LoRa parameters
    if (radio.setFrequency(433.0) != RADIOLIB_ERR_NONE) {
        Serial.println("Failed to set frequency.");
        tft.setCursor(10, 60);
        tft.println("Freq Set Failed!");
    }

    if (radio.setBandwidth(250.0) != RADIOLIB_ERR_NONE) {
        Serial.println("Failed to set bandwidth.");
        tft.setCursor(10, 80);
        tft.println("BW Set Failed!");
    }

    if (radio.setSpreadingFactor(9) != RADIOLIB_ERR_NONE) {
        Serial.println("Failed to set spreading factor.");
        tft.setCursor(10, 100);
        tft.println("SF Set Failed!");
    }

    if (radio.setOutputPower(5) != RADIOLIB_ERR_NONE) {
        Serial.println("Failed to set output power.");
        tft.setCursor(10, 120);
        tft.println("PWR Set Failed!");
    }

    Serial.println("LoRa Initialized Successfully!");
    tft.setCursor(10, 140);
    tft.println("LoRa Ready!");
}

void loop() {
    // Transmit a test message
    static int counter = 0;
    String message = "Hello LoRa " + String(counter++);
    Serial.print("Transmitting: ");
    Serial.println(message);

    tft.fillRect(0, 180, TFT_WIDTH, 20, TFT_BLACK); // Clear previous message
    tft.setCursor(10, 180);
    tft.print("TX: ");
    tft.print(message);

    int status = radio.transmit(message.c_str());
    if (status == RADIOLIB_ERR_NONE) {
        Serial.println("Transmission successful!");
        tft.setCursor(10, 200);
        tft.print("TX OK");
    } else {
        Serial.print("Transmission failed. Error code: ");
        Serial.println(status);
        tft.setCursor(10, 200);
        tft.print("TX Fail");
    }

    delay(2000); // Delay for next transmission
}

jamesgbahr, 2024-12-26:
I should add im getting readings when i flash meshtasic signaling that the radio is working on meshtastic but not with radio lib here is the pinout meshtastic is using and the pinout directly from heltec https://github.com/meshtastic/firmware/commit/9ec7dbd69504581c789d865624761fa841a6e7c4#diff-3b755673736dd249b8d853ca58c35050f060224e273801f7af8100fc86602ab3
[image]

jamesgbahr, 2024-12-26:
I just noticed its returning RLB_DBG: Expected string: SX1261 when it should be looking for a 1262 not sure why it would be doing that?

StevenCellist, 2024-12-26:
Are you really, really sure you have a T190 that has the LoRa module installed?

jgromes, 2024-12-26:
Couple of things:

1. We have an issue template that asks for specific information and even helps you format them so that the post is not a kilometer in length. I have edited your previous posts so that code is formatted as code and debug log is collapsible. Please take a look at those edits.
2. I don't know what you're expeciting to happen here: `if (!radio.begin())` - methods in this library return `RADIOLIB_ERR_NONE`, which is 0 (as a well-behaved API should). I suspect you have copy-pasted code from some other library. Please take a look at our extensive collection of examples and start with those. Only add additional stuff (such as the TFT) once you are confident you can control the radio.
3. Meshtastic uses RadioLib. If it works, then so does this library and the issue is in your code.

Most importantly though, you seem to have completely skipped all of the documentation we have. That makes me quite sad - I've spent a lot of time on it. We have a Wiki, full API reference, a troubleshooting guide and even online tools for decoding the status codes and debug log. All of it linked from the readme - I really don't know what else to do to get people to read it.

I'm about 90% confident you have an incorrect SPI interface. More than one of the wiki pages describes this, so I'm confident you can find it on your own.

closed 2024-12-26 by jgromes

jamesgbahr, 2024-12-26:
I just ran the exact code from your examples and got the same result in debug, so its not my code 
RLB_DBG: 00000320: 00 00 00 00 00 00 00 00 00 00 00 00RLB_DBG: SX126x not found! (5 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
RLB_DBG: Expected string: SX1261
RLB_DBG: SX126x not found! (6 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
RLB_DBG: Expected string: SX1261
RLB_DBG: SX126x not found! (7 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
RLB_DBG: Expected string: SX1261
RLB_DBG: SX126x not found! (8 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
RLB_DBG: Expected string: SX1261
RLB_DBG: SX126x not found! (9 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
RLB_DBG: Expected string: SX1261
RLB_DBG: SX126x not found! (10 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
RLB_DBG: Expected string: SX1261
RLB_DBG: No SX126x found!
failed, code -2

jamesgbahr, 2024-12-26:
yes it was spi issue
 LoRa module reset.
RLB_DBG:
RadioLib Info
Version:  "7.1.1.0"
Platform: "ESP32"
Compiled: "Dec 25 2024" "23:56:15"
RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
RLB_DBG:
RLB_DBG: M      SX126x
LoRa module initialized successfully!