jgromes/RadioLib #1366: CC1101 Parameters issue, receiving packets only with invalid bandwidth parameter set

jmatilai, 2024-12-30:
<details><summary>Using _CC1101_Receive_Interrupt_ and _CC1101_Transmit_Interrupt_, any bandwidth setting does not work, except setting unacceptable value such as "100"</summary>

<p>
I'm unable to get the module receive anything with the example templates, except one empty ghost message after booting. However by accident I changed the bandwidth filter value and by setting "unacceptable" value such as "100" I can get messages sent/received. With 6.6.0. this works, I get -104 [RADIOLIB_ERR_INVALID_RX_BANDWIDTH] error at start, but after this it works fine. With 7.1.1. I get -104 error also from startReveive, which prevents this "workaround". I have tried few acceptable values for bandwidth filter based on datasheet with no avail.

```
[CC1101] Data:          Hello World! #512
[CC1101] RSSI:          -30.00 dBm
[CC1101] LQI:           6
[CC1101] Received packet!
[CC1101] Data:          Hello World! #513
[CC1101] RSSI:          -30.00 dBm
[CC1101] LQI:           5
```

I use platformio in VScode with 
* RadioLib 6.6.0, 
* SPI 1.1.0 and 
* Arduino_Core_STM32 seems to be 2.8.1. 

I tried also ELECHOUSE based library https://github.com/LSatan/SmartRC-CC1101-Driver-Lib, which works, but the STM32 crashes after sending for a while. I did not investigate that further since I found your library, but now this made me think is there something wrong with my hardware?

RX END (only modified pins and radio.begin):
```c++
/*
  RadioLib CC1101 Receive with Interrupts Example

  This example listens for FSK transmissions and tries to
  receive them. Once a packet is received, an interrupt is
  triggered.

  To successfully receive data, the following settings have to be the same
  on both transmitter and receiver:
  - carrier frequency
  - bit rate
  - frequency deviation
  - sync word

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration#cc1101

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/
*/
#include <SPI.h>
#include <Arduino.h>
// include the library
#include <RadioLib.h>

#define CC1101_GDO0 PB0
#define CC1101_CS PA4
#define CC1101_SCK PA5
#define CC1101_MOSI PA7
#define CC1101_MISO PA6
#define CC1101_GDO2 PA3
#define LED_ONBOARD PC13

// CC1101 has the following connections:
// CS pin:    10
// GDO0 pin:  2
// RST pin:   unused
// GDO2 pin:  3
CC1101 radio = new Module(CC1101_CS, CC1101_GDO0, RADIOLIB_NC, CC1101_GDO2);

// or detect the pinout automatically using RadioBoards
// https://github.com/radiolib-org/RadioBoards
/*
#define RADIO_BOARD_AUTO
#include <RadioBoards.h>
Radio radio = new RadioModule();
[... 260 lines cut ...]
  }
}
```

</p>
</details> 

**Hardware setup**
Both

* "Bluepill" STM32F103C8T6
* Unknown brand generic CC1101 module "V2.0 433MHZ" text, blue with 2x4 pin header and SMA antenna connection
* connections CC1101 - STM32: GDO0 PB0, CS PA4, SCK PA5, MOSI PA7, MISO PA6, GDO2 PA3
* some SMA antenna with unknown properties
* Mini560 3.3V step down as supply

**Debug mode output**

<details><summary>Debug mode output</summary>
<p>

</p>
</details> 

**Additional info (please complete):**

jmatilai, 2024-12-30:
After reading [#1261](https://github.com/jgromes/RadioLib/issues/1261) I tried 5.7.0 and replaced

```
  //radio.setPacketReceivedAction(setFlag);
  radio.setGdo0Action(setFlag, RISING);
```
and
```
  //radio.setPacketSentAction(setFlag);
  radio.setGdo2Action(setFlag,FALLING);
```

TX seems to work, but RX node seems to get interrupt continuously

```
CRC error!
[CC1101] Received packet!
[CC1101] Data:          Hllo World! #2
[CC1101] RSSI:          -41.50 dBm
[CC1101] LQI:           49
CRC error!
[CC1101] Received packet!
[CC1101] Data:          Hllo World! #2A▒D▒˞▒
                                            5▒,p▒5i▒▒\▒7▒-▒▒h▒▒(:n▒▒S/▒▒&▒▒R▒s▒]▒BFHllo World! #2A▒D▒˞▒

[CC1101] RSSI:          -47.50 dBm
[CC1101] LQI:           94
CRC error!
CRC error!
[CC1101] Received packet!
[CC1101] Data:          Helo World! #2
[CC1101] RSSI:          -41.50 dBm
[CC1101] LQI:           49

```

It receives packets however every now and then.

jgromes, 2024-12-30:
This is interesting, I guess it is possible we broke the Rx bandwidth configuration at some point. Though quickly checking with ESP32+CC1101, the configuration seems to be correct, at least for the default 58.0 kHz value. 

> I tried 5.7.0 and replaced

I would suggest sticking to the latest release, or better yet the master, there were quite a few bufixes to CC1101 since 5.7.0. 6.6.0 is also quite outdated.

jmatilai, 2024-12-30:
Did some more testing, now with master, doing a ping pong test (A transmitting at interval, B transmitting response to A), packet loss with different bandwidths

* 58: 100%
* 102: 100%
* 135: 33..90%
* 162: 0..87%
* 206: >92%
* 270: 100%

Settings used (only changed BW from these):
`radio.begin(434.0, 4.8, 25.4, 135.0, 10, 16);`

I have not calibrated the modules or changed anything. But I expected that is not required with low bit rates. Should I do something like that?

jgromes, 2025-01-02:
Seems like there's definitely something weird going on. I'll try to get a test setup with transmitter and receiver up and running.

No calibration should be necessary here. The default bitrate is rather slow, the signal is narrow and the receiver bandwidth should be wide enough.

jgromes, 2025-03-16:
Unfortunately I was not able to replicate the packet loss issues reported here (at comparable power level, the CC1101 receiver reports -45 to -46 dBm). There is the occasional lost packet, but the overall loss rate was low (2-5%), and independent of the configured bandwidth. It should be noted that I measured with SX126x as the transmitter and CC1101 only receiving.

Since I was not able to reproduce this issue, I will close this thread - feel free to reopen later if more information becomes available.

closed 2025-03-16 by jgromes