jgromes/RadioLib #1368: SX1278 setFrequency() does not work anymore in 7.1.1

BBluemer, 2025-01-03:
Hello,
with my setup and minimal working example code (see below) the frequency setting of the SX1278 does not work anymore with library version 7.1.1. Using library version 7.1.0 it works as expected. 
The Debug output below is identical with both Library Versions.
Thanks for any help.

<details><summary>Debug mode output</summary>
<p>

```
Hello World!
SX1278: INIT returns 0
SX1278: TRANSMIT_DIRECT returns 0
```

</p>
</details> 

<details><summary>Sketch that is causing the module fail</summary>
<p>

```c++
#include <RadioLib.h>

SX1278 radio = new Module(10, 4, 5, 3);
int sx1278_ret = 0;

/*------------------------------------------------------------------------------------------*/
/* setup()                                                                                  */
/*------------------------------------------------------------------------------------------*/
void setup() {

  // System
  Serial.begin(115200);
  delay(500);
  printf("Hello World!\n\r");

  // SX1278 Init
  radio.reset();
  sx1278_ret = radio.beginFSK(450, 19.2, 7.7, 25, 17, 16, false);
  printf("SX1278: INIT returns %d\n\r", sx1278_ret);

  sx1278_ret = radio.transmitDirect();
  printf("SX1278: TRANSMIT_DIRECT returns %d\n\r", sx1278_ret);

}

/*------------------------------------------------------------------------------------------*/
/* loop()                                                                                   */
/*------------------------------------------------------------------------------------------*/
void loop() {
}
```

</p>
</details> 

**Expected behavior**

The example code sets the Module to 450.0 MHz and switches the transmitter on. With library version 7.1.0 the spectrum analyzer shows the carrier at 450.0 MHz. Compiling with library version 7.1.1 the carrier is at 434.0 Mhz. Additional setFrequency(450) after initialisation do not help.

**Hardware / Software Setup**
 - MCU: Teensy 4.1
 - Wireless module type SX1278 (RA-02)
 - Arduino IDE version 2.3.4
 - Library version 7.1.1 or 7.1.0

CrispyAlice2, 2025-01-03:
I just encountered this issue as well, it seems like the issue was the module not actually changing the frequency until the LSB frequency register was written. I wrote a quick PR, I'll submit that in a second.

jgromes, 2025-01-03:
@CrispyAlice2 I was able to replicate this too - indeed, since 29ede2c3545321f90ac8a8ea4583d3472c17ff80 register access has been optimized to remove redundant write operations. But if the module is sensitive to change in the LSB frequency register, we will need to figure out a different way. Looking forward to the PR ;)

BBluemer, 2025-01-03:
Gentlemen,

in the mean time i had a deeper look at the manual, at the Register description there is nothing mentioned about a necessary write to the lsb-register to update the whole 3 registers, but on page 82 i found the note: 'The Frf setting is split across 3 bytes. A change in the center frequency will only be taken into account when the least significant byte FrfLsb in RegFrfLsb is written.'

Thank you so much for finding this issue that quick!
Regards, Bert

jgromes, 2025-01-03:
Resolved by merging #1369 - thank you @CrispyAlice2! This bug is quite significant, so I will go ahead and release this as 7.1.2 in the next few days.

Also props to @BBluemer for a clear and concise bug report that allowed to quickly locate and resolve this issue!

closed 2025-01-03 by jgromes