jgromes/RadioLib #1392: Add support to set nRF24l01 LNA gain

BLLuis, 2025-01-17:
Implementing a channel scanner and due to the lack of nRF24l01 signal meter I turn off the LNA to reduce in 1.5 db signal sensitivity getting at least 3 levels for signal strength.
The library don't have a function to set LNA or sensitivity so I'm using RADIOLIB_GODMODE and RADIOLIB_LOW_LEVEL to have acces to Module pointer using getMod and writting to nRF24 to change this setting.

These are some lines of my code as example:

`

	#define RADIOLIB_NRF24_RF_LNA_OFF                           0b00000000  //  0     0   LNA gain: Off
	#define RADIOLIB_NRF24_RF_LNA_ON                            0b00000001  //  0     0             On

	/*
	getMod is available defining
	
	#define RADIOLIB_GODMODE    (1)
	#define RADIOLIB_LOW_LEVEL  (1)

	in component/src/BuildOptUser.h file
	*/

	Module * pModuleRadio0 = m_pRadio0->getMod();
	radioState = pModuleRadio0->SPIsetRegValue(RADIOLIB_NRF24_REG_RF_SETUP, RADIOLIB_NRF24_RF_LNA_ON, 0, 0);
`

This is good not only to reduce sensitivity but also to reduce current consumption in 0.8ma, my code is not tested on a real device yet.
Wish to see this in the library to avoid using God mode.

jgromes, 2025-01-18:
I'm not against adding this; the API can be quite simple, `int16_t nRF24::setLNA(bool enable);`. Once this is tested on real hardware, feel free to open a PR adding this - thanks!

BLLuis, 2025-01-19:
Tested on two radios, can't measure current consumption due to my limited hardware at the moment, but got changes counting channels with carrier.

`I (4485) main: Test LNA

I (4485) main: Testing LNA consumption
I (4485) main: LNA Off
I (9495) main: LNA On
I (14495) main: LNA Off
I (19495) main: LNA On

I (24495) main: Testing LNA gain
I (49495) main: R0 Ch count with LNA:4 No LNA:0
I (49495) main: R1 Ch count with LNA:5 No LNA:2
`

Will do a PR soon to add this function.
Regards.

jgromes, 2025-02-13:
@BLLuis FYI I added a public method to control the nRF24 LNA based on your example. Thank you for the suggestion!

closed 2025-02-13 by jgromes