jgromes/RadioLib #1398: Formatting issue of debug output in terminal

serialrf433, 2025-01-26:
**Describe the bug**

I enabled debugging to get more interesting output. The formatting of this output that i get out of the serial does not look like it should.
The normal output is correctly formatted [Radio]... . Its just the debugging output that does not look correctly.

**To Reproduce**

remove the // in the 4 lines before the #define 
https://github.com/jgromes/RadioLib/blob/master/src/BuildOptUser.h

**Expected behavior**

Line1
Line2
Line3
Line4
...

**Screenshots**
[image]

**Additional info (please complete):**
 - MCU: [compiled with arduino-cli version 1.1.1 for ESP32]
 - Computer-OS: Linux (using different ones, same on all)

HeadBoffin, 2025-01-27:
You may want to learn about line endings over serial - there are three combinations and you need to tell your command line software which one to translate in to line endings for the terminal.

serialrf433, 2025-01-30:
But the line endings over serial should be constant. I am using serial with other projects and esp32 or other MCU's but never had such a issue. Serial simply always worked and was correctly formatted. Here its not. The normal output have correct formatting but the debugging have not. Or to say it with 'line endings over serial': they should be consistent independent if debugging is enabled or not.

StevenCellist, 2025-01-30:
RadioLib uses `\n` or LF as line endings, which most systems can parse just fine. Especially Linux is a LF-based OS, but some terminals still want to see a CR or '\r' as well. I recently encountered PuTTY, where I needed to check "Implicit line break on LF" as well. 

@jgromes up to you to add `\r` or not.

jgromes, 2025-02-11:
I think I know what's going on now - @serialrf433 is using Arduino, which uses CR+LF for `Serial.println`, that's where the inconsistency is coming from. Of course, everything would have been displayed correctly had his (as of yet unspecified) serial console application treated LF as actual line end (as most well behaved serial consoles do). Having a mix of CR+LF and LF endings is not uncommon, even the default Arduino Serial monitor handles this correctly.

Internally, RadioLib consistently uses LF only.

That being said, there is a lot of users of RadioLib on Arduino, so I will do two things:

1. Make the RadioLib internal line feed configurable via a build-time option
2. Use CR+LF as the default.

closed 2025-02-11 by jgromes

StevenCellist, 2025-02-11:
Did you just call PuTTY not well-behaved @jgromes? ;)