jgromes/RadioLib #1399: Idea: use sizeof(byteArr) instead of a fixed number in the examples

serialrf433, 2025-01-26:
I am using the  byte byteArr[]  option from the examples and try out different data. Having to count on my own the length does not have any benefit and is just work that have to be done.
Instead of the line   
state = radio.startTransmit(byteArr, 8);
you could also use
state = radio.startTransmit(byteArr, sizeof(byteArr);

Then changes at line https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_Transmit_Interrupt/SX126x_Transmit_Interrupt.ino#L83 can be done freely without the need to count by hand and enter a number in the line below.

Or am i missing something some reason why a fixed number should be entered there?

HeadBoffin, 2025-01-27:
Quite often we statically define the maximum sized array for the payload but some transmissions don't use it all.

When building the payload you can use a variable to count along and then use that for the transmit function. But if we put that in, we'd have to explain that and that's an intermediate level implementation.

jgromes, 2025-01-27:
@serialrf433 this is intentional. The examples are rather simple and use a local, statically allocated array, so sizeof will work fine there (as will a constant number). However, it would be easy for new users to come to the conclusion that "sizeof gives me the length of an array", which is not true in general. Once the array becomes allocated dynamically, or passed from somewhere else via a pointer, it would be easy to make a mistake by leaving that sizeof in place.

closed 2025-01-27 by jgromes

serialrf433, 2025-01-30:
>  However, it would be easy for new users to come to the conclusion that "sizeof gives me the length of an array", which is not true in general.

This was already my conclusion. Thanks for letting me learn something new about that.