jgromes/RadioLib #1406: RF69 changing bitrate randomly

viraptor, 2025-01-31:
**Describe the bug**
I'm using an RFM69HCW connected through ESP32 with the esp-idf hal provided here.
While the module seems to overall work, there are no state errors reported, and there are no power draw issues, when using longer messages, the transmission bitrate seems to sometimes split in half randomly.
See here:

[image]

The red bar is where the message repeats, the green area is the desired bitrate (340 samples / symbol). The orange area is halved (171 samples / sym).

The behaviour alternates consistently - 18ms of correct bitrate, 9ms of halved bitrate + extra noise, so that seems to exclude most hardware wiring problems.

I'm honestly lost and not sure how to exclude faulty hardware in this case - is there some test I can run? Any other pointers welcome.

**To Reproduce**
<details><summary>Sketch that is causing the module fail</summary>
<p>

```c++
#include "RadiolibHal.h"

EspHal* hal = new EspHal(5, 19, 27);
RF69 radio = new Module(hal, 21, 14, 22);

ESP_LOGD("main", "Setting up radiolib");
radio.reset();
int16_t state = radio.begin(
  916.0,
  9.3,
  145.0,
  RADIOLIB_RF69_DEFAULT_RXBW,
  RADIOLIB_RF69_DEFAULT_POWER,
  0);
ESP_LOGD("main", "post-begin state: %d", state);
state = radio.disableAES();
ESP_LOGD("main", "post-aes state: %d", state);
state = radio.packetMode();
ESP_LOGD("main", "post-packet state: %d", state);
state = radio.setOOK(false);
ESP_LOGD("main", "post-ook state: %d", state);
state = radio.setOutputPower(20, true);
ESP_LOGD("main", "post-power state: %d", state);
state = radio.disableAddressFiltering();
ESP_LOGD("main", "post-address state: %d", state);
state = radio.disableSyncWordFiltering();
ESP_LOGD("main", "post-sync state: %d", state);
state = radio.setCrcFiltering(0);
ESP_LOGD("main", "post-crc state: %d", state);
radio.setAmbientTemperature(25);
ESP_LOGD("main", "temp: %d", radio.getTemperature());

uint8_t data[] = {
  0xaa, 0xaa, 0xaa, 0xaa, 0x2d, 0xd4, 0x69, 0x88, 0x1, 0x11, 0x3, 0xf9,
  0xaa, 0xaa, 0xaa, 0xaa, 0x2d, 0xd4, 0x69, 0x88, 0x1, 0x11, 0x3, 0xf9,
  0xaa, 0xaa, 0xaa, 0xaa, 0x2d, 0xd4, 0x69, 0x88, 0x1, 0x11, 0x3, 0xf9,
};
radio.transmit(data, sizeof(data), 0);
```

</p>
</details> 

**Expected behavior**
Consistent bitrate over the whole message.

**Additional info (please complete):**
 - MCU: ESP32
 - Wireless module type RFM69HCW
 - Library version 7.1.2

viraptor, 2025-01-31:
Just discovered that this is a problem with the receiver I'm using. The transmitter behaves exactly as expected 🤦

jgromes, 2025-01-31:
It happened to me more than once that I found a solution to a problem after posting about it :) Out of curiosity, what was the cause? Something changing the receiver (SDR?) sample rate?

viraptor, 2025-02-01:
The sdr dongle I was using couldn't handle the sample rate I needed continuously, even though there was still free cpu available. I'll blame it on "just macos things" - normally I'm using that device under linux and don't experience that issue there.