jgromes/RadioLib #1422: undefined reference to setNodeAddress

BugProg, 2025-02-14:
**Describe the bug**
Failed to build example code, got `undefined reference to SX126x::setNodeAddress(unsigned char)`.

**To Reproduce**
- Test a basic example code in FSK mod and set an address with `setNodeAddress`.

<details><summary>Sketch that is causing the module fail</summary>
<p>

```c++
void setup() {
    Serial.begin(9600);

    // initialize SX1262 with default settings
    Serial.print(F("[SX1262] Initializing ... "));
    int state = radio.beginFSK(868.3, 15.0, 57.136417, 234.3, 10, 32);

    if (state == RADIOLIB_ERR_NONE) {
        Serial.println(F("success!"));
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        while (true) { delay(10); }
    }

    radio.setNodeAddress(0x23);

    uint8_t sync_word[] = {0xAA, 0x2D};
    state = radio.setSyncWord(sync_word, 2);

    if (state == RADIOLIB_ERR_NONE) {
        Serial.println(F("success!"));
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        while (true) {
            delay(10);
        }
    }

    // set the function that will be called
    // when packet transmission is finished
    radio.setPacketSentAction(setFlag);
    radio.setEncoding(RADIOLIB_ENCODING_WHITENING);

    // start transmitting the first packet
    Serial.print(F("[SX1262] Sending first packet ... "));

    // you can transmit C-string or Arduino string up to
    // 256 characters long
    transmissionState = radio.startTransmit("Hello World!");

    delay(1000);
}
```

```
/.var/app/cc.arduino.IDE2/cache/.arduinoIDE-unsaved2025114-59-1qjfc97.t335/SX126x_Transmit_Interrupt/SX126x_Transmit_Interrupt.ino:71: undefined reference to `SX126x::setBroadcastAddress(unsigned char)'
```

</p>
</details> 

**Additional info (please complete):**
 - MCU: Arduino UNO r4
 - Wireless module type SX1268
 - Arduino IDE version  2.3.4
 - Library version 7.1.2.0

jgromes, 2025-02-14:
This method has been removed from source and the examples (see https://github.com/jgromes/RadioLib/issues/1268#issuecomment-2549519398 for details as to why), but it looks like I forgot to also remove it from the header.

@BugProg I'm guessing you got this either from the header or the Doxygen (which is generated from the header), could you confirm? I'm asking because if this is from somewhere else, then that would need to be updated.

BugProg, 2025-02-15:
Yes found it on the Doxygen: https://jgromes.github.io/RadioLib/class_s_x126x.html#adf54f7b83aa47ad0b803147622b3c580
And from the header in my IDE.

Thank you for your quick answer. 😀

jgromes, 2025-02-15:
OK, in that case it should all be fixed now. Thanks for reporting!

closed 2025-02-15 by jgromes