jgromes/RadioLib #1424: Undefined references to existing functions when using RadioLib with sources on Raspberry Pi 5

hardcore-thinking, 2025-02-15:
**Describe the bug**
I don't think it's a bug, but I can't seem to make it work. I'm just using the example provided with the code. But I get some undefined references and I don't understand why. And unfortunately with such errors I can't even get debug mode output.

**To Reproduce**
```c++
#include <iostream>
#include <cstdlib>
#include <chrono>
#include <thread>
#include <array>

#include "RadioLib/src/RadioLib.h"
#include "RadioLib/src/hal/RPi/PiHal.h"

std::ostream& operator<<(std::ostream& out, std::array<uint8_t, 256> const& arr) {
	for (auto& c : arr) {
		out << c;
	}

	return out;
}

// uncomment the following only on one
// of the nodes to initiate the pings
//#define INITIATING_NODE

PiHal* hal = new PiHal(0);
SX1262 radio = new Module(hal, 10, 2, 3, 9);

// or detect the pinout automatically using RadioBoards
// https://github.com/radiolib-org/RadioBoards
/*
#define RADIO_BOARD_AUTO
#include <RadioBoards.h>
Radio radio = new RadioModule();
*/

int transmissionState = RADIOLIB_ERR_NONE;
bool transmitFlag = false;
volatile bool operationDone = false;

void setFlag(void) {
  operationDone = true;
}

int main() {
	std::cout << "[SX1262] Initializing ... " << std::endl;
  	int state = radio.begin();
  
	if (state == RADIOLIB_ERR_NONE) {
    		std::cout << "success!" << std::endl;
  	}
	
	else {
    		std::cout << "failed, code " << state << std::endl;
    
		while (true) {
			std::this_thread::sleep_for(std::chrono::milliseconds(10));
		}
  	}

  	radio.setDio1Action(setFlag);

  	#if defined(INITIATING_NODE)
    	std::cout << "[SX1262] Sending first packet ... " << std::endl;
    	transmissionState = radio.startTransmit("Hello World!");
    	transmitFlag = true;
  	
	#else    
	// start listening for LoRa packets on this node
    	std::cout << "[SX1262] Starting to listen ... " << std::endl;
    	state = radio.startReceive();
    	if (state == RADIOLIB_ERR_NONE) {
      		std::cout << "success!" << std::endl;
    	}
	
	else {
      		std::cout << "failed, code " << state << std::endl;
      	
		while (true) {
			std::this_thread::sleep_for(std::chrono::milliseconds(10));
		}
    	}
  	
	#endif

	while (true) {
  		if(operationDone) {
    			operationDone = false;

    			if(transmitFlag) {
      				if (transmissionState == RADIOLIB_ERR_NONE) {
					std::cout << "transmission finished!" << std::endl;
				}
				
				else {
        				std::cout << "failed, code " << transmissionState << std::endl;
	      			}

      				radio.startReceive();
      				transmitFlag = false;

    			}
			
			else {
				std::array<uint8_t, 256> str {};
      				int state = radio.readData(str.data(), str.size());

      				if (state == RADIOLIB_ERR_NONE) {
        				std::cout << "[SX1262] Received packet!" << std::endl;

       	 				std::cout << "[SX1262] Data:\t\t" << str << std::endl;
					std::cout << "[SX1262] RSSI:\t\t" << radio.getRSSI() << " dBm" << std::endl;
					std::cout << "[SX1262] SNR:\t\t" << radio.getSNR() << " dB" << std::endl;
      				}

				std::this_thread::sleep_for(std::chrono::milliseconds(10));

      				std::cout << "[SX1262] Sending another packet ... " << std::endl;;
      				transmissionState = radio.startTransmit("Hello World!");
      				transmitFlag = true;
  	  		}
  		}
	}

	return EXIT_SUCCESS;
}
```

**Expected behavior**
Just make the SX1262's PingPong example work fine on my Raspberry Pi 5.

**Screenshots**
No screenshots but I do have the errors:
```
/usr/bin/ld: /tmp/ccA6P5sh.o: in function `main':
main.cpp:(.text+0x230): undefined reference to `SX1262::begin(float, float, unsigned char, unsigned char, unsigned char, signed char, unsigned short, float, bool)'
/usr/bin/ld: main.cpp:(.text+0x488): undefined reference to `SX126x::readData(unsigned char*, unsigned long)'
/usr/bin/ld: main.cpp:(.text+0x514): undefined reference to `SX126x::getRSSI()'
/usr/bin/ld: main.cpp:(.text+0x560): undefined reference to `SX126x::getSNR()'
/usr/bin/ld: main.cpp:(.text+0x5dc): undefined reference to `PhysicalLayer::startTransmit(char const*, unsigned char)'
/usr/bin/ld: /tmp/ccA6P5sh.o: in function `__static_initialization_and_destruction_0(int, int)':
main.cpp:(.text+0x6d0): undefined reference to `Module::Module(RadioLibHal*, unsigned int, unsigned int, unsigned int, unsigned int)'
/usr/bin/ld: main.cpp:(.text+0x6e4): undefined reference to `SX1262::SX1262(Module*)'
/usr/bin/ld: /tmp/ccA6P5sh.o: in function `PiHal::PiHal(unsigned char, unsigned int, unsigned char, unsigned char)':
main.cpp:(.text._ZN5PiHalC2Ehjhh[_ZN5PiHalC5Ehjhh]+0x38): undefined reference to `RadioLibHal::RadioLibHal(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int)'
/usr/bin/ld: /tmp/ccA6P5sh.o:(.data.rel.ro._ZTV5PiHal[_ZTV5PiHal]+0xb0): undefined reference to `RadioLibHal::pinToInterrupt(unsigned int)'
/usr/bin/ld: /tmp/ccA6P5sh.o:(.data.rel.ro._ZTI5PiHal[_ZTI5PiHal]+0x10): undefined reference to `typeinfo for RadioLibHal'
collect2: error: ld returned 1 exit status
```

**Additional info (please complete):**
 - Host machine: Raspberry Pi 5
 - Wireless module type: SX1262
 - Library version: 7.1.2

hardcore-thinking, 2025-02-15:
I forgot to mention, I adapted the code. But I don't think it comes from there. The compilation is fine, but not linking.

jgromes, 2025-02-16:
Seems like you're not actually linking the library. Can you show how you're building this, e.g. your CMake file?

hardcore-thinking, 2025-02-16:
I'm using the commands in the "Porting to non-Arduino platforms" chapter in the wiki. I don't use any Cmake made myself as I barely have any experience with it. But it pretty much is my problem. Doesn't the Cmake file in the repository build everything ? Isn't there a way to just build  it but not install it (sorry in advance if it is in the wiki, I didn't notice) ?

jgromes, 2025-02-16:
> Doesn't the Cmake file in the repository build everything

No, it just tells your build system how the build the library itself. But you still need to build your main.cpp. See the [Raspberry Pi example](https://github.com/jgromes/RadioLib/tree/master/examples/NonArduino/Raspberry) for details.

hardcore-thinking, 2025-02-16:
I didn't notice that as well. Good to know. I'll take a look at this. Thank you for your help. I'm closing the issue.

closed 2025-02-16 by hardcore-thinking