jgromes/RadioLib #1430: [Si4432] failure to receive fixed size FSK packet

lyusupov, 2025-02-19:
**Describe the bug**

I transmit a fixed size FSK modulated packet from SX1276 radio by the demo sketch 
https://github.com/jgromes/RadioLib/blob/master/examples/SX127x/SX127x_FSK_Modem/SX127x_FSK_Modem.ino

The Tx packet size is set to 8 bytes:

```
  state = radio.fixedPacketLengthMode(8);
```

However, the data received from FIFO is a 0x01 0x01 0x01  0x01  0x01  0x01  0x01 garbage.

See the SPI log below

<details><summary>Debug mode output</summary>
<p>

```
RLB_SPI: R      7       1
RLB_SPI: W      7       1
RLB_SPI: R      7       1
RLB_SPI: R      8       0
RLB_SPI: W      8       2
RLB_SPI: R      8       2
RLB_SPI: R      8       2
RLB_SPI: W      8       0
RLB_SPI: R      8       0
RLB_SPI: R      3       20      12
RLB_SPI: W      5       3
RLB_SPI: W      6       0
RLB_SPI: W      7       5
RLB_SPI: R      3E      8
RLB_SPI: R      3       21      32
RLB_SPI: R      7F      1       1       1       1       1       1       1      1
RLB_SPI: R      7       5
RLB_SPI: W      7       1
RLB_SPI: R      7       1
RLB_SPI: R      3       A0      12
[Si4432] Received packet!
[Si4432] Data: 11111111
```

</p>
</details> 

**To Reproduce**

Use https://github.com/jgromes/RadioLib/blob/master/examples/SX127x/SX127x_FSK_Modem/SX127x_FSK_Modem.ino sketch to transmit and https://github.com/jgromes/RadioLib/tree/master/examples/Si443x/Si443x_Receive_Interrupt to receive.

**Expected behavior**

Received data is expected to be 0x0123456789abcdef

**Screenshots**

URH tool can confirm that Tx signal is fine. 
It has
- 16 bits of 0x5555 preamble
- 0x12AD sync word
- 0x0123456789abcdef of payload data

[image]

**Additional info (please complete):**
 - MCU: ESP32
 - Wireless module type SX1276 (TX) and Si4432 (RX)
 - Library version - 7.1.2

> [!TIP]
> In the issue report https://github.com/jgromes/RadioLib/discussions/442
> you say that

[image]

jgromes, 2025-02-20:
I did indeed add this feature in 19109bc7d84e462b1309b7e05332a8f112b4dcb4, so I'm not entirely sure what's going on here. It seems odd that the only byte returned is `0x01`, which just so happens to be the first byte you send. Could you try sending with a different first byte to see if the returned data is different? If so then it would seem that for some reason it's always returning the first byte only. Although there do seem to be 8 bytes, which checks out.

lyusupov, 2025-02-20:
> Could you try sending with a different first byte to see if the returned data is different?

The MCVE above is based on the RadioLib demo sketch. However, the issue originally found in 'real world' use case.
Yes, the byte value returned depends on the data been sent but every time it consists of multiple occurrence of a single byte value.

The transmit feature of a fixed sized packet works fine to me. A packet sent by Si4432 gets received by SX1276 with no issues. Si4432 receive feature is the only affected.

jgromes, 2025-03-14:
I was able to replicate this issue with SX126x transmitter, regardless of whether fixed packet mode was enabled or not. It turns out the actual issue was due to the following:

1. Si443x driver had no support for CRC configuration, hence it remained on the default value (enabled).
2. IRQ flag for CRC error on receive was enabled, but not processed

The Si443x received the packet, but its default CRC configuration is different from the SX126x default, hence it evaluated the packet as broken (regardless whether CRC was enabled on the transmitter or not). However, RadioLib did not handle this correctly (by returning an error code), and for some reason, reading the FIFO after CRC error just repeatedly returns the first byte.

I most likely never caught this issue since I tested Si443x receiver against Si443x transmitter - the issue will not manifest when both of them are misconfigured in the same way.

To fix this, multiple things were needed:

1. Proper handling of the CRC error flag - `readData` now returns the appropriate error for mismatched CRC
2. Adding support for Si443x CRC - this allows to set up the CRC config of Si443x via `Si443x::setCRC`. Unfortunately the configuration is pretty limited - I was not able to find one that would work with SX127x/6x. So the only solution is to disable CRC on both ends with `radio.setCRC(false);`.

All should now be resovled - thanks for reporting!

closed 2025-03-14 by jgromes