jgromes/RadioLib #1433: [si4432] Connecting the stm32duino FreeRTOS library disrupts the interrupt

Dmitrij-Solovjev, 2025-02-24:
**Describe the bug**
Connecting the stm32duino FreeRTOS library breaks the interrupt (on the basic example of asynchronous message sending for si4432 (it is never called for startTransmit, although if you touch the pin the interrupt is triggered).

**To Reproduce**
Basic example of asynchronous sending and connection of stm32duino FreeRTOS

**Additional info (please complete):**
 - MCU: stm32f411 (Blackpill v2)
 - Link to Arduino core: [https://github.com/stm32duino/STM32FreeRTOS/releases/tag/10.3.2](STM32FreeRTOS v10.3.2)
 - Wireless module type: si4432
 - PlatformIO Core:6.1.17, Home:3.4.4
 - Library version: 7.1.2

StevenCellist, 2025-02-24:
Hi @Dmitrij-Solovjev, thank you for taking your time to report an issue here on RadioLib! And condoning the edits, your final edit is a pretty nicely filed issue.

While I'm not a definitive STM32 expert, I'll try a guess based on my experience with the STM32WLE5:
The STM32 platform in itself does not 'persist' interrupts (at least for them STM32WLE5): if an interrupt is processed, the interrupt must be removed in order for it to stop 'firing'. Thus, the interrupt is never fired again. The STM32duino core has a workaround for this as far as I can remember, but I can't find it at this instance. Maybe including the FreeRTOS library breaks this workaround again - it's the best guess I can do right now.

In the support code for the STM32WLE5 in RadioLib, [this workaround](https://github.com/jgromes/RadioLib/blob/cf776230a448faf7414f2809e12b95b68ada9014/src/modules/SX126x/STM32WLx.cpp#L104-L114) is used where the interrupt is restarted when the interrupt is being cleared. For the STM32WLE5, the radio is integrated into the controller using a custom SPI interface (SubGHz) with its own functions - it isn't possible to copy this workaround. A simple idea however could be to re-set the `setPacketSentAction()` function after every transmission, making sure that the interrupt is re-attached every time.

jgromes, 2025-02-24:
From experience, this seems like incorrectly configured interrupt priority in FreeRTOS. What is your FreeRTOS configuration and how are the tasks set up?

Dmitrij-Solovjev, 2025-02-24:
That's the irony. I did NOT create the tasks, I did NOT run the scheduler, I just specified the library dependency in `platformio.ini` (I didn't even connect it via the #include directive, which is funny). To exclude the development environment factor, I will try to connect the libraries as local (in the project/lib folder) and will write about the results within 24 hours, but it is unlikely that connecting the libraries differently will fix the situation.

Dmitrij-Solovjev, 2025-02-24:
I created a [repository](https://github.com/Dmitrij-Solovjev/IoT-Sniffer) in which I put an almost identical (with exact pinout and LED_BUILDIN blinking) to the original example with [asynchronous sending](https://github.com/jgromes/RadioLib/tree/cf776230a448faf7414f2809e12b95b68ada9014/examples/Si443x/Si443x_Transmit_Interrupt). When running this code, the following strange behavior is observed: if you comment out `#include <FreeRTOS.h>`, sending will be successful, if you leave it as it is, sending will not work.
I apologize for the extra fuss, this is probably where I should have started 😅

Dmitrij-Solovjev, 2025-02-24:
Anyway, I realized what the problem was: I used to connect the STM32FreeRTOS library via lib_deps, which allowed me to use the incorrect connection via `#include <FreeRTOS.h>`. Now that I got rid of lib_deps and tried to write an example using the locally installed library in the project/lib folder, I got errors that made me think the library was connected incorrectly :).
I apologize for the aimlessly wasted time.

p.s. It's funny, but this is not the first project where I've connected the library using `#include <FreeRTOS.h>` (which turned out to be wrong) instead of `#include <STM32FreeRTOS.h>`, and I've never had such problems before. How amazingly the world works

closed 2025-02-24 by Dmitrij-Solovjev