jgromes/RadioLib #1434: LR1121 On a ESP32-U4WDH

gc9n, 2025-02-27:
hello there all

i have a ELRS RX board with ESP32-U4WDH,  pin mapping is ok however i got some freezing uppon   radio.begin();
its very weird , i dont understand what is going on , so i added a lot of debbung steps 

```
`#include <RadioLib.h>
#define NSS_PIN 5
#define IRQ_PIN 34
#define NRST_PIN 2
#define BUSY_PIN 35
#define LED_PIN_GREEN 33  // Define LED pin
#define LED_PIN_RED 25    // Define LED pin

SPIClass newSPI(HSPI);
 LR1121 radio = new Module(NSS_PIN, IRQ_PIN, NRST_PIN, BUSY_PIN, newSPI);
 
// SPI bus pins:
#define LR1121_SCK 18   // Use GPIO18 for SCK
#define LR1121_MOSI 23  // Use GPIO23 for MOSI
#define LR1121_MISO 19  // Use GPIO19 for MISO
#define LR1121_NSS 5    // Use GPIO5 for NSS (CS)
void setup() {
  Serial.begin(115200);
  Serial.println("Setup Start");

  pinMode(LED_PIN_GREEN, OUTPUT);
  pinMode(LED_PIN_RED, OUTPUT);
  digitalWrite(LED_PIN_GREEN, HIGH);
  digitalWrite(LED_PIN_RED, HIGH);
  delay(1000);

  Serial.println("Initializing SPI bus...");
  newSPI.begin(LR1121_SCK, LR1121_MISO, LR1121_MOSI, LR1121_NSS);
  newSPI.setFrequency(125000);  // Lower SPI speed
  delay(10);
  Serial.println("SPI Initialized.");

  Serial.println("Resetting LR1121...");
  pinMode(NRST_PIN, OUTPUT);
  digitalWrite(NRST_PIN, LOW);
  delay(10);
  digitalWrite(NRST_PIN, HIGH);
  delay(10);
  Serial.println("LR1121 Reset complete.");

  Serial.println("Setting NSS HIGH...");
  pinMode(NSS_PIN, OUTPUT);
  digitalWrite(NSS_PIN, HIGH);
  delay(10);
  Serial.println("NSS Set.");

  Serial.print("Checking BUSY pin: ");
  pinMode(BUSY_PIN, INPUT);
  int busyState = digitalRead(BUSY_PIN);
  Serial.println(busyState ? "HIGH" : "LOW");

  if (busyState == HIGH) {
    Serial.println("ERROR: LR1121 BUSY pin is HIGH! Device might be stuck.");
    while (true) { delay(100);  digitalWrite(LED_PIN_RED, LOW);}
  }

  Serial.println("Power cycling LR1121...");
  digitalWrite(NSS_PIN, LOW);
  delay(100);
  digitalWrite(NSS_PIN, HIGH);
  delay(100);
  Serial.println("Power cycle complete.");

  Serial.println("Testing SPI communication...");
[... 244 lines cut ...]
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[LR1110] SNR:\t\t"));
      Serial.print(radio.getSNR());
      Serial.println(F(" dB"));

    } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);

    }
  }
}
`
 
```

do i have to check somehting in the library it self , a parameter a setting or somehting? 
*i am using Arduino IDE 2.3.3

jgromes, 2025-02-27:
> so i added a lot of debbung steps

First of all, please don't do that. RadioLib has a [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode) for this. Enable basic and SPI debug and then post the result. Your "debugging sequence" is also incorrect, as it only samples BUSY pin once and ignores the fact that the LR11xx typically takes some 270 ms to start up after pulling the reset pin. That's why the BUSY pin is high - see the device datasheet.

>  pin mapping is ok

How do you know that? Is this some custom board? If so, it is impossible for us to know whether the connections are correct. Pinout of the ESP32 contains no useful information in this case, as it does not show how you have connected the LR1121 to it.

In short, post the needed information and debug mode (as per the issue template), otherwise we unfortunately cannot help you.

gc9n, 2025-02-27:
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDW	0	0	
RLB_SPI: SI			
RLB_SPI: SO	4	13	
RLB_SPI: CMDW	1	1C	
RLB_SPI: SI			0	
RLB_SPI: SO	4	13	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDW	1	D	
RLB_SPI: SI			
RLB_SPI: SO	4	13	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	
RLB_SPI: SO	6	0	0	
RLB_SPI: CMDW	1	17	
RLB_SPI: SI			0	0	0	A3	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDW	2	13	
RLB_SPI: SI			1	
RLB_SPI: SO	4	13	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDW	1	14	
RLB_SPI: SI			1B	F8	F	FC	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDW	1	13	
RLB_SPI: SI			0	0	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	0	0	0	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_DBG: GPIO post-transfer timeout, is it connected?
RLB_DBG: -705 at C:\Users\Gchat\OneDrive\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279

gc9n, 2025-02-27:
i assume is SPI issue wrong pins or something like that

gc9n, 2025-02-28:
according to the debug results the chip is recognised so i guess the SPI transfers some data, 
RLB_DBG: Found LR11x0: RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03
RLB_DBG: Base FW version: 1.1
RLB_DBG: M	LR11x0
 
any thoughts @jgromes

StevenCellist, 2025-02-28:
@gc9n there's no use pasting the exact same log again (even more as you didn't even format it, which Mr. Gromes has nicely done for you twice now, so check out how to do it yourself).

You did add an observation

> according to the debug results the chip is recognised so i guess the SPI transfers some data

but haven't shown us a picture of your setup or posted other things you have tried, like changing the pins used (or is the radio soldered, in which case we would like to see a schematic). And we're not obliged to help you within the 13 hours that it took you to re-post the same log (there's not any obligation whatsoever), you're advised to practice some patience if you're really out of ideas. But I think there's some stuff that you can try and more info you can share first.

gc9n, 2025-02-28:
> [@gc9n](https://github.com/gc9n) there's no use pasting the exact same log again (even more as you didn't even format it, which Mr. Gromes has nicely done for you twice now, so check out how to do [...]

yeah , i had a felling that i didnt copy paste the exact Debug output sorry about that.
Anyway , for the picture of my setup its just a RC radio RX for ELRS is very tiny,  i dont have the scematic , i just got the pinout mapping from a source , i tryied to verify the connections from MCU and LR1121 and yes its the exact pins  , 

So far i tested various things Like Hardware SPI , VSPI , adding the SPI settings on the Init 
 LR1121 radio = new Module(NSS_PIN, IRQ_PIN, NRST_PIN, BUSY_PIN, newSPI,SPISettings(2000000, MSBFIRST, SPI_MODE0));

however the Results is exactly the same 
RLB_DBG: GPIO post-transfer timeout, is it connected?
RLB_DBG: -705 at C:\Users\.\OneDrive\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
i am also intentionally changing the SPI pins to wrong ones to see what i got from the debug and the result is deferent ofcourse not even recognises the Radio.

i have a feeling that theres is something wrong with the ESP32-U4WDH and Arduino IDE it self , i mean , theres no Board to select (ESP32-U4WDH) on the IDE , i am using ESP32 DEV Module and /or ESP32 PICO D4. 
i continue the investigation , if i find something i will post my results

jgromes, 2025-02-28:
From the debug log, the communication seems OK so there's no need to be experimenting with different SPI pins. What is not OK is the timeout on the BUSY signal, which should go low after the command is completed. It has one second to do so, and the command at which this happens is `RADIOLIB_LR11X0_CMD_SET_DIO_IRQ_PARAMS`, which shouldn't take long.

Couple of different things could be happening here:

1. Timing issue - if for some reason the `millis()` counter of the ESP32 is ticking too fast, then it would cause the timeout to elapse sooner than the 1000 ms specified. This can be checked by just verifying something like `delay(10000);` actually stops execution for 10 seconds.
2. Some hardware issue - power, unreliable connections etc., but without further information about the hardware as pointed out by @StevenCellist, we cannot assist here.

gc9n, 2025-03-01:
i added in LR11x0.cpp this line

```

int16_t LR11x0::SPIcommand(uint16_t cmd, bool write, uint8_t* data, size_t len, uint8_t* out, size_t outLen) {
  int16_t state = RADIOLIB_ERR_UNKNOWN;
      **RADIOLIB_DEBUG_BASIC_PRINTLN("Sending Command... = 0x%02x",cmd); ///More Debug stuff**

  if(!write) {
    // the SPI interface of LR11x0 requires two separate transactions for reading
    // send the 16-bit command

    state = this->mod->SPIwriteStream(cmd, out, outLen, true, false);
    RADIOLIB_ASSERT(state);

    // read the result without command
    this->mod->spiConfig.widths[RADIOLIB_MODULE_SPI_WIDTH_CMD] = Module::BITS_0;
    state = this->mod->SPIreadStream(RADIOLIB_LR11X0_CMD_NOP, data, len, true, false);
    this->mod->spiConfig.widths[RADIOLIB_MODULE_SPI_WIDTH_CMD] = Module::BITS_16;

  } else {
    // write is just a single transaction
    state = this->mod->SPIwriteStream(cmd, data, len, true, true);
  
  }
  
  return(state);
}
```

you said the `and the command at which this happens is RADIOLIB_LR11X0_CMD_SET_DIO_IRQ_PARAMS`

according to the debug i have 

`RLB_DBG: Found LR11x0: RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03
RLB_DBG: Base FW version: 1.1
RLB_DBG: M	LR11x0
RLB_SPI: CMDW	0	0	
RLB_SPI: SI			
RLB_SPI: SO	4	13	
RLB_DBG: Sending Command... = 0x11c
RLB_SPI: CMDW	1	1C	
RLB_SPI: SI			0	
RLB_SPI: SO	4	13	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDW	0	0	
RLB_SPI: SI			
RLB_SPI: SO	4	13	
RLB_DBG: Sending Command... = 0x11c
RLB_SPI: CMDW	1	1C	
RLB_SPI: SI			0	
RLB_SPI: SO	4	13	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_DBG: Sending Command... = 0x10d
RLB_SPI: CMDW	1	D	
RLB_SPI: SI			
RLB_SPI: SO	4	13	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	
RLB_SPI: SO	6	0	0	
RLB_DBG: Sending Command... = 0x117
RLB_SPI: SI			
RLB_SPI: SO	4	13	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	
RLB_SPI: SO	6	0	0	
RLB_DBG: Sending Command... = 0x117
RLB_SPI: CMDW	1	17	
RLB_SPI: SI			0	0	0	A3	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_DBG: Sending Command... = 0x213
RLB_SPI: CMDW	2	13	
RLB_SPI: SI			1	
RLB_SPI: SO	4	13	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_DBG: Sending Command... = 0x114
RLB_SPI: CMDW	1	14	
RLB_SPI: SI			1B	F8	F	FC	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
RLB_DBG: Sending Command... = 0x113
RLB_SPI: CMDW	1	13	
RLB_SPI: SI			0	0	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	0	0	0	0	
RLB_SPI: CMDR	
RLB_SPI: SI	0	0	0	0	0	0	
RLB_SPI: SO	4	13	0	0	0	0	
**RLB_DBG: Sending Command... = 0x10f**
RLB_DBG: GPIO post-transfer timeout, is it connected?
RLB_DBG: -705 at C:\Users\Gchat\OneDrive\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
`
RLB_DBG: Sending Command... = 0x10f   if i am not mistaken is 
#define RADIOLIB_LR11X0_CMD_CALIBRATE                           (0x010F)

lyusupov, 2025-03-01:
Since it does not look like a true issue with the library - it could make sense to transfer it into [**Discussions**](https://github.com/jgromes/RadioLib/discussions) area.

gc9n, 2025-03-01:
OK **Major Update** it seems that the issue is TCXO or Freq or some sort of default value that corrupts it.

i added   int state = radio.begin(915.0, 250.0, 7, 5, 0x34, 14, 8, 0);  //radio.begin();
and is not stucking anymore , i was able to make a Firmware Update  on the LR1121 firmware as well.
But i dont like it , i still havent found the issue
 running the example like Wifi Scan , on radio.Begin() stucks of course ,

gc9n, 2025-03-01:
i believe my chip has XTAL  i mean is not supporting TCXO

gc9n, 2025-03-01:
The problem resolved completly as soon as i set the LR11x0::XTAL to true
i just    int state = radio.begin();   withou issue ,

jgromes, 2025-03-02:
I'm glad you were able to resolve the issue (though I will point out that had you shown us the schematic, we would have been able to point out that you are not using a TCXO).

This behvaior is [typical on the SX126x](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide#sx126xsx128xlr11x0-incorrect-oscillator), selecting incorrect oscillator will prevent it from starting up. However, it looks like in the case of LR11x0, this manifests slightly differently, in the BUSY pin timeout. I have added an appropriate section in the troubleshooting guide, since this is my first time encountering a LR11x0 device with XTAL.

It also helped me to fix a corner case where the debug output was not displayed after BUSY timeout - that's why I mistook the command that was actually failing.

gc9n, 2025-03-02:
So the next step is to test the RX and TX capabilities ,   i have 2 units,  i load up the Modem for LR1121 
i can send/Transmit , but i cannot recieve absolutly anythjing, its amazing !   i can see the data on 915  , on SDR , but the other node cant read anythjing , i am using mostly examples. 
..i will post anything i find here 
There is no schematic.

i know that DIO5 is RFSW0 i use it like 
```

static const uint32_t rfswitch_dio_pins[] = { 
  RADIOLIB_LR11X0_DIO5, RADIOLIB_NC,
  RADIOLIB_NC, RADIOLIB_NC, RADIOLIB_NC
};

static const Module::RfSwitchMode_t rfswitch_table[] = {
  // mode                  DIO5  DIO6 
  { LR11x0::MODE_RX,     { LOW  } },
  { LR11x0::MODE_TX,     { HIGH } },
  END_OF_MODE_TABLE,
};
```

if there is anything to look for and check i am mostly interesting for Recieving stuff

jgromes, 2025-03-02:
> There is no schematic.

Unfortunately for you that is the only source of information that will tell you definitively how the RF switch is wired up. This is a question that is specific to your hardware, so you should probably ask whoever made it.

> i will post anything i find here

Please don't. Issues are supposed to be used to report things (potentially) wrong with the library, so that we can fix them. This is not a place for people to document their troubleshooting. [Discussions](https://github.com/jgromes/RadioLib/discussions) maybe, but not here.

Closing as the original issue has been resolved.

closed 2025-03-02 by jgromes