jgromes/RadioLib #1435: reserved identifier violation

elfring, 2025-02-27:
I would like to point out that an identifier like “[`_RADIOLIB_MODULE_H`](https://github.com/jgromes/RadioLib/blob/master/src/Module.h#L1-L2 "Update candidate")” [does not fit](https://wiki.sei.cmu.edu/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier#DCL51CPP.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

jgromes, 2025-02-27:
Technically you are correct, macro names starting with underscores are not compliant with C++ spec. However, since all guard macros here start with `_RADIOLIB`, the chance for clashing with anything defined in C++ standard library or elsewhere is virtually zero. Seeing as thise change would be in literally every header here (since I am at least consistent), I'd rather keep it as is.

closed 2025-02-27 by jgromes

elfring, 2025-02-27:
* :eyes: Please take another look at [name conventions](https://stackoverflow.com/questions/228783/what-are-the-rules-about-using-an-underscore-in-a-c-identifier#answer-228797 "What are the rules about using an underscore in a C++ identifier?").
  How do you think about to avoid that this software [depends on undefined behaviour](https://en.cppreference.com/w/cpp/language/identifiers#In_declarations "Documentation")? :thinking:
* :thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/bugprone/reserved-identifier.html "Checking the application of reserved identifiers")?

jgromes, 2025-02-27:
Do you think the chances of C++ standard library introducing a macro like `_RADIOLIB_MODULE_H` are high? I remain unconvinced. But if you can point to this issue manifesting with RadioLib, then I will fix it immediately.

> How do you think about to avoid that this software [depends on undefined behaviour](https://en.cppreference.com/w/cpp/language/identifiers#In_declarations)

For header guard macros (which is the only usage of underscored macros in RadioLib), this will never reach undefined behavior. Just having a macro that starts with underscore is not undefined; what is undefined is `#define`ing a macro twice, which could happen if C++ standard library happened to define something like `_RADIOLIB_MODULE_H`. But the first thing that happens is an `#if defined()` check, which would prevent inclusion if the macro was already defined. Will that break the library compilation? Probably. Is it undefined? Maybe, on a technicality of the reserved name itself, but by my reading of the C++ spec, I don't think it is.

> Would you become interested to [use a development tool like “clang-tidy”](https://clang.llvm.org/extra/clang-tidy/checks/bugprone/reserved-identifier.html)

No I would not, I hate automated linters with a passion.