jgromes/RadioLib #1436: Remove an unnecessary null pointer check

elfring, 2025-02-27:
[An extra null pointer check is not needed](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") in [the destructor of the class “ExternalRadio”](https://github.com/jgromes/RadioLib/blob/cf776230a448faf7414f2809e12b95b68ada9014/src/protocols/ExternalRadio/ExternalRadio.cpp#L37-L41 "Update candidate").

jgromes, 2025-02-27:
You're right, this check is technically redundant, so I removed it. Then again, it doesn't really hurt anything...

closed 2025-02-27 by jgromes

elfring, 2025-02-27:
Thanks for [your small source code improvement](https://github.com/jgromes/RadioLib/commit/c9690a7955bbcf6234abb50d64bc29584bb2aff5 "Remove redundant null check").