jgromes/RadioLib #1449: Would you like me to add some examples of the xiaoe-esp32s3+wio-sx1262

hpssjellis, 2025-03-06:
**IMPORTANT: Check the docs**
Checked.

**Is your feature request related to a problem? Please describe.**
The sx1262 examples do not work  out of the box on the esp32s3-sx1262 from Seeedstudio

**Describe the solution you'd like**
I can make a sub folder of examples that run, are you interested in me sending a PR with those 

**Describe alternatives you've considered**
I have a working LoRa PP pingPong and will start working on a LoRaWan example.

Is there any chance you could get AES encryption working on the library. I would really like to be able
to use the Arduino IDE to work with some meshtastic send and receive.

**Additional context**
For $9.90 these kits are very affordable   https://wiki.seeedstudio.com/wio_sx1262_with_xiao_esp32s3_kit/

jgromes, 2025-03-06:
> I can make a sub folder of examples that run, are you interested in me sending a PR with those

I don't think that's a good idea. What exactly did you have to modify for the examples to work? Basic stuff like pinout and SPI configuration is expected to be different for every board. If we were to have an example for each one, that would be hundreds of examples ...

> Is there any chance you could get AES encryption working on the library.

There is every chance as RadioLib comes with a built-in AES support. See here: https://jgromes.github.io/RadioLib/class_radio_lib_a_e_s128.html

hpssjellis, 2025-03-06:
I think lots of examples for different boards is a really good thing. Lots of makers give up when the first few examples fail. The problem with this board is Seeedstudio did not make it very obvious which pins to use. 

Here several people had issues:

https://forum.seeedstudio.com/t/xiao-esp32s3-wio-sx1262-arduino-ide-lora-error/284419/15

I can't seem to get a LoRaWan example running for these boards.

I am curious about using your encryption with meshtastic. I think it is ridiculous that meshtastic does not maintain an arduino library. Your encryption is 128 I think Meshtastic needs 256. Any suggesstions?

hpssjellis, 2025-03-06:
Thanks for chatting.  I will keep my examples on my curriculum site.

I did use the RadioBoards to get the correct pins and really appreciate that. I agree that I will need to find a different encryption for Meshtastic. 

What about this board with LoRaWan. I don't see an OTAA example. Any suggestions?

jgromes, 2025-03-06:
Exactly, but I also highly suggest to read the .md files in there, I can guarantee the information in there will be useful to you.

closed 2025-03-06 by jgromes