jgromes/RadioLib #1451: I cannot get sx1262 to initialize. Error -2

Celppu, 2025-03-07:
I cannot get sx1262 to initialize.
Using version 7.1.2
Platformio, esp32-s3 devkitc and arduino core

I can only get error -2 . Also with oscilloscope there seem to be something in MISO so it does answer?

Code
`
#include <RadioLib.h>

//LORA 1
#define LORA1_MISO 14
#define LORA1_MOSI 12
#define LORA1_SCK 13
#define LORA1_NSS 9
#define LORA1_RST 10
#define LORA1_DIO0 47 // Not connected
SPISettings spiSettings(4000000, MSBFIRST, SPI_MODE0);

SX1262 radio = new Module(LORA1_NSS, LORA1_DIO0, LORA1_RST, RADIOLIB_NC, SPI, spiSettings);

#define LORA_SS  37
#define LORA_RST 10
#define LORA_DIO0 8

void setup() {
  Serial.begin(1000000);

  Serial.print(F("[SX1262] Init... "));

  SPI.begin(13, 14, 12, 9);

  int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 10, 8, 0, false);
    
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
    Serial.print(F("fs, code "));
    Serial.println(state);
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop(){

}`

And prints
`[SX1268] Init... RLB_DBG:
RadioLib Info
Version:  "7.1.2.0"
Platform: "ESP32"
Compiled: "Mar  7 2025" "22:42:52"
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (1 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (2 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (3 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (4 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (5 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (6 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (7 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (8 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (9 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0  00       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22      20      20 222      0       20      22      0       20      22      0
RLB_DBG: SX126x not found! (10 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_DBG: No SX126x found!
RLB_DBG: -2 at .pio/libdeps/esp32-s3-devkitc-1/RadioLib/src/modules/SX126x/SX126x.cpp:40
RLB_DBG: -2 at .pio/libdeps/esp32-s3-devkitc-1/RadioLib/src/modules/SX126x/SX1262.cpp:13
failed, code -2`

lyusupov, 2025-03-07:
RadioLib driver for SX1262 will likely NOT work properly when BUSY is NC

[image]

Celppu, 2025-03-07:
> RadioLib driver for SX1262 will likely NOT work properly when BUSY signal is NC
> 
> [image]

I added busy pin, but still no luck. 

This code works on arduino IDE ??????

```C++
/*
  RadioLib SX126x Blocking Transmit Example

  This example transmits packets using SX1262 LoRa radio module.
  Each packet contains up to 256 bytes of data, in the form of:
  - Arduino String
  - null-terminated char array (C-string)
  - arbitrary binary data (byte array)

  Other modules from SX126x family can also be used.

  Using blocking transmit is not recommended, as it will lead
  to inefficient use of processor time!
  Instead, interrupt transmit is recommended.

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
#define LORA1_MISO 14
#define LORA1_MOSI 12
#define LORA1_SCK 13
#define LORA1_NSS 9
#define LORA1_RST 10
#define LORA1_DIO0 47 // Not connected
SPISettings spiSettings(1000000, MSBFIRST, SPI_MODE0);

SX1261 radio = new Module(LORA1_NSS, LORA1_DIO0, LORA1_RST, 47/*RADIOLIB_NC*/, SPI, spiSettings); // changed to SX1268
//RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
#define LORA_SS  37
#define LORA_RST 10
#define LORA_DIO0 8

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

// or using CubeCell
//SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);

void setup() {
  Serial.begin(115200);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  SPI.begin(13, 14, 12, 9);

  int state = radio.begin(868.0, 125.0, 9, 7, 0x12, 10, 8, 0, false);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
[... 210 lines cut ...]
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0      00       0       0       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      2       0       20      22      22     20       20      2       22      0       20      22      0       20      22      0       
RLB_DBG: SX126x not found! (9 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".    
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW   0
RLB_SPI: SI
RLB_SPI: SO     FF
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     80      80
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0      00       0       0       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      53      58      31      32      36     30       20      56      32      44      20      32      44      30      32      0       
RLB_DBG: SX126x not found! (10 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 30 20 56 32 44 20 32 44 30 32 00  SX1260 V2D 2D02.    
RLB_DBG: Expected string: SX1261
RLB_DBG: No SX126x found!``` 

SX1260 ????

Celppu, 2025-03-08:
Ok it gets more interesting. On arduino ide i get 7.1.1 version to work if all debug prints are on ?!? and if not depends on reset cycle

```
ode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[SX1262] Initializing ... failed, code -2
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[SX1262] Initializing ... success!
[SX1262] Transmitting packet ... ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[SX1262] Initializing ... success!
[SX1262] Transmitting packet ... success!
[SX1262] Datarate:	606.06 bps
[SX1262] Transmitting packet ... success!
[SX1262] Datarate:	606.06 bps
[SX1262] Transmitting packet ... success!
[SX1262] Datarate:	606.06 bps```

lyusupov, 2025-03-08:
[image]

fdkuiper, 2025-03-08:
Hello, I believe I had something similar in the past. Then I switched in platform.ini to a different platform:
`platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip`

This fixed the problem for me at that time. You can consider to try this (after inspecting the specific github repository to see if you trust it).

Celppu, 2025-03-08:
> [image]

I fixed this part and verified connections!
``` C++
#define LORA1_MISO 14
#define LORA1_MOSI 12
#define LORA1_SCK 13
#define LORA1_NSS 9
#define LORA1_RST 10
#define LORA1_DIO0 8
#define LORA1_BUSY 47
SPISettings spiSettings(1000000, MSBFIRST, SPI_MODE0);

SX1261 radio = new Module(LORA1_NSS, LORA1_DIO0, LORA1_RST, LORA1_BUSY /*RADIOLIB_NC*/, SPI, spiSettings); // changed to SX1268
```

But!! The code still works only if all library debug prints are on. I'm using this module :
[https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_Vollgo-SX1262S8S-T-X1_C718838.pdf](https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_Vollgo-SX1262S8S-T-X1_C718838.pdf)

Any idea why this happens?

Here using same code v 7.1.1
```
RLB_SPI: SI				0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
RLB_SPI: SO	A2	A2	A2	A2	2	0	20	22	22	20	20	2	22	0	20	22	0	20	22	0	
RLB_DBG: SX126x not found! (3 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	A2	A2	
RLB_SPI: CMDR	1D	3	20	
RLB_SPI: SI				0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
RLB_SPI: SO	A2	A2	A2	A2	2	0	20	22	22	20	20	2	22	0	20	22	0	20	22	0	
RLB_DBG: SX126x not found! (4 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 02 00 20 22 22 20 20 02 22 00 20 22 00 20 22 00  .. ""  .". ". ".
RLB_DBG: Expected string: SX1261
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	A2	A2	
RLB_SPI: CMDR	1D	3	20	
RLB_SPI: SI				0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
RLB_SPI: SO	A2	A2	A2	A2	53	58	31	32	36	31	20	56	32	44	20	32	44	30	32	0	
RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
RLB_DBG: 
RLB_DBG: M	SX126x
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	A2	A2	
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
``` 

notice multiple tries, but eventually the example code starts executing as should but without debugs ->
```
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[SX1262] Initializing ... failed, code -2
```

jgromes, 2025-03-08:
@Celppu the module you have linked to seems ... suspicious, I have never seen one like that and Chinese-only datasheet does not inspire confidence. So my first guess would be some hardware/wiring/power issue. After that, some difference in configuration between Arduino and Platformio.

However, based on what I can see so far, this does not seem like an issue in the library, so converting to discussion.