jgromes/RadioLib #1459: HopeRF RFM95 - SX1276 - does not sleep

clavisound, 2025-03-18:
Hello there.

first time user of RadioLib, so maybe I am doing something wrong.

To my understanding, RadioLib does not put to sleep the HopeRF RFM95 SX1276 module. After transmitting, I have 2mA of current from the HopeRF. ATmega is on Deep Sleep.

I defined all the pins `NS/CSS`, `IRQ/DIO0`, `RST`, `DIO1`.

StevenCellist, 2025-03-18:
Well... there's `radio.sleep()`.

jgromes, 2025-03-19:
@clavisound generally, RadioLib does not do anything you don't tell it to. The reason is that the library does not know what your use case is. It probably makes sense for you to put the module to sleep. For other users, it might not because their usage is not constrainted by power, but needs the radio to react quickly. 

Since this is not really an issue, I will move this thread to discussions.