jgromes/RadioLib #1464: ERROR JOIN ChirpStack RADIOLIB_ERR_NO_JOIN_ACCEPT

jpayalateva, 2025-03-20:
Hi everyone,

I am having issues with the JOIN process using the STM32WL chip and the RadioLib library. My network server is ChirpStack, which correctly registers the JoinAccept, but the library seems not to listen to the response and returns the error RADIOLIB_ERR_NO_JOIN_ACCEPT.

I have verified that the JoinRequest is sent correctly and that the JoinAccept is received by ChirpStack, but the device does not seem to process this response. Has anyone had a similar problem or knows how to solve it?

Thanks in advance for your help.

[image]

[image]

[LORA] [STM32WL] Initializing ...
[LORA] radio.begin(868.0)
[LORA] success!
[LORA] setPacketReceivedAction
[LORA] beginOTAA LoRaWAN 1.0.4
joinEUI: 00000001e
DevUid: 4233501820613199
nwkKey: 6EADBFD6D9114FD5A6411EFFFFFFFFFF
appKey: 2B7E151628AED2A6ABF7158FFFFFFFFF
[LORA] success!
Join LoRaWAN Network joinEUI: 00000001f
[4.438644] RLB_PRO: Setting up dynamic channels
[4.442205] RLB_PRO: UL: 0 1 (0 - 5) | DL: 0 1 (0 - 5)
_[4.447686] RLB_PRO: UL: 1 1 (0 - 5) | DL: 1 1 (0 - 5)
_[4.454468] RLB_PRO: UL: 2 1 (0 - 5) | DL: 2 1 (0 - 5)
_[4.457443] RLB_PRO: [MAC] 0x03
[4.460336] RLB_PRO: 00000000: 30 0
[4.471234] RLB_PRO: LinkAdrReq: dataRate = 3, txSteps = 0, nbTrans = 0
[4.477083] RLB_PRO: LinkAdrAns: 07
[4.480661] RLB_PRO: [MAC] 0x04
[4.482599] RLB_PRO: 00000000: 07 .
[4.493576] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
[4.499151] RLB_PRO: [MAC] 0x05
[4.499484] RLB_PRO: 00000000: 00 d2 ad 84 ....
[4.510353] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq =
[4.516470] RLB_PRO: [MAC] 0x08
[4.516670] RLB_PRO: 00000000: 01 .
[4.527097] RLB_PRO: RXTimingSetupReq: delay = 1 sec
[4.532672] RLB_PRO: [MAC] 0x09
[4.532893] RLB_PRO: 00000000: 05 .
[4.543862] RLB_PRO: [MAC] 0x0c
[4.544085] RLB_PRO: 00000000: 65 e
[4.555031] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
[4.560698] RLB_PRO: [MAC] 0x0f
[4.560906] RLB_PRO: 00000000: fa .
[4.571789] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
[4.646314] RLB_PRO:
[4.646394] RLB_PRO: PHY: Frequency = MHz, TX = 16 dBm
[4.649505] RLB_PRO: LoRa: SF = 9, BW = kHz, CR = 4/5, IQ: U
[4.864115] RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
[4.865780] RLB_PRO: 00000000: 00 1e 00 00 00 00 00 00 00 99 31 61 20 18 50 33 ..........1a .P3
[4.876457] RLB_PRO: 00000010: 42 00 00 42 18 b3 d1 B..B...
[4.887706] RLB_PRO:
[4.887787] RLB_PRO: PHY: Frequency = MHz, TX = 16 dBm
[4.890887] RLB_PRO: LoRa: SF = 9, BW = kHz, CR = 4/5, IQ: D
[9.864041] RLB_PRO: Opening Rx1 window (110 ms timeout)... <-- Rx Delay end
[9.978003] RLB_PRO: Closing Rx1 window
[9.978563] RLB_PRO:
[9.978647] RLB_PRO: PHY: Frequency = MHz, TX = 16 dBm
[9.984803] RLB_PRO: LoRa: SF = 12, BW = kHz, CR = 4/5, IQ: D
x[10.864048] RLB_PRO: Opening Rx2 window (670 ms timeout)... <-- Rx Delay end
[11.537003] RLB_PRO: Closing Rx2 window
[LORA] failed, code RADIOLIB_ERR_NO_JOIN_ACCEPT

lyusupov, 2025-03-20:
**duplicates** Discussions thread https://github.com/jgromes/RadioLib/discussions/1463

HeadBoffin, 2025-03-20:
Indeed, a duplicate that's unlikely to be an issue in the library, more the implementation.

closed 2025-03-20 by HeadBoffin