jgromes/RadioLib #1466: LLCC68 incorrect pin check on CAD callback

RobertMorton56, 2025-03-24:
Hi,

Firstly, what a great library. Thanks for the code. I've worked with several LoRa modules over the past while and they all work great with this library but the following doesn't make sense to me.

* Hardware Aithinker RA-01H

The code for the SX127X states DIO0 is activated:

    /*!
      \brief Interrupt-driven channel activity detection method. DIO0 will be activated when LoRa preamble is detected.
      DIO1 will be activated if there's no preamble detected before timeout.
      \returns \ref status_codes
    */

And sure enough checks getIrq() in the while loop:

```
while(!this->mod->hal->digitalRead(this->mod->getIrq())) {
    this->mod->hal->yield();
    if(this->mod->hal->digitalRead(this->mod->getGpio())) {
      return(RADIOLIB_PREAMBLE_DETECTED);
    }
  }

```

**Here's the rub:**

* Hardware LLCC68 module (from Dreamlnk)

From SX126X:
    For startChannelScan 

    /*!
      \brief Interrupt-driven channel activity detection method. DIO1 will be activated
      when LoRa preamble is detected, or upon timeout. Defaults to CAD parameter values recommended by AN1200.48.
      \returns \ref status_codes
    */

Yet scanChannel does:

```
while(!this->mod->hal->digitalRead(this->mod->getIrq())) {
    this->mod->hal->yield();
  }
```

So for the LLCC68 radio, if you swap the module init pin definitions it works:

`LLCC68 radio = new Module( PIN_SPI_SS, PIN_SPI_DIO1, PIN_SPI_RST, PIN_SPI_DIO0 );`

I know I'm nitpicking here but should the scanChannel() code in the SX126X not check getGpio() in the while loop?

Thanks again,
Bob

jgromes, 2025-03-24:
The key thing to realize here that LLCC68 (aka SX126x), unlike SX127x, does not have DIO0. So the real question here is where you got the pin mapping for the LLCC68, because its last pin is not DIO0, but BUSY. It's not an interrupt signal.

The logic is that the second argument of `Module` is the default/most used interrupt pin - on SX127x it's DIO0, on SX126x it's DIO1. The final argument is an additional GPIO pin, which may be an interrupt or some other signal.

RobertMorton56, 2025-03-24:
Ah! OK, got you. So proving all pins are wired up to the micro, a valid initialising function for the LLCC68 module might be the following?

`LLCC68 radio = new Module( NSS, DIO1, RST, DIO3 );`

[image]

Does Radiolib use the Busy pin for the LLCC68 module at all? 

Thanks again...

closed 2025-03-24 by RobertMorton56

jgromes, 2025-03-24:
> LLCC68 radio = new Module( NSS, DIO1, RST, DIO0 );

No, as per the SX126x examples, the last pin is BUSY. 

> Does Radiolib use the Busy pin for the LLCC68 module at all?

Yes, without it the SPI communication is less reliable, as the SX126x/LLCC68 (same thing) uses the BUSY line to signal when the device is ready to accept new commands. If your design is missing the BUSY pin, you can use `RADIOLIB_NC` but at that point RadioLib will just wait for some time between consecutive SPI commands, which does not guarantee something won't collide.

RobertMorton56, 2025-03-24:
Brilliant, understood. Thanks for the clarification and help!