jgromes/RadioLib #1468: SX1262 Receiving Its Own Packets

bowser-jacob, 2025-03-26:
**Describe the bug**
When I attempt to receive a packet after transmitting one (even with a delay) the first packet to be received will not be real, ie. no packet was actually received but the transmitter *thinks* there is one. This "imaginary" packet contains the data field of the previous transmission and the RSSI, SNR, etc of the previous packet received. I don't know how transmit and receive data are getting mixed, but here we are.

<details><summary>Debug mode output. NOTE: every second block of serial output is a "bad" packet </summary>
<p>

```
[SX1262] Received packet!
[SX1262] Data:		IDLE
[SX1262] RSSI:		-4.00 dBm
[SX1262] SNR:		4.75 dB
[SX1262] Frequency error:	-3616.22 Hz

[SX1262] Received packet!
[SX1262] Data:		TEST
[SX1262] RSSI:		-4.00 dBm
[SX1262] SNR:		4.75 dB
[SX1262] Frequency error:	-3616.22 Hz

[SX1262] Received packet!
[SX1262] Data:		IDLE
[SX1262] RSSI:		-4.00 dBm
[SX1262] SNR:		5.00 dB
[SX1262] Frequency error:	-3625.43 Hz

[SX1262] Received packet!
[SX1262] Data:		TEST
[SX1262] RSSI:		-4.00 dBm
[SX1262] SNR:		5.00 dB
[SX1262] Frequency error:	-3625.43 Hz

```

</p>
</details> 

**To Reproduce**
<details><summary>Sketch that is causing the module to fail</summary>
<p>

```c++
#include <RadioLib.h>

#define LORA_MISO 8   // confirmed
#define LORA_SCK 7    // confirmed
#define LORA_MOSI 9  // confirmed
#define LORA_CS 41    //NSS
#define LORA_DIO2 38
#define LORA_DIO1 39     // irq
#define LORA_RESET 42
#define LORA_BUSY 40
#define LORA_RF_SW 38

float frequency = 915.0;
float bandwidth = 62.5;   //125.0;
uint8_t spreadingFactor = 12;
uint8_t codingRate = 5;  // 5;
uint8_t syncWord = 0xF1;    // 0x12;    // try   Serial.println("Set syncronize word to 0x3444");
float outputPower = 22;   //17;
uint16_t preambleLength = 8;
String DataString;
String RSSIString;
String SNRString;
String HzString;

SX1262 radio = new Module(LORA_CS, LORA_DIO1, LORA_RESET, LORA_BUSY);

volatile bool receivedFlag = false;

void setFlag(void) {
  receivedFlag = true;
}

void setup() {
  Serial.begin(9600);
  pinMode(38, OUTPUT);
  digitalWrite(38, HIGH);

  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin(frequency, bandwidth, spreadingFactor, codingRate, syncWord, outputPower, preambleLength, 1.6, false);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  radio.setPacketReceivedAction(setFlag);
  radio.setDio2AsRfSwitch(true);

  Serial.print(F("[SX1262] Starting to listen ... "));
  state = radio.startReceive();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

}

void loop() {
   
  if(receivedFlag) {
    receivedFlag = false;

    String str;
    int state = radio.readData(str);
    if (state == RADIOLIB_ERR_NONE) {
      Serial.println(F("[SX1262] Received packet!"));
      Serial.print(F("[SX1262] Data:\t\t"));
      Serial.println(str);

      if(str=="IDLE"){

        digitalWrite(38, LOW); //RF Switch to RF1
        Serial.println("Transmitting...");
        //Serial.println(RSSIString);
        radio.transmit("TEST");
        delay(2000);
        digitalWrite(38, HIGH); //RF Switch to RF2
        radio.setPacketReceivedAction(setFlag);
        radio.startReceive();
      }

    } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);

    }
  }
}
```

</p>
</details> 

**Expected behavior**
Using another transmitter, I am sending the word "IDLE" every 8 seconds. The serial output should therefore print "Recieved Packet!" followed by "Data: IDLE", because "IDLE" is the only data being transmitted. The radio should then transmit a packet with "TEST", then receive another packet with "IDLE". However, the radio receives the first packet, then after transmitting "receives" a *second* packet with the same RSSI, SNR, delta_Freq, but a data field that says "TEST". It is as if when I transmit, packet to be transmitted gets copied into the recieve data (except for RSSI and other figures which remain constant).

**Additional info (please complete):**
 - MCU: ESP32S3
 - Link to Arduino core: https://github.com/espressif/arduino-esp32
 - Wireless module type SX1262 (Part of the SEEED Wio-SX1262 module).
 - Arduino IDE version: 2.3.4
 - Library version: 7.1.2

jgromes, 2025-03-26:
The problem here is that you seem to have combined two different approaches - an interrupt-driven reception using `startReceive`/`readData` and the blocking `transmit`. Just like packet reception, transmission end is also signaled on the DIO1 pin, which will trigger an interrupt. That interrupt will set the flag causing the next iteration of the `loop` to immediately go to `readData`, where it will pull out data from the SX126x shared transmit/receive buffer.

To solve this, you have to stick with a single approach, I highly recommend interrupt-driven. You will have to call `startTransmit` and handle the signalization from `setFlag` differently based on whether it was caused by receiving new packet, or sending one. The ping pong example should give you an idea how to achieve this.

argetlam-coder, 2025-03-31:
I have the same problem when I try to use the SX1262 module as LoRaWAN node. I get a RADIOLIB_ERR_CRC_MISMATCH error after every successful uplink if the uplink is not followed by a downlink. If the uplink is followed by a real downlink, then it works without error.

jgromes, 2025-04-05:
@argetlam-coder I don't think this is the same issue - @bowser-jacob did not report getting `RADIOLIB_ERR_CRC_MISMATCH` and does not use LoRaWAN. `RADIOLIB_ERR_CRC_MISMATCH` will also get reported by the LoRaWAN stack if the message integrity code check of a downlink fails.

jgromes, 2025-04-10:
Closed deu to inactivitiy.

closed 2025-04-10 by jgromes