jgromes/RadioLib #1471: Join Failed - RADIOLIB_ERR_NO_JOIN_ACCEPT (-1116) on TTGO LoRa32 V1 on AU915

ElectronicallyE, 2025-03-30:
Hello,

I'm been trying to get my TTGO LoRa32 connected to a V3 instance of The Things Network. I can confirm the gateway is configured for V3. I can't see to get it to connect.

I'm getting the error (-1116), which I understand means Source: [https://jgromes.github.io/RadioLib/group__status__codes.html](https://jgromes.github.io/RadioLib/group__status__codes.html)

**LoRaWAN_Starter.ino**
```
/*
  RadioLib LoRaWAN Starter Example

  ! Please refer to the included notes to get started !

  This example joins a LoRaWAN network and will send
  uplink packets. Before you start, you will have to
  register your device at https://www.thethingsnetwork.org/
  After your device is registered, you can run this example.
  The device will join the network and start uploading data.

  Running this examples REQUIRES you to check "Resets DevNonces"
  on your LoRaWAN dashboard. Refer to the network's 
  documentation on how to do this.

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/

  For LoRaWAN details, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/LoRaWAN

*/

#include "config.h"

void setup() {
  Serial.begin(115200);
  while(!Serial);
  delay(5000);  // Give time to switch to the serial monitor
  Serial.println(F("\nSetup ... "));

  Serial.println(F("Initialise the radio"));
  int16_t state = radio.begin();
  debug(state != RADIOLIB_ERR_NONE, F("Initialise radio failed"), state, true);

  // Set spreading factor to SF12
  radio.setSpreadingFactor(12);
  
  // Setup the OTAA session information
  state = node.beginOTAA(joinEUI, devEUI, nwkKey, appKey);
  debug(state != RADIOLIB_ERR_NONE, F("Initialise node failed"), state, true);
  Serial.println(F("Join ('login') the LoRaWAN Network"));
  state = node.activateOTAA();
  debug(state != RADIOLIB_LORAWAN_NEW_SESSION, F("Join failed"), state, true);

  Serial.println(F("Ready!\n"));
}

void loop() {
  Serial.println(F("Sending uplink"));

  // This is the place to gather the sensor inputs
  // Instead of reading any real sensor, we just generate some random numbers as example
  uint8_t value1 = radio.random(100);
  uint16_t value2 = radio.random(2000);

  // Build payload byte array
  uint8_t uplinkPayload[3];
  uplinkPayload[0] = value1;
[... 260 lines cut ...]
    "device_channel_index": 13,
    "consumed_airtime": "0.370688s"
  },
  "correlation_ids": [
    "gs:uplink:01JQKA6APVM4FEYZT20D6SA64A"
  ],
  "origin": "ip-10-102-7-79.ap-southeast-2.compute.internal",
  "context": {
    "tenant-id": "CgN0dG4="
  },
  "visibility": {
    "rights": [
      "RIGHT_APPLICATION_TRAFFIC_READ"
    ]
  },
  "unique_id": "01JQKA6APZ61B8E7FA6CBXWKBG"
}
```

**Additional Info:**
 - MCU: TTGO LoRa32 SX1276 V1
 - Link to Arduino core: esp32 by Espressif Systems https://github.com/espressif/arduino-esp32 (Board: TTGO LoRa32-OLED)
 - Wireless module type: SX1267
 - Arduino IDE version: 2.3.4
 - Library version: 7.1.2

StevenCellist, 2025-03-30:
Hi @ElectronicallyE, great report with lots of crucial info! As far as I scanned the details now, it looks pretty solid. I think we'll need to see the protocol debug log of your device to confirm whether your device is correctly listening at the right moments for the JoinAccept. You can check the wiki how to enable (protocol) debug output. Please enable timestamps in your terminal: they're the most important part of the log!

Oh.. and LoRaWAN overrides all calls to `radio` functions such as your `setSpreadingFactor(12)`. The right way to do this would be `node.activateOTAA(0)` (this uses datarates, not spreading factor), however, on AU915 dwelltime limitations are in place, and the specification forces JoinRequests to be sent at DR2 = SF10, so that trick won't work!

ElectronicallyE, 2025-03-31:
Hi @StevenCellist,

Thank you for your quick reply and positivity. It's refreshing to have positive feedback!

I have enabled RADIOLIB_DEBUG_PROTOCOL in BuildOptUser.h as per [https://github.com/jgromes/RadioLib/wiki/Debug-mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

This was tested LoRaWAN Specification 1.1.0 and RP002 Regional Parameters 1.0.4. Using the code above, with the spreading factor line removed.

Arduino serial monitor output:
```
13:59:34.910 -> RLB_PRO: RXTimingSetupReq: delay = 1 sec
13:59:34.942 -> RLB_PRO: [MAC] 0x09
13:59:34.942 -> RLB_PRO: 00000000: 1d                                               .                 
13:59:34.942 -> RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
13:59:34.942 -> RLB_PRO: [MAC] 0x0c
13:59:34.942 -> RLB_PRO: 00000000: 65                                               e                 
13:59:34.974 -> RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
13:59:34.974 -> RLB_PRO: [MAC] 0x0f
13:59:34.974 -> RLB_PRO: 00000000: fa                                               .                 
13:59:35.014 -> RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
13:59:35.047 -> RLB_PRO: 
13:59:35.047 -> RLB_PRO: PHY:  Frequency = 918.200 MHz, TX = 30 dBm
13:59:35.092 -> RLB_PRO: LoRa: SF = 10, BW = 125.0 kHz, CR = 4/5, IQ: U
13:59:35.453 -> RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
13:59:35.453 -> RLB_PRO: 00000000: 00 00 00 00 00 00 00 00 00 5a f8 06 d0 7e d5 b3  .........Z...~..
13:59:35.453 -> RLB_PRO: 00000010: 70 00 00 f6 e1 53 91                             p....S.           
13:59:35.499 -> RLB_PRO: 
13:59:35.499 -> RLB_PRO: PHY:  Frequency = 927.500 MHz, TX = 30 dBm
13:59:35.499 -> RLB_PRO: LoRa: SF = 10, BW = 500.0 kHz, CR = 4/5, IQ: D
13:59:43.108 -> RLB_PRO: Opening Rx1 window (77 ms timeout)... <-- Rx Delay end 
13:59:43.225 -> RLB_PRO: Closing Rx1 window
13:59:43.271 -> RLB_PRO: 
13:59:43.271 -> RLB_PRO: PHY:  Frequency = 923.300 MHz, TX = 30 dBm
13:59:43.271 -> RLB_PRO: LoRa: SF = 12, BW = 500.0 kHz, CR = 4/5, IQ: D
13:59:44.684 -> RLB_PRO: Opening Rx2 window (191 ms timeout)... <-- Rx Delay end 
13:59:44.948 -> RLB_PRO: Closing Rx2 window
13:59:44.948 -> Join failed - RADIOLIB_ERR_NO_JOIN_ACCEPT (-1116)
```

Here is a screenshot from the TTN Console.
[image]

StevenCellist, 2025-03-31:
Wow, this is wild, I've never seen an ESP32 perform so badly. I guess Australia is warmer than the Netherlands but 50% off is insane. Do you by any chance have some normal ESP32 boards around, just to confirm that they run more normally?

RadioLib offers a method for clock drift compensation but its intended to fix (stable) drifts in the order of 10s of milliseconds per second - this issue must be fixed in another way. As I have never seen anything like this, I wouldn't have a clue how to resolve it.

ElectronicallyE, 2025-03-31:
Thanks for another quick reply @StevenCellist.

I've just tried the same delay sketch on an ESP-WROOOM32 which appears to work fine (using ESP32 Dev Module as the board). Looks like it doesn't face the same problem

```
16:01:22.344 -> 1 second
16:01:23.360 -> 1 second
16:01:24.344 -> 1 second
16:01:25.381 -> 1 second
16:01:26.363 -> 1 second
16:01:27.370 -> 1 second
16:01:28.388 -> 1 second
16:01:29.389 -> 1 second
```

I've tried a third TTGO LoRa32 V1 which has the same amount of drift of 1.5 seconds for every second.

The ESP32 chip on the TTGO LoRA32 V1 is a ESP32 DOWDQ6.

In order to read the serial monitor, I have been setting the baud rate to 74880, even though in the script it is 115200. My understanding is because the chip uses a 26 MHz crystal, instead of a 40 MHz crystal ((26/40) * 115200 = 74880). 115200/74880 = 1.538, which is close enough to the drift. I didn't think until now that this would mean this would also affect the chips time (obvious now given it affects the baud rate)!

[https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32c2/get-started/establish-serial-connection.html](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32c2/get-started/establish-serial-connection.html)
The default console baud rate on ESP32-C2 is 115200 when a 40 MHz XTAL is used, or 74880 when a 26 MHz XTAL is used.

Here's a few links describing a similar problem:
https://github.com/espressif/arduino-esp32/issues/9837 <- This is an issue called out about the board library and calls out "[Heltec Lora boards](https://github.com/espressif/arduino-esp32/issues/9837#issuecomment-2162483466)" which are very similar to mine.
https://forum.arduino.cc/t/baudrate-changes-after-boot/1337352
https://www.esp32.com/viewtopic.php?t=43788
https://github.com/sparkfun/ESP32_Thing_Plus/issues/7
https://community.sparkfun.com/t/sparkfun-esp32-thing-issue-with-espressif-version-3-0-0-core-clock-incorrect-frequency/47531

Downgrading the _esp32 by Espressif Systems_ board library to 2.0.17 seems to work a treat!

I reupload the following code:
```
void setup() {
    Serial.begin(115200); // Start serial communication at 115200 baud
}

void loop() {
    Serial.println("1 second"); // Print message to serial
    delay(1000); // Wait for 1 second
}
```

Please see the latest results below. **Can you confirm if the variation will cause any connection issues (how much leniency is there)?** 

[image]

Tried reuploading the code and I'm in business! [You little ripper!](https://youtu.be/Gp6f1EMn1lo?si=CU9eHgUqeJ9c31xx&t=246)

Screenshot from the console (might have to zoom in):
[image]

Output from serial monitor:
```
17:27:05.646 -> ets Jun  8 2016 00:22:57
17:27:05.646 -> 
17:27:05.646 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
17:27:05.646 -> configsip: 0, SPIWP:0xee
17:27:05.646 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:27:05.646 -> mode:DIO, clock div:1
17:27:05.646 -> load:0x3fff0030,len:1184
17:27:05.646 -> load:0x40078000,len:13260
17:27:05.691 -> load:0x40080400,len:3028
17:27:05.691 -> entry 0x400805e4
17:27:10.804 -> 
17:27:10.804 -> Setup ... 
17:27:10.804 -> Initialise the radio
17:27:10.804 -> Join ('login') the LoRaWAN Network
[... 88 lines cut ...]
17:27:22.254 -> RLB_PRO: 00000020: 8a 8c a7 63 4d 59                                ...cMY            
17:27:22.254 -> RLB_PRO: [MAC] 0x0b
17:27:22.254 -> RLB_PRO: 00000000: 01                                               .                 
17:27:22.254 -> RLB_PRO: RekeyConf: server version = 1.1
17:27:22.254 -> RLB_PRO: [MAC] 0x09
17:27:22.254 -> RLB_PRO: 00000000: 1d                                               .                 
17:27:22.286 -> RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
17:27:22.286 -> RLB_PRO: [MAC] 0x03
17:27:22.286 -> RLB_PRO: 00000000: 41 00 ff 00 00 00 00 00 00 00 00 00 00 01        A.............    
17:27:22.286 -> RLB_PRO: LinkAdrReq: dataRate = 4, txSteps = 1, nbTrans = 1
17:27:22.286 -> RLB_PRO: UL:   8 1 916.800 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.318 -> RLB_PRO: UL:   9 1 917.000 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.318 -> RLB_PRO: UL:  10 1 917.200 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.318 -> RLB_PRO: UL:  11 1 917.400 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.318 -> RLB_PRO: UL:  12 1 917.600 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.318 -> RLB_PRO: UL:  13 1 917.800 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.318 -> RLB_PRO: UL:  14 1 918.000 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.355 -> RLB_PRO: UL:  15 1 918.200 (0 - 5) | DL:   0 0   0.000 (0 - 0)
17:27:22.355 -> RLB_PRO: LinkAdrAns: 07
17:27:22.355 -> RLB_PRO: [MAC] 0x06
17:27:22.355 -> RLB_PRO: DevStatusAns: status = 0xff06
17:27:22.355 -> Received a downlink
17:27:22.355 -> Next uplink in 300 seconds
17:27:22.355 -> 
```

StevenCellist, 2025-03-31:
Great digging! Weird results with the baud rate - in the latest link I see someone having success again on Arduino-ESP32 core v3.0.2, what version were you on?

The clock stability isn't looking particularly wonderful: I see variations of up to 45ms on your table. I recommend you keep your device running for a while and try sending it a few downlinks every now and then to see if it receives all of them. 
If not, you may want to increase the value of `node.scanGuard` (which basically widens the Rx window). See here:

https://github.com/jgromes/RadioLib/blob/ec785005aa607b4da7c0e5f4367ebf379a43e477/src/protocols/LoRaWAN/LoRaWAN.h#L856-L868

The current value of 5ms is the minimum stable value across a variety of modern boards we've encountered, including a bunch of LoRa32 v2's and v3's - not sure we've had recent reports of v1's. But the same MCU is on the v2 vs the v1, and we haven't seen any stability problems with ESP32. So maybe there's simply a lot of buffering on the UART which makes it _look_ like it is instable, and in reality it works just fine with the 5ms scan guard. Just test a bit for yourself :)

StevenCellist, 2025-03-31:
No reason to change the scanGuard if all went well. 

> The TTN console reports a "RX1 Delay" of "5" for all downlinks, even those which aren't "join accept" downlinks that should be 1s for RX1. Do you know why this is the case? Does your library adjust the delay correctly for a "join accept" downlink versus a "normal" downlink?

TTN uses a 5 second Rx1 delay which is communicated through the JoinAccept which is parsed by RadioLib. This 5 second delay allows e.g. webhooks to schedule downlinks based on the uplink. With the default 1s delay, it can be pretty hard to schedule a downlink on time.

> Does TTN restrict the number of downlinks and is there any logic behind this (which ones it allows versus blocks)?

TTN FUP limits to at most 10 downlinks a day, but you should regard LoRaWAN as uplink-only except for maybe a configuration change every few days or weeks. Or maybe a 'ping' once a day. Any use case that requires regular downlinks isn't fit for (public) LoRaWAN networks. TTN doesn't automatically block downlinks, but if you happen to spike the usage graphs enough for someone at the TTI towers to notice, someone will push a button to terminate your activity. So for testing, best to get a TTS Discovery instance which doesn't limit FUP.

> On another note, I figured out how to do confirmed uplinks via LoRaWAN_Reference.ino. Is this the best source for all the commands and parameters that can be applied or is there a list with descriptions elsewhere?

It's surely the most comprehensive, but otherwise there's the API documentation which lists all available functions.

closed 2025-03-31 by StevenCellist

ElectronicallyE, 2025-04-01:
Thanks @StevenCellist for your explanations once again.

> TTN uses a 5 second Rx1 delay which is communicated through the JoinAccept which is parsed by RadioLib.

If the default RX1 downlink window is 5 seconds (regardless of a "JoinAccept" or not, why do they say it is 1s? Am I missing something?

> TTN doesn't automatically block downlinks, but if you happen to spike the usage graphs enough for someone at the TTI towers to notice, someone will push a button to terminate your activity.

So in my case, would you suspect this is was manually capped/restricted? It seemed to allow a couple of downlinks on the reboot without issue.

StevenCellist, 2025-04-01:
RP002-1.0.4 recommends Rx1Delay to be 5 seconds, which TTN apparently put in their docs (is that a screenshot from the v2 docs? formatting looks more like v2 than v3 - TTN did use 1s on v2). They use 5s though.

A handful of downlinks doesn't spike the graphs at once - your screenshot didn't show though that you're missing downlinks?

ElectronicallyE, 2025-04-01:
**RX1 Delay**
That screenshot regarding delays is from https://www.thethingsnetwork.org/docs/lorawan/regional-parameters/ which must be for V2. Not sure why they keep this online if it's out of date?

I've found the latest documentation here:
https://www.thethingsindustries.com/docs/getting-started/glossary/

_Class A
Class A devices support bi-directional communication between a device and a gateway. Uplink messages (from the device to the server) can be sent at any time (randomly). The device then opens two receive windows at specified times after an uplink transmission. The first receive window (Rx1) is set to 5 seconds and can be modified. The second receive window (Rx2) always comes 1 second after Rx1.
https://www.thethingsnetwork.org/docs/lorawan/classes/#class-a_

**Missing Downlinks**
Correct, no downlinks are being missed, but the code requests confirmed uplinks...

StevenCellist, 2025-04-01:
An uplink is confirmed by the LNS by sending a downlink with the Ack bit set. Which, if you inspected the downlink JSON, you would have seen set to `true`.

ElectronicallyE, 2025-04-01:
Correct, however I should have clarified that better.

I meant that every 30 seconds an uplink was being sent requesting it be confirmed. For the bulk of the transmissions, no confirmation downlink was received. The above list were exclusively uplinks with accompanying downlinks. There were a number without one.