jgromes/RadioLib #1472: Issue with SX1262 Transmission Frequency

PawnMa, 2025-04-02:
Hello,

I am using an interrupt-based method to send data packets. During the process, I noticed that the time taken from sending to the interrupt is always approximately 6ms slower than the time calculated by `getTimeOnAir`. . I have ensured that there are no unnecessary delays in the program. For example, when the theoretical frequency is calculated as 106Hz, the actual transmission rate is only around 65Hz.

I would like to ask if this delay is necessary, and if it can be optimized or reduced through configuration. This is critical for our application, as we are working on an RC model project where the fastest possible communication frequency is very important.

Thank you for your help!

Best regard

jgromes, 2025-04-02:
`getTimeOnAir` does what it says in the name - it returns the time on air of a packet for given length and LoRa configuration. However, before the packet starts transmission, a lot of things need to happen with the radio - it needs to be configured properly, and the data itself has to be sent over the SPI, that's where the "extra delay" comes from. There are ways to optimize this (e.g. by using faster than default SPI), but without context about your project, I cannot tell you if that would be enough.

SX126x and LoRa in general is IMO not a good choice for an RC model, because it communicates via packets. For an RC model, I would try to find a radio that can transmit/receive continuously in a stream. SX127x can do it, but there are probably better things out there.

PawnMa, 2025-04-02:
Thank you very much for your response. I have tried changing the SPI speed, but the improvement was minimal. I suspect that the delay is caused internally by the chip. However, I did indeed refer to the open-source [ExpressLRS ](https://github.com/ExpressLRS/ExpressLRS)configuration, which uses the SX127X/SX1280. Their communication frequency can even reach up to 500Hz.

It is possible that this is indeed a limitation of the SX126X itself.

Thank you again for your help!