jgromes/RadioLib #1478: RX2 Delay Incorrect

pchwalek, 2025-04-07:
When performing OTAA, I was analyzing the timings of RX1 and RX2. RX1 appears good but RX2 was off by 2-3 seconds. After looking at the code, I found that Radiolib goes through two iterations of a for loop where it calculates the estimated time delay required before the current receive window. However, upon the second iteration, the time delay estimation should be an addition of the reference clock minus the current clock plus both the current and last time window delay. In the current version of the code, the [below line](https://github.com/jgromes/RadioLib/blob/a88ec8dc0860914e6fad0c266c23a90daf99e8ea/src/protocols/LoRaWAN/LoRaWAN.cpp#L1437C5-L1437C81) will always overflow (i.e., the arithmetic would yield a negative number) for the second window:

` RadioLibTime_t waitLen = tReference + dlDelays[window] - mod->hal->millis();`

StevenCellist, 2025-04-08:
Hi @pchwalek, the line that you mention is indeed sensitive to over(under)flow, but if the radio is properly connected and LNS is properly configured, it should never occur.
To get a couple of things straight:
1. The Rx2 delay is also defined relative to `tReference` (or, as phrased in the spec: relative to `RxDelayEnd`). Which means it should _not_ be calculated as `tReference` + Rx1 + Rx2, but simply as `tReference` + Rx2.
2. The Rx1 window, if not used by the LNS, will never exceed ~700ms. The longest possible empty LoRa frame takes [663.6ms](https://avbentem.github.io/airtime-calculator/ttn/eu868/0,0), plus a little bit of `scanGuard` (for US915, the longest possible empty frame is just [206.8ms](https://avbentem.github.io/airtime-calculator/ttn/us915/0,0)) . As the difference between Rx2 and Rx1 is 1000ms, this will always be safe.

A possible reason for the behavior you notice is that your BUSY or DIO/IRQ pin is not configured correctly and therefore the Rx1 window is not closed correctly. Please verify that your radio is configured properly.

pchwalek, 2025-04-09:
I apologize, I misread the spec. I thought RX2 was referenced from RX1. Thank you for the concise but detailed answer!

closed 2025-04-09 by pchwalek